thesofproject/sof #580: The DMIC driver does not prevent too short microphone clock enables from happening

singalsu, 2018-11-14:
Some multi-mode microphones have a state machine for sleep/sensor/normal modes. It is known that some audio servers probe the audio system with very short capture tests. If the microphone clock burst is too short some microphones may freeze erroneously until next power cycle.

The issue could be prevented by passing a min. clock enabled time parameter for FW. If the capture is at trigger stop shorter than such required time the clock stop can be delayed. The FIFO can be stopped immediately so this feature would not capture extra samples.

Need to decide driver architecture for this. In SOF the DSP could do this entirely if the microphone clock requirements could be passed in similar way as other existing parameters like duty cycle range and mic clock range.

lgirdwood, 2018-11-14:
@singalsu best decide ASAP what data you may or may not need from topology or kernel as ABI will be upstreaming soon.

singalsu, 2018-11-15:
@lgirdwood @plbossart The parameters to add would be "wake_up_time" and "mode_change_time". They should be for any microphone less than 100 ms, more likely 10 - 30 ms. If the unit would be microseconds they would fit well two uint32_t types.

singalsu, 2018-11-15:
I created SOF PR #584 and matching Linux PR https://github.com/thesofproject/linux/pull/292 IPC for parameters those enable doing this enhancement with later done patches.

plbossart, 2018-11-15:
@singalsu as commented on
https://github.com/thesofproject/linux/pull/292, I don't quite get how the two parameters match the problem statement you described initially?

singalsu, 2018-11-15:
OK, I think I now get your point. Would it be better that these two parameters would only control mute (some microphones keep their output tri-stated until wake_up_time is reached) and add third min_clock_on_time?

The parameter is not found from datasheets but we can add comment to header and topology files that set it to wake_up_time if not known.

Would this be clear?

plbossart, 2018-11-15:
my understanding is this:
- wake_up_time: delay before clock on and data provided to clients
- min_clock_on_delay: clock can only be turned off once this delay expires. Data discarded immediately on stop

I don't get what mode_change relates to and the 'Time from clock change to data (us)' comment means. 
Also not sure about how this related to mute?

singalsu, 2018-11-15:
The data sheet that mentioned tri-stating until wake_up_time did not describe what happens for data line when microphone mode is changed (by setting the mic clock below or above the mode change threshold). 

I didn't know the fact that the line is tri-stated intentionally in some microphones when capture starts. In that case it's best to mute the entire time the CIC and FIR to prevent the tri-stated PDM data line to mess up DCCOMP.

My idea of using mode_change_time would be to mute the DMIC HW for mode_change_time if the running clock rate has been changed, e.g. 768 kHz <-> 2.4 MHz (needs also re-program of decimation ratio related registers and possibly even FIR coefficients).

The minimum change to do now would be really only the **min_clock_on_time** and leave out those two.

But since we upstream the IPC now I thought that these mic data sheets parameters would be good to become reserved too since we can use them to control muting in HW.

plbossart, 2018-11-15:
I am not sure how we would handle a dynamic clock frequency change without stopping the capture entirely and restarting it, e.g. when moving to a low-power usage with voice trigger handled on the DSP. If we had this requirement we'd need quite a bit of plumbing across the stack.

wake_up and min_clock_on_time are the only things we can really test without significant changes across firmware+driver for now, so let's add them as placeholders. We can the implementation later and use arbitrarily large delays for testing and check what happens.

singalsu, 2018-11-15:
OK, I'll update both PRs this way. Thanks!

michalgrodzicki, 2019-03-26:
@singalsu any update on this?

jajanusz, 2019-06-06:
@singalsu Was it addressed in some PR? Can I close it?

singalsu, 2019-06-06:
@jajanusz I do have now suitable test HW with the mic model that is sensitive to short clock bursts to verify if this can happen. I'll update shortly my results. If it happens the fix will be quite simple on FW side.

Though we might need another DMIC token into driver side. The IPC struct has been already updated with placeholders for the needed data.

singalsu, 2019-06-07:
@jajanusz @michalgrodzicki @plbossart I did the re-verify. The conclusion is that I'm not able to get the microphone state freeze to mute happen with any very short captures prior to a longer test capture. 

I tested with 2 ms increment to do capture of 10, 12, 14, ... 50 ms. The range 50 - 100 ms range I tested with 10 ms steps. I inspected the PCM samples level from the 5.0s long capture after each of these very short captures. I also checked from sof-logger output the time stamps for trigger START and PAUSE/STOP and they matched my capture time time parameter.

It may be that the issue has been fixed in newer HW revision of the reported microphone. My DMICs of this type have been purchased this year while this report is from last year. 

Related to test setup for this, arecord doesn't support capture (seconds) duration as floating point number but ecasound does. The commands I used for recording were, e.g. 

```
ecasound -f:s16_le,2,48000 -b:128 -t 0.020 -i alsahw,0,6 -o rec48k.wav
ecasound -f:s16_le,2,48000 -b:128 -t 5.0 -i alsahw,0,6 -o rec48k.wav

```

I feel it's safe to close this issue now.

singalsu, 2019-06-07:
Closing this issue.

closed 2019-06-07 by singalsu