thesofproject/sof #581: APL gcc toolchain cannot build xtensa-smp

jajanusz, 2018-11-14:
Our toolchain for apl is bugged, it should have threadptr support but, it does not.
What results in errors like:
`Error: unknown opcode or format name 'rur.threadptr'` when trying to compile apl for smp

lgirdwood, 2018-11-14:
@jajanusz which GCC version do you have ? iirc, I added threadptr a while ago

jajanusz, 2018-11-14:
@lgirdwood gcc 8.1 from today from https://github.com/thesofproject/crosstool-ng
Travis had the same problem: https://travis-ci.org/thesofproject/sof/jobs/454734132#L1663

lgirdwood, 2018-11-14:
@xiulipan is the travis CI container using latest overlay HEAD (which has THREADPTR register support)

lgirdwood, 2018-11-14:
@jajanusz doubled checked and some overlays were missing. now pushed.  @xiulipan CI container will need updated.

xiulipan, 2018-11-15:
@lgirdwood 
I will update the docker image ASAP.

xiulipan, 2018-11-15:
@lgirdwood @jajanusz 
docker image update done. Please have a push to have a test with #570

jajanusz, 2018-11-15:
@xiulipan @lgirdwood 
Thanks for updates, now it works

closed 2018-11-15 by jajanusz