thesofproject/sof #583: scratching noise when pause/release HDMI audio on Yorp

keqiaozhang, 2018-11-15:
This noise issue is happened when pause/release HDMI playback, the possibility is less than 5% and there's no obvious error trace.
The recorded wav file is attached. 
BTW, I tried to reproduce this issue with sine wav, but failed.
[noise.zip](https://github.com/thesofproject/sof/files/2583835/noise.zip)

Kernel-topic/cros-sof-v4.14-rebase: a38231
DSP FW-origin/glk-004-drop-stable: 1f799207
Topology-origin/master: 5894ed

lgirdwood, 2018-11-15:
@keqiaozhang is this easy to reproduce with your orig wav file and cannot be reproduces with sine wave ? Please check sine wave more (and check with audacity) as this would possibly indicate an issue with gain ? rather than DMA. I assume there is also no such issue with SSP and HDMI is using passthrough pipeline ?

slawblauciak, 2018-11-16:
I noticed a couple rather interesting peculiarities with this.
For once, often times **on pause**, soon before this bug manifests itself I can hear a very short audible burst of this metallic noise.
Secondly, very often the bug doesn't show up immediately after releasing, but after some time, like half a second or so. It also sometimes goes away after a second too.

I maintain a very low reproduction rate, around 1%.

keqiaozhang, 2018-11-27:
@slawblauciak 
I got the same result as you. The noise happens only when pausing or releasing the playback and goes away after <1 second and the possibility is low, less than 5%.
Before the noise will last until next pause/release. 
Tested with
kernel-cros-sof-v4.14-rebase: f8f388ef
sof-master: 443233
tplg-master: ad3f3e4

lgirdwood, 2018-11-29:
@keqiaozhang can you provide some audacity screens shots and wav of the noise. Thamks

keqiaozhang, 2018-11-30:
@lgirdwood 
I tested a lot with 48K sine wave, but failed to reproduce it.
@slawblauciak 
Could you help to provide the audacity screens shots? I cannot play 8k sine wave without the plug plugin on Yorp.

slawblauciak, 2018-11-30:
@keqiaozhang you shouldn't need the plug if the sampling rate is 48 kHz. Just the sinewave should be at 8 kHz.

I will try to provide you with a screenshot

keqiaozhang, 2018-11-30:
Thanks @slawblauciak. I misunderstood, I just generated a 48K_16bit_2ch_800hz sine wav.  I will try it again.

michalgrodzicki, 2019-03-26:
@mengdonglin what is a current status here?

mengdonglin, 2019-03-29:
@keqiaozhang please help to check latest status with glk 1.1

keqiaozhang, 2019-03-29:
@mengdonglin 
I didn't observe this issue in regular test. but I still need to run the stress test to double confirm it. will update it soon

Jiangxinx, 2019-03-29:
Not reproduce on APL-UP2-PCM521X platform.

Env:
sof master: 742f83b
kernel sof-dev: e29122d6
tplg: sof-apl-pcm512x.tplg

lgirdwood, 2019-03-29:
@Jiangxinx @keqiaozhang can we close ?

keqiaozhang, 2019-04-01:
This issue cannot be reproduced on GLK. We can close this issue now.
SOF/tplg: glk-011-drop-stable / 55ae67b5
Kernel driver: cros-sof-v4.14-rebase/a1d6bf0d

closed 2019-04-01 by keqiaozhang