thesofproject/sof #586: EQ configuration data via IPC is corrupt

singalsu, 2018-11-16:
I noticed this with topology sof-apl-eq-dmic.tplg. The trace shows that IIR prepare fails in start of capture.

"pipeline.c:499 	component_op_upstream() error: Component failed, err = -22"

The blob seems to be corrupt except what's in the fixed header.

lgirdwood, 2018-11-16:
@singalsu is this a tools bug or FW bug ? i.e. where is the data being corrupted

singalsu, 2018-11-16:
@lgirdwood I think it's because there's "uint32_t reserved[4]" added to into structs. The binary data in EQ topologies and data for sof-eqctl do not contain them.

We could remove those from eq.h or add the zeros to data. What do you prefer?

lgirdwood, 2018-11-16:
Tools need to be updated to use new ABI, and also send the ABI header at the start of any data. The driver should not insert this header (but should check the ABI being used against driver ABI and FW ABI versions)..

singalsu, 2018-11-16:
OK, I'll do the fixes next week into SOFT topologies, tools, and possible bugs in SOF FW with handling the gap inside configuration data.

singalsu, 2018-11-16:
Just FYI @ranj063 @juimonen There may be an issue in the kernel too. It should have failed with non-matching EQ ABI header. I can fix that next week too.

lgirdwood, 2018-11-16:
@singalsu I'm working on kernel ABI boot check part now.

singalsu, 2018-11-28:
I'm now testing a fixed EQs system. It works for me and I will publish today PRs for kernel and FW to address this. 

Once SOFT is merged to SOF I can make PR for topologies. I have the tools fixed too but that will need to steps. A quick fix PR would be soon ready. The later proper fix PR will have similar Matlab objects ABI as the C headers.

plbossart, 2018-12-06:
@singalsu both kernel and firmware are fixed, can this be closed or do you still have topology/tools updates?

singalsu, 2018-12-07:
Yes, it works now. I'll improve later the the Matlab tools support for ABI but there's no need to keep this open.

closed 2018-12-07 by singalsu