thesofproject/sof #590: Remove pipelines and components on module unload

ranj063, 2018-11-16:
Today the DSP is turned off when the SOF kernel modules are unloaded. Instead when the topology is freed, the firmware should remove the corresponding pipelines and components to be symmetric with what happens during topology loading.

tlauda, 2019-03-23:
@ranj063 Don't we already free pipelines and components on topology unload or am I missing something?

ranj063, 2019-03-23:
@tlauda I suppose the title is not very clear on the requirement. What I meant was today when we load the topology, the driver sends IPC's to the DSP to set up the components and pipelines. But on the other hand when there are no IPC's sent to the DSP when the topology is unloaded. All we do is just power off the DSP. 
I suppose the ask here is to be able to tear down a pipeline and set up a new one just like we do at first boot today.

lgirdwood, 2019-03-24:
@ranj063 the FW infrastructure is already place to tear down pipelines, so we are good to go when the kernel side is ready. I think the difficulty is adding/removing a topology at runtime from userspace. There is currently no kernel interface defined to do this (could be a new ALSA ioctl though).

tlauda, 2019-03-24:
@lgirdwood Should we close this issue and open one on kernel side?

ranj063, 2019-03-25:
@lgirdwood @tlauda I'll move this issue to the kernel repo

closed 2019-03-25 by ranj063