thesofproject/sof #592: Fail to load nocodec topology on UP2

markyang, 2018-11-19:
**Summary:**
  Fail to load nocodec topology on UP2.  kernel 21947cd7 on Nov. 16 can work normally.
  sof-logger is empty.  
  ls -l /lib/firmware/intel/
```
lrwxrwxrwx  1 root root      39 1  29 00:05 sof-apl-nocodec.tplg -> /home/up2/20181119/sof-apl-nocodec.tplg
lrwxrwxrwx  1 root root      29 11 19  2018 sof-apl.ri -> /home/up2/20181119/sof-apl.ri
```
**dmesg:**
```
[    5.927345] sof-audio sof-audio: created machine skl_hda_dsp_generic
[    6.061305] HDMI HDA Codec ehdaudio0D2: Max dais supported: 3
[    6.068121] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp1
[    6.068124] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp2
[    6.068126] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp3
[    6.068206] sof-audio sof-audio: loading topology:intel/sof-hda-generic.tplg
[    6.070404] sof-audio sof-audio: Direct firmware load for intel/sof-hda-generic.tplg failed with error -2
[    6.070410] sof-audio sof-audio: error: tplg intel/sof-hda-generic.tplg load failed with -2
[    6.070473] sof-audio sof-audio: error: failed to load DSP topology -2
[    6.070510] sof-audio sof-audio: ASoC: failed to probe component -2
[    6.070563] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -2
[    6.070649] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -2
[    6.171901] systemd-udevd (1432) used greatest stack depth: 11880 bytes left
[    6.820312] IPv6: ADDRCONF(NETDEV_UP): enp2s0: link is not ready
[    6.822504] Generic PHY r8169-200:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=r8169-200:00, irq=IGNORE)
```
**Test steps:**
aplay -l
aplay: device_list:270: no soundcards found...

**Test env:**
sof master: 977fdd57
sof tool: a01fd666
kernel sof-dev: 38a7c34e
tplg: sof-apl-nocodec.tplg-a01fd666

**Log:**
[logger-apl-t.log](https://github.com/thesofproject/sof/files/2594874/logger-apl-t.log)
[dmesg-apl-nocodec.log](https://github.com/thesofproject/sof/files/2594875/dmesg-apl-nocodec.log)

lgirdwood, 2018-11-19:
@markyang it's looking for "sof-hda-generic.tplg" which is missing from your directory.

plbossart, 2018-11-19:
It's more that you need to disable HDAudio support when you want to try 
nocodec.

xiulipan, 2018-11-20:
@plbossart 
Now if we use config from kconfig, we will fallback to  `skl_hda_dsp_generic` but not `sof_nocodec`?
But the auto test of https://github.com/thesofproject/linux/pull/302 with kconfig will fall back to `sof_nocodec`. From the test logs, I find this machine driver change happens since https://github.com/thesofproject/linux/pull/282

I am quite confused here. If it means we'd like to disable HDA link for default from https://github.com/thesofproject/linux/pull/302. And later we can still use sof-apl-nocodec.tplg for test?

mengdonglin, 2018-11-20:
I feel we need to keep HDA link by default since SOF needs skl_hda_dsp_generic for WHL.  And OSV may also tend to enable it to support as many platforms as possible for end users in the future.

And now we enabled nocodec mode for validation.

What will happen on UP2 w/o the codec, if we enable both HDA links/codec and nocodec mode?
Ideally, we hope to fallback to the nocodec machine driver anyway.

plbossart, 2018-11-20:
And to be clearer, we want to have the HDAudio link by default to enable HDMI but not the HDaudio codec. That's what my latest patches enable. You could have a default sof-defconfig, and then an extension that adds both SOF audio codec support and all required codecs in hdaudio-codecs-defconfig

plbossart, 2018-11-20:
Not sure why it's assigned to me, I am clearly not going to do anything about it. You can deal with the problem by changing the kconfig, or wait until https://github.com/thesofproject/linux/issues/284 is fixed. That latter issue is really the one which @keyonjie should be involved.

mengdonglin, 2018-11-22:
@xiulipan need your help to revise kconfig used by the building server to make daily image work on UP2.

xiulipan, 2018-11-22:
@mengdonglin 
Will do that after we have basic test cases.

mengdonglin, 2018-11-24:
@markyang @DarrenxWang Please help to manually revise the kconfig to build a kernel image without HDA support to  test UP2 in nocodec mode, before the the building server support is ready.

markyang, 2018-11-26:
**Summary:**
  This issue cannot be reproduced on UP2  w/t codec any more.

**Test env:**
sof master: 6e14892
sof tool: ad3f3e4
kernel sof-dev: aff80d74
tplg: sof-apl-nocodec.tplg-ad3f3e4f

markyang, 2018-11-28:
**Summary:**
This issue cannot be reproduced on nocodec UP2 with kernel on Nov 26.
This issue can be reproduced on nocodec UP2 with kernel on Nov 28.

**Test env:**
sof master: 31cde1c
sof tool: ad3f3e4
kernel sof-dev: 4fd5f872
tplg: sof-apl-nocodec.tplg-ad3f3e4f

**Log:**
[config-4.19.0-daily-20181128.txt](https://github.com/thesofproject/sof/files/2623767/config-4.19.0-daily-20181128.txt)
[config-4.19.0-daily-20181126.txt](https://github.com/thesofproject/sof/files/2623773/config-4.19.0-daily-20181126.txt)

mengdonglin, 2018-12-11:
@markyang Could you help to clarify if this kconfig issue is solved ? If yes, please close the issue.

And actually UP2 in nocodec mode failed with #660 now.

mengdonglin, 2018-12-12:
@markyang Please share the commits of the daily image and attach kconfig here. It will be used as an anchor for reference in the future.