thesofproject/sof #608: trace: error trace only show the latest one

xiulipan, 2018-11-22:
error trace only show the latest log

From @singalsu 
```diff
diff --git a/src/tasks/audio.c b/src/tasks/audio.c index 74fbc305..459e2871 100644
--- a/src/tasks/audio.c
+++ b/src/tasks/audio.c
@@ -79,6 +79,11 @@ int do_task_master_core(struct sof *sof)
        /* let host know DSP boot is complete */
        platform_boot_complete(0);

+       trace_event(TRACE_CLASS_PIPE, "pb1");
+       trace_error(TRACE_CLASS_PIPE, "eb1");
+       trace_event(TRACE_CLASS_PIPE, "pb2");
+       trace_error(TRACE_CLASS_PIPE, "eb2");
+
        /* main audio IPC processing loop */
        while (1) {
                /* sleep until next IPC or DMA */
```
```bash
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1         PIPE             74887.552083     74887.554688              audio.c:85  eb2
```

xiulipan, 2018-11-22:
@singalsu 
Reproduced you bug on AP2
Should be fixed by #607

mengdonglin, 2018-11-28:
Closing as fixed.

closed 2018-11-28 by mengdonglin