thesofproject/sof #609: The sof does not work on UP2 with pcm512x.

markyang, 2018-11-22:
**Summary:**
  The sof does not work on UP2 with pcm512x.
  sof-logger is empty. sof-logger -t log is attached below.
**dmesg:**
```
[    4.361126] random: crng init done
[    4.361130] random: 7 urandom warning(s) missed due to ratelimiting
[    4.492483] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    4.492617] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    4.492683] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    4.492800] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    4.492862] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    4.492980] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    4.493028] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    4.493150] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    4.493267] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
[    4.493284] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
[    4.951842] systemd-udevd (1441) used greatest stack depth: 11368 bytes left
[    4.993285] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
[    4.993301] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
[    5.160103] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    5.160251] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    5.160315] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    5.160435] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    5.160808] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    5.160937] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    5.160998] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    5.161121] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    5.579899] IPv6: ADDRCONF(NETDEV_UP): enp2s0: link is not ready
[    5.583319] Generic PHY r8169-200:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=r8169-200:00, irq=IGNORE)
[    5.696134] IPv6: ADDRCONF(NETDEV_UP): enp2s0: link is not ready
[    5.706674] IPv6: ADDRCONF(NETDEV_UP): enp3s0: link is not ready
[    5.708733] Generic PHY r8169-300:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=r8169-300:00, irq=IGNORE)
[    5.820665] IPv6: ADDRCONF(NETDEV_UP): enp3s0: link is not ready
[    5.820728] r8169 0000:03:00.0 enp3s0: Link is Down
[    5.993379] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
[    5.993401] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
[    7.282210] sof-audio sof-audio: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[    7.282471] sof-audio sof-audio: error: reply expected 0x40 got 0xc bytes
[    7.282516] sof-audio sof-audio: error: ipc error for 0x40010000 size 0x40
[    7.282528] sof-audio sof-audio: error: ctx_save ipc error during suspend -22
[    7.282542] sof-audio-pci 0000:00:0e.0: can't suspend (snd_sof_runtime_suspend [snd_sof] returned -22)
[    7.418897] pcm512x i2c-104C5122:00: No SCLK, using BCLK: -2
```

**Test steps:**
1. arecord -l
```
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 0: Port5 (*) []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 1: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 2: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 3: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

**Test env:**
sof master: 7a793433
sof tool: 6da07a91
kernel sof-dev: b846ee56
tplg: sof-apl-pcm512x.tplg-6da07a91

**Log:**
[dmesg-apl.log](https://github.com/thesofproject/sof/files/2606750/dmesg-apl.log)
[logger-apl-t.log](https://github.com/thesofproject/sof/files/2606751/logger-apl-t.log)

xiulipan, 2018-11-22:
@tlauda @jajanusz @lgirdwood 
This bug is a regression caused by https://github.com/thesofproject/sof/pull/603

tlauda, 2018-11-22:
@xiulipan Then we also need to modify kernel. Let's not make any exceptions in terms of ipc replies. Every other ipc uses some kind of reply type message, even if it's the custom one.

xiulipan, 2018-11-22:
@tlauda 
What do you mean by **exceptions** 
@ranj063 
I think the reply here will be used to store some important info at the suspend?

tlauda, 2018-11-22:
@xiulipan I mean, that we shouldn't use the same struct as a reply. It's not done for any other ipc. And I know, that in the future it will be used to store some info, but let's not get ahead of ourselves. Either way, the struct for reply will need to be changed.

xiulipan, 2018-11-22:
@tlauda 
Ok, maybe we have some misuse here. @ranj063 Please give @tlauda some explanation about why the PM need use the same structure as reply.

But I still did not know what is the purpose of https://github.com/thesofproject/sof/pull/603
> Uses generic reply for pm_context ipc messages.
Why do you need the use generic reply for this CTX_SAVE IPC?

And if you need to change some IPC related stuff, please notify us ahead to have some buffer time to make the change work smoothly between Linux and FW.

tlauda, 2018-11-22:
@xiulipan Change in kernel is here: https://github.com/thesofproject/linux/pull/310.

xiulipan, 2018-11-22:
@tlauda 
Thanks, wait for @ranj063 review.

lgirdwood, 2018-11-22:
@xiulipan merged.

xiulipan, 2018-11-23:
close via https://github.com/thesofproject/linux/pull/310

markyang, 2018-11-26:
**Summary:**
  This issue cannot be reproduced on UP2 w/ pcm512x any more.

closed 2018-11-27 by mengdonglin

markyang, 2019-01-07:
@stevyan