thesofproject/sof #619: SRC: possible signal overflow with 8/16/32 to 48 kHz conversions

juimonen, 2018-11-23:
There's possibly similar issue in SRC as #416. So with extreme pumped up/compressed test input I can hear crackling noise in output.

singalsu, 2018-11-23:
Thanks for testing & reporting this! Yep I will check & fix.

singalsu, 2019-02-27:
@juimonen This may be a bug in PCM512x. If set this this gain to -1 dB most annoying rattle disappears:

`amixer set Digital 205`

The default 0 dB can be restored with

`amixer set Digital 207`

I'll test with other device too to confirm. Though since there's a smaller rattle sound left with this tricky music sample I wonder if the SRC coefficients should be re-calculated for e.g. -1 dB attenuation to avoid the filter overshoots to saturate. Now the gain is such that steady state sine waves do not clip (downscaled by filter passband ripple only).

singalsu, 2019-02-27:
@juimonen Another GLK + DA7219 based device does the rattle too though it sounds a lot more like "impulse noise". So better to not close this and continue investigation.

singalsu, 2019-02-28:
I did test code that outputs to trace samples values for a short time after reaching -1 dBFS positive or negative PCM code level. The traced values were parsed and plotted with Octave. The peaks seen at volume component input (=SRC output) and volume output are OK looking signal peaks often flattened to min/max PCM code without opposite polarity peaks. It looks very much like the frequent hit into min/max PCM causes the two DACs that I have access to make some rattle sound. I'll test next adding some small attenuation into SRC filter coefficients.

singalsu, 2019-03-05:
The PR #1069 is now merged so this can be closed.

closed 2019-03-05 by singalsu