thesofproject/sof #620: PR #606 Build simple ROMs for CI testing of APL etc  breaks compilation with xcc

lbetlej, 2018-11-23:
Looks like PR#606 breaks compilation of UP & SMP targets (APL, CNL etc.) with xcc.
See compilation logs of failed Internal Intel CI at https://quickbuild.igk.intel.com/build/3259834

jajanusz, 2018-11-23:
#606 it looks like gcc has some defaults that xcc is missing, idk, something like that makes it build for APL:
https://github.com/jajanusz/sof/commit/d0722090244a39c84998a58e35e7738d54426560
but I'm not sure that it should be like that.

@lgirdwood 
I guess that will be needed only for qemu+gcc environment, so if there is no quickfix for xcc I can add flag like --enable-rom, so you will build it only for your CIs and we won't when we don't need it.

lgirdwood, 2018-11-23:
@jajanusz yes thanks, only needed for qemu CI boot test for each PRs, so not needed for other reasons.

jajanusz, 2018-11-23:
With #621 ROMs are build only when --enable-roms is passed to ./confgiure script

closed 2018-11-23 by jajanusz