thesofproject/sof #632: topology: need virtual widgets for legacy boards

plbossart, 2018-11-27:
We can remove patches for Linux machine drivers by declaring virtual widgets in BYT/CHT topology files:

````text
cht_bsw_rt5672.c
	{"AIF1 Playback", NULL, "ssp2 Tx"},
 	{"ssp2 Tx", NULL, "codec_out0"},
 	{"ssp2 Tx", NULL, "codec_out1"},
 	{"codec_in0", NULL, "ssp2 Rx"},
 	{"codec_in1", NULL, "ssp2 Rx"},
 	{"ssp2 Rx", NULL, "AIF1 Capture"},

cht_bsw_rt5645.c
 	{"AIF1 Playback", NULL, "ssp2 Tx"},
 	{"ssp2 Tx", NULL, "codec_out0"},
 	{"ssp2 Tx", NULL, "codec_out1"},
 	{"codec_in0", NULL, "ssp2 Rx" },
 	{"codec_in1", NULL, "ssp2 Rx" },
 	{"ssp2 Rx", NULL, "AIF1 Capture"},

       {"AIF2 Playback", NULL, "ssp2 Tx"},
 	{"ssp2 Tx", NULL, "codec_out0"},
 	{"ssp2 Tx", NULL, "codec_out1"},
 	{"codec_in0", NULL, "ssp2 Rx" },
 	{"codec_in1", NULL, "ssp2 Rx" },
 	{"ssp2 Rx", NULL, "AIF2 Capture"},

	{"AIF1 Playback", NULL, "ssp0 Tx"},
 	{"ssp0 Tx", NULL, "modem_out"},
 	{"modem_in", NULL, "ssp0 Rx" },
 	{"ssp0 Rx", NULL, "AIF1 Capture"},

        {"AIF2 Playback", NULL, "ssp0 Tx"},
 	{"ssp0 Tx", NULL, "modem_out"},
 	{"modem_in", NULL, "ssp0 Rx" },
 	{"ssp0 Rx", NULL, "AIF2 Capture"},

cht_bsw_max98090_ti.c
	{"HiFi Playback", NULL, "ssp2 Tx"},
 	{"ssp2 Tx", NULL, "codec_out0"},
 	{"ssp2 Tx", NULL, "codec_out1"},
 	{"codec_in0", NULL, "ssp2 Rx" },
 	{"codec_in1", NULL, "ssp2 Rx" },
 	{"ssp2 Rx", NULL, "HiFi Capture"},

bytcht_da7213.c
	{"ssp2 Tx", NULL, "codec_out0"},
 	{"ssp2 Tx", NULL, "codec_out1"},
 	{"Playback", NULL, "ssp2 Tx"},
 	{"ssp2 Rx", NULL, "Capture"},

````

jajanusz, 2019-06-06:
@plbossart Is it done? I guess it need changes in topology files, so someone from SW have to look into it.

jajanusz, 2019-06-10:
@plbossart @ranj063 @kv2019i ping

closed 2020-02-20 by plbossart