thesofproject/sof #642: sof-logger error

keqiaozhang, 2018-11-28:
Today I tried the new sof-logger( in sof repo, under /tools/logger/sof-logger), but run to the following error:
```
chronos@localhost ~ $ sudo sof-logger -l sof-apl.ldc
Error: abi version in sof-apl.ldc file does not coincide with abi version used by logger.
chronos@localhost ~ $ sudo sof-logger -l sof-apl.ldc -t
Error: abi version insof-apl.ldc file does not coincide with abi version used by logger.
```
But sof-logger in soft repo is working well.

Tested with:
kernel-cros-sof-v4.14-rebase: f8f388ef
sof-master: 31cde1
tplg-master: ad3f3e4

cujomalainey, 2018-11-28:
The logger should be exactly the same with the same history as the top of SOFT. I think something is a little strange here.

keyonjie, 2018-11-28:
yes I see the same issue today.

bkokoszx, 2018-11-28:
Hi @keqiaozhang 
Could you try to rebuild rimage, then sof ad try to reproduce issue?

markyang, 2018-11-29:
**Summary:**
  This issue can be reproduced on BYT with ALC5651.
**Test env:**
sof master: 920a3bce
sof tool:   ad3f3e4f
kernel sof-dev: 405e866b
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg-ad3f3e4f

markyang, 2018-11-29:
Because the version change in abi.h, if the following steps were done, it will be OK:
1. git clean -dxf 

2. rebuild rimage:
auotgen.sh 
./configure --enable-rimage
make
sudo make install 

3: rebuild sof-apl.ldc

keqiaozhang, 2018-11-29:
Thanks @markyang  and @bkokoszx 
This issue can be closed now.

closed 2018-11-29 by keqiaozhang