thesofproject/sof #644: The firmware did not work on UP2 w/ pcm512x

markyang, 2018-11-28:
**Summary:**
    The latest firmware did not work on UP2 w/ pcm512x.
    it worked fine if the firmware(443233c9) on Nov. 27 was used.

**dmesg:**
```
[    4.180155] sof-audio sof-audio: tplg: config SSP5 fmt 0x4001 mclk 24576000 bclk 3072000 fclk 48000 width (24)32 slots 2 mclk id 0
[    4.180178] sof-audio sof-audio: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.180291] sof-audio sof-audio: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.180297] sof-audio sof-audio: warn: unexpected DAI config count 0!
[    4.180299] sof-audio sof-audio: tplg: config HDA0 fmt 0x0
[    4.180322] sof-audio sof-audio: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.180394] sof-audio sof-audio: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.180400] sof-audio sof-audio: warn: unexpected DAI config count 0!
[    4.180402] sof-audio sof-audio: tplg: config HDA1 fmt 0x0
[    4.180423] sof-audio sof-audio: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.180506] sof-audio sof-audio: error: ipc error for 0x80010000 size 0xc
[    4.180560] sof-audio sof-audio: error: failed to set DAI config for HDA1
[    4.180597] sof-audio sof-audio: ASoC: physical link loading failed
[    4.180630] sof-audio sof-audio: error: tplg component load failed -5
[    4.180679] sof-audio sof-audio: error: failed to load DSP topology -22
[    4.180717] sof-audio sof-audio: ASoC: failed to probe component -22
[    4.180779] bxt-pcm512x bxt-pcm512x: ASoC: failed to instantiate card -22
[    4.180822] bxt-pcm512x bxt-pcm512x: Topology forgot to remove link Port5?
[    4.180823] bxt-pcm512x bxt-pcm512x: Topology forgot to remove link HDMI1?
[    4.180825] bxt-pcm512x bxt-pcm512x: Topology forgot to remove link HDMI2?
[    4.180826] bxt-pcm512x bxt-pcm512x: Topology forgot to remove link HDMI3?
[    4.180854] bxt-pcm512x bxt-pcm512x: snd_soc_register_card failed -22
[    4.180904] bxt-pcm512x: probe of bxt-pcm512x failed with error -22
```

**sof-logger:**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          DMA            159219.322917    159219.328125            hda-dma.c:448    hda-dmac: 7 no free channel 0
    0      1          DAI            159225.260417         5.937500                dai.c:734    dai_config() error: dma_channel_get() failed
    0      1          IPC            159229.791667         4.531250                ipc.c:361    eCD
```

**Test steps:**
aplay -l
```
aplay: device_list:270: no soundcards found...
```

**Test env:**
sof master: 31cde1ce
sof tool: ad3f3e4f
kernel sof-dev: 4fd5f872
tplg: sof-apl-pcm512x.tplg-ad3f3e4f

**Log:**
[dmesg.log](https://github.com/thesofproject/sof/files/2623194/dmesg.log)
[logger-t.log](https://github.com/thesofproject/sof/files/2623195/logger-t.log)

keqiaozhang, 2018-11-28:
This issue is same as thesofproject/linux/issues/322.

mengdonglin, 2018-11-28:
Closing this since it's a duplicate of thesofproject/linux/issues/322.

closed 2018-11-28 by mengdonglin

stevyan, 2018-11-30:
**Summary:**
This issue cannot be reproduced on UP2(pcm512x) with HDA configuration enabled.

**Test env:**
sof master: d189651
sof tool: d189651
kernel sof-dev: 2a933464
tplg: sof-apl-pcm512x.tplg-d189651d

**Reference:**
test@cd-apl-up2-pcm512x-01:~$ uname -a
Linux **cd-apl-up2-pcm512x-01 4.19.0-daily-hda-20181130** #88 SMP Fri Nov 30 06:59:17 CST 2018 x86_64 x86_64 x86_64 GNU/Linux
test@cd-apl-up2-pcm512x-01:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
test@cd-apl-up2-pcm512x-01:~$