thesofproject/sof #649: pulseaudio did not work on BYT with ALC5651

markyang, 2018-11-29:
**Summary:**
  pulseaudio did not work on BYT with ALC5651. 
  sof-logger is empty. sof-logger -t log is attached below.

**Test steps:**
1. aplay -Dhw:0,0 -c2 -fS16_LE -vv -i ~/Music/18_WAV_48000Hz_stereo_16bit.wav # OK
2. paplay ~/Music/18_WAV_48000Hz_stereo_16bit.wav # no voice

**Test env:**
sof master: 920a3bce
sof tool:   920a3bce
kernel sof-dev: 405e866b
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg-ad3f3e4f

**Log:**
[dmesg.txt](https://github.com/thesofproject/sof/files/2628252/dmesg.txt)
[logger-t.log](https://github.com/thesofproject/sof/files/2628253/logger-t.log)

lgirdwood, 2018-11-29:
@markyang  log is corrupt 
```
[0m    0      2          DMA         3457979531.302083              nan             dw-dma.c:418 	dw-dma: 1 channel 2 start[0m
[0m    0      2          DMA         3457980527.395833       996.093750             dw-dma.c:679 	dw-dma: 1 channel 2 config[0m
```
How are you creating/attaching log ?

Btw, paplay issues are not supported as they use a different PA config to aplay.

lgirdwood, 2018-11-30:
@mengdonglin paplay is not supported, if it plays silce with no DSP or driver errors then it's a PA routing/volume issue and not a SOF bug. If there are kernel IPB bus or logger errors then keep this open otherwise it should be closed.

mengdonglin, 2018-12-03:
Previously Keqiao tested paplay on Minnowboard with ALC5651 codec.  We can check if this is a PA routing issue caused by both legacy-HDA and SOF sound cards being enabled on Minnow and PA routes audio to legacy HD-A (HDMI audio) by default.

@markyang could you help to list and share the audio devices seen by both ALSA and Pulseaudio?
(1) aplay -l
(2) pacmd
    >>list_sinks

mengdonglin, 2018-12-03:
@lgirdwood It seems we need more debugging here. There is only 1 sound card and 1 pcm sink device. The wav is same as the sample spec of this sink.

@markyang Thanks for the info! Could you help to further check if there is any difference on the amixer settings when you're use aplay and paplay?

mengdonglin, 2018-12-04:
@lgirdwood Okay, let's set it to a low priority.

libinyang, 2018-12-05:
Please run `pactl set-sink-volume 0 80%` before paplay.

libinyang, 2018-12-06:
@markyang 
Could you please verify if pactl cmd can fix this issue?

markyang, 2018-12-06:
@libinyang  Neither ```pactl set-sink-volume 0 80%``` nor ```paplay --volume=50000 xxx.wav``` works.

markyang, 2018-12-07:
**Summary:**
  This issue cannot be reproduced on BYT with ALC5651 using the latest build.

**Test env:**
sof master: b5d6c71b
kernel sof-dev: a71221dc
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg

mengdonglin, 2018-12-10:
@markyang @libinyang could you help to clarify which PR fix this issue?

libinyang, 2018-12-10:
I don't need any PR. Just the setting issue.

I'm not sure Mark's situation.

markyang, 2018-12-10:
@mengdonglin @libinyang I retested this issue with the build on Dec 5 and 6, it cannot be reproduced.
it should be a setting issue as libinyang mentioned above. 
I'm just a little puzzled why ```pactl set-sink-volume 0 80%``` didn't work on last Thu.

mengdonglin, 2018-12-11:
@markyang @libinyang   Closing it since it cannot be reproduced now. We can keep an eye and file new bug/reopen it if this issue happens later.

closed 2018-12-11 by mengdonglin