thesofproject/sof #670: "make dist" to release sof tarball

zhuyingjiang, 2018-12-05:
Make tarball release of SOF, using the version number getting from git tag. automatically  generate sof-x.y.tar.gz.

jajanusz, 2018-12-05:
I can add tar packaing that follows kernel convention, so it will be something like `make tar-src-pkg`.
Only issue that it has comparing to autotools' `make dist` is that it has to be done in git repository (I guess it's nonissue at all). Other than that it's easier to develop, cos it's based on git-archive feature and easier to maintain, cos developers don't have to add anything to makefiles so `make dist` could track what to include in tar

zhuyingjiang, 2018-12-07:
@jajanusz Good to know, let's follow kernel convention. And there is a problem with previous "make dist" is that it can't compile using the same compile steps in git after untar the tarball. Please note me when your solution is ready, I will double check and make it work with this.

zhuyingjiang, 2019-02-20:
Close, this is move to #954, and solved in #951

closed 2019-02-20 by zhuyingjiang