thesofproject/sof #671: The playback is no sound on WHL RVP

DarrenxWang, 2018-12-05:
Summary:
The playback is no sound on WHL RVP

dmesg:
no error

Test steps:
aplay -D hw:0,0 -r 48000 -c 2 -f s16_le Music/California_Dreaming_48k_24bit.wav -vv

Test env:
sof master: 12e11b76
kernel sof-dev: d0fbf1c6
tplg: sof-hda-generic.tplg.tplg-12e11b76

Log:
[dmesg.log](https://github.com/thesofproject/sof/files/2648157/dmesg.log)
[logger.txt](https://github.com/thesofproject/sof/files/2648158/logger.txt)

lgirdwood, 2018-12-05:
@DarrenxWang what output are you expecting audio from and what are the alsamixer settings ? ASLA setting can be dumped to text file and attached using "alsactl store -f filename"

mengdonglin, 2018-12-05:
@lgirdwood This error is not caused by amixer setting.  @RanderWang confirmed that this regression is caused by he revised the DAI format from 32 bit to 24 bit, and need further debugging on firmware side. He found both playback and capture by ALC700 is silent now.

mengdonglin, 2018-12-05:
@DarrenxWang However, please help to share the amixer settings. It will be helpful to involve firmware engineers for further debugging.

zhuyingjiang, 2018-12-07:
The result is 16bit and 32bit can play, but with noise, 24bit is silent. Capture can't work. I will debug on this.

mengdonglin, 2018-12-11:
@RanderWang @zhuyingjiang So this issue is blocked or actually a duplicate of #700, right?

jajanusz, 2018-12-11:
@mengdonglin 
I think you can close this issue because it's too broad, only 24b is bugged and #700 address it better.

zhuyingjiang, 2018-12-11:
Yes, this can be closed, since 16/32 playback and capture works.

zhuyingjiang, 2018-12-12:
linux version :https://github.com/thesofproject/linux: 4052ad89a474a84b2b

kconfig : https://github.com/thesofproject/kconfig : 1c7649a29db5b1930cc
make defconfig; 
scripts/kconfig/merge_config.sh .config <PATH>/kconfig/base-defconfig <PATH>/kconfig/sof-defconfig <PATH>/kconfig/hdaudio-codecs-defconfig
make menuconfig  un-select HDA audio PCI

sof version: https://github.com/thesofproject/sof : 12e11b7693f04c1658b7e
build sof-cnl.ri

tplg: change tools/topology/sof-hda-generic.m4 s24le to s16le when build 16bit tplg, change to s32le when build 32bit tplg
Result: 16 bit /32 bit playback capture and HDMI DP are all OK. 24bit remain silent

closed 2018-12-12 by zhuyingjiang