thesofproject/sof #675: Failed to set sof-audio hw params after the device was resumed from S3 on BYT with ALC5651.

markyang, 2018-12-06:
**Summary:** 
  Failed to set sof-audio hw params after the device was resumed from S3 on BYT with ALC5651.
  Both 'sof-logger' and 'sof-logger -t' are empty.

**dmesg:**
```
[  932.541122] sof-audio sof-audio: pcm: open stream 0 dir 0
[  932.541130] sof-audio sof-audio: period min 192 max 16384 bytes
[  932.541134] sof-audio sof-audio: period count 2 max 16
[  932.541137] sof-audio sof-audio: buffer max 65536 bytes
[  932.599609] sof-audio sof-audio: rate_min: 48000 rate_max: 48000
[  932.599620] sof-audio sof-audio: channels_min: 2 channels_max: 2
[  932.599633] sof-audio sof-audio: rate_min: 48000 rate_max: 48000
[  932.599641] sof-audio sof-audio: channels_min: 2 channels_max: 2
[  932.599658] sof-audio sof-audio: rate_min: 48000 rate_max: 48000
[  932.599665] sof-audio sof-audio: channels_min: 2 channels_max: 2
[  932.599678] sof-audio sof-audio: pcm: hw params stream 0 dir 0
[  932.599690] sof-audio sof-audio: generating page table for 000000009ffeee5a size 0xff00 pages 16
[  932.599699] sof-audio sof-audio: stream_tag 0
[  932.907635] sof-audio sof-audio: error: ipc timed out for 0x60010000 size 0x48
[  932.907886] sof-audio sof-audio: error: unexpected fault 0xffffffff trace 0xffffffff
[  932.907902] sof-audio sof-audio: error: waking up any trace sleepers
[  932.908043] sof-audio sof-audio: ASoC: sof-audio hw params failed: -110
[  932.908085]  Passthrough: ASoC: hw_params FE failed -110
[  932.908429] sof-audio sof-audio: pcm: free stream 0 dir 0
[  933.209242] sof-audio sof-audio: error: ipc timed out for 0x60030000 size 0xc
[  933.209493] sof-audio sof-audio: error: unexpected fault 0xffffffff trace 0xffffffff
[  933.209510] sof-audio sof-audio: error: waking up any trace sleepers
[  933.211235] sof-audio sof-audio: pcm: close stream 0 dir 0

```

**Test steps:**
Disable pulseaudio first(rm /usr/bin/pulseaudio), and execute the following steps at once after reboot:
1. aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofbytcrrt5651 [sof-bytcr-rt5651], device 0: Passthrough (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
2. ps aux | grep pulseaudio | grep -v grep  # make sure pulseaudio has been closed.
3. dmesg > dmesg-before-s3.txt
4. run 'echo mem > /sys/power/state' to suspend the device and press any key to resume. 
5. dmesg > dmesg-after-s3.txt
6. aplay -Dhw:0,0 -c2 -fS16_LE -vv -i ~/Music/18_WAV_48000Hz_stereo_16bit.wav
```
Playing WAVE '/home/byt-jx/Music/18_WAV_48000Hz_stereo_16bit.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 84000
PERIOD_SIZE: 4032
PERIOD_BYTES: 16128
PERIODS: (4 5)
BUFFER_TIME: 340000
BUFFER_SIZE: 16320
BUFFER_BYTES: 65280
TICK_TIME: 0
```
7. dmesg > dmesg-after-aplay.txt

**Test env:**
sof master: ab52fdc3
kernel sof-dev: 62e8db9d
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg

**Log:**
[dmesg-after-s3.txt](https://github.com/thesofproject/sof/files/2652341/dmesg-after-s3.txt)
[dmesg-after-aplay.txt](https://github.com/thesofproject/sof/files/2652342/dmesg-after-aplay.txt)
[dmesg-before-s3.txt](https://github.com/thesofproject/sof/files/2652343/dmesg-before-s3.txt)

lgirdwood, 2018-12-06:
@markyang Minnowboard does not support S3, can you try this on another BYT/CHT/BSW based board like Asus T100, or RVP.

markyang, 2018-12-07:
@lgirdwood mengdonglin let me check if this is only a sof-audio issue or not. if it's just a system issue, we'll close it at once. I'll attach more info later.

markyang, 2018-12-11:
**Summary:**
  No sound module was loaded, do suspend/resume, then sof module cannot be loaded.
  But if the suspend/resume(the step 4 below) is skipped, it can be done successfully.
  so it seems to be a system issue.

**dmesg:**
```
[  641.733047] usb 1-2: reset low-speed USB device number 3 using xhci_hcd
[  642.012631] acpi LNXPOWER:05: Turning OFF
[  642.013092] acpi LNXPOWER:02: Turning OFF
[  642.013194] acpi LNXPOWER:01: Turning OFF
[  642.013398] acpi LNXPOWER:00: Turning OFF
[  642.013556] OOM killer enabled.
[  642.013561] Restarting tasks ... done.
[  642.016503] video LNXVIDEO:00: Restoring backlight state
[  642.016520] PM: suspend exit
[  644.209700] igb 0000:03:00.0 enp3s0: igb: enp3s0 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX/TX
[  664.721422] sof-audio-acpi 80860F28:01: ACPI DSP detected
[  664.721431] sof-audio-acpi 80860F28:01: BYT-CR not detected
[  664.721608] sof-audio sof-audio: probing SOF DSP device....
[  664.721612] sof-audio sof-audio: using platform alias sst-mfld-platform
[  664.721617] sof-audio sof-audio: LPE PHY base at 0x90a00000 size 0x200000
[  664.721668] sof-audio sof-audio: LPE VADDR 000000005b054519
[  664.721671] sof-audio sof-audio: IMR base at 0x20000000 size 0x100000
[  664.721690] sof-audio sof-audio: IMR VADDR 00000000d8251699
[  664.721732] sof-audio sof-audio: using IRQ 29
[  664.728739] sof-audio sof-audio: pre-allocate 8 IPC messages
[  664.728782] sof-audio sof-audio: loading firmware
[  664.728788] sof-audio sof-audio: header size=0x143a8 modules=0x1 abi=0x1 size=16
[  664.728832] sof-audio sof-audio: new module size 0x1439c blocks 0x10 type 0x0
[  664.728836] sof-audio sof-audio: block 0 type 0x1 size 0xf0 ==>  offset 0xc0000
[  664.728996] sof-audio sof-audio: block 1 type 0x1 size 0x16a ==>  offset 0xc0400
[  664.729352] sof-audio sof-audio: block 2 type 0x1 size 0x6 ==>  offset 0xc057c
[  664.729362] sof-audio sof-audio: block 3 type 0x1 size 0x6 ==>  offset 0xc059c
[  664.729373] sof-audio sof-audio: block 4 type 0x1 size 0x6 ==>  offset 0xc05bc
[  664.729384] sof-audio sof-audio: block 5 type 0x1 size 0x6 ==>  offset 0xc05dc
[  664.729394] sof-audio sof-audio: block 6 type 0x1 size 0x6 ==>  offset 0xc05fc
[  664.729405] sof-audio sof-audio: block 7 type 0x1 size 0x3 ==>  offset 0xc061c
[  664.729412] sof-audio sof-audio: block 8 type 0x1 size 0x6 ==>  offset 0xc063c
[  664.729423] sof-audio sof-audio: block 9 type 0x1 size 0x4 ==>  offset 0xc0658
[  664.729426] sof-audio sof-audio: block 10 type 0x1 size 0x17 ==>  offset 0xc065c
[  664.729450] sof-audio sof-audio: block 11 type 0x1 size 0x6 ==>  offset 0xc067c
[  664.729461] sof-audio sof-audio: block 12 type 0x1 size 0x1158a ==>  offset 0xc06a0
[  664.789714] sof-audio sof-audio: block 13 type 0x2 size 0x1dd0 ==>  offset 0x100008
[  664.796144] sof-audio sof-audio: block 14 type 0x2 size 0xc78 ==>  offset 0x101dd8
[  664.798850] sof-audio sof-audio: block 15 type 0x2 size 0x68 ==>  offset 0x103238
[  664.799002] sof-audio sof-audio: booting DSP firmware
[  665.841119] sof-audio sof-audio: error:  unable to run DSP firmware
[  665.841612] sof-audio sof-audio: error: unexpected fault 0xffffffff trace 0xffffffff
[  665.841628] sof-audio sof-audio: error: failed to reset DSP
[  665.841644] sof-audio sof-audio: error: failed to boot DSP firmware -19
```

**Test steps:**
1. add all the snd* and sof* modules in the blacklist.
2. reboot device.
3. lsmod | egrep "snd|sof" # make sure no sound module was loaded.
4. run 'echo mem > /sys/power/state' to suspend the device and press any key to resume.
5. sof_insert.sh
6. aplay -l
aplay: device_list:270: no soundcards found...

**Test env:**
sof master: b5d6c71b #Dec 7
kernel sof-dev: 8cf9a6a7
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg

mengdonglin, 2018-12-11:
@lgirdwood Could you share more info about "Minnowboard does not support S3". Is it an BIOS/ACPI issue or board issue for Minnowboard? If yes, I'll closing this issue as "won't fix".

It seems the audio DSP firmware cannot boot after S3, even if the firmware can still be loaded.

We don't have working BYT/CHT/BSW based board like Asus T100, or RVP now.  But since other BYT/CHT can support S3, so this is at least not a silicon issue of BYT but a Minnnow-specific issue.

lgirdwood, 2018-12-19:
@mengdonglin The minnowboard BIOS does not correctly re-enable the ADSP PCI device - any PCI/BAR IO always will read 0xffffffff. Probably not a big deal as Minnow will always be plugged into a PSU, although we will need a flag in the driver platform data to prevent RTD3 on Minnowboard.

tlauda, 2019-03-23:
@lgirdwood @mengdonglin Should we close this one, as it's not FW issue?

closed 2019-03-24 by lgirdwood