thesofproject/sof #681: debug: fault injection: sign firmware with invalid key

plbossart, 2018-12-06:
To make sure the Linux driver error flows get tested in a regular basis, add an invalid key to check how the driver probe releases all allocated resources.

lbetlej, 2019-03-27:
driver may respond with a meaningful dmesg entry on a specific ROM error code documented in sof-docs

lgirdwood, 2019-03-27:
Test key should be generated and added to git alongside public dev key. rimage already takes the key location as an argument so it would be easy enough to add a make target to sign with the test key.

jajanusz, 2019-06-06:
It will just add unnecessary burden to the codebase. It's something that just CI/occasional checks should do, some check with random key and try to load it. For sure it shouldn't be placed next to valid keys.
The most we can add to the repo for that is some script with its invalid key.

@mengdonglin Probably @xiulipan is the best person to be assigned for this one.

xiulipan, 2019-06-10:
@jajanusz @plbossart @lgirdwood 
What would this key used for?
Do you mean add some fail case test in CI?
I will take that as a new feature request.

lgirdwood, 2019-06-10:
@xiulipan this is to test driver handling paths for FW boot/signing failure. Maybe best option is to have rimage cmd line option that signs with random data. This way we don't have to carry "testing keys" and rimage can still sign using random data (as the private key).

jajanusz, 2019-06-10:
@lgirdwood I prefer to clutter some deep folder that no1 looks into instead of rimage codebase. 
Also CI should be deterministic, I know it's near to impossible to match valid key, but still, it makes CI non-deterministic.

lgirdwood, 2019-06-10:
@jajanusz I'm also fine with signing with an empty or 0 key (as long as driver error flow is tested)

lgirdwood, 2020-08-18:
@plbossart did you manage to have a solution for testing kernel boot flow for signing ?

kv2019i, 2023-02-15:
@mengdonglin @plbossart I think this should be moved to validation backlog, assigning to TBD milestone until then.