thesofproject/sof #682: topology: add bad configurations to test IPC errors

plbossart, 2018-12-06:
add e.g. unsupported MCLKs rates, invalid SSPs or obvious mistakes to check how Linux deals with topology errors. Those bad configurations shall be used only to test that that the driver releases all allocated resources and the modules can be reloaded once a valid topology file is provided.

lbetlej, 2019-03-27:
to be covered by IPC fuzzer

jajanusz, 2019-06-08:
@plbossart @lgirdwood @lbetlej Do we need this? Fuzzer should test it without need for keeping bad topologies in repo. Probably we should close it and stick to #1279

lgirdwood, 2019-06-10:
@jajanusz yep, I think @ranj063 should be able to confirm that fuzzer can take existing valid topollogies and mangle them to check Linux driver error paths.

plbossart, 2019-06-10:
@jajanusz This is a different level of validation. A bad config can be applied and tested on a regular basis by QA.  It's like module load/unload, it should be part of the basis acceptance tests. 
Fuzzing has elements of randomness to it that will prevent it from being used on a regular basis, and the interpretation of the results could take time.
Different tools really.

jajanusz, 2019-06-10:
@plbossart You are right, for regular (CI) tests you need something else than fuzzer. I just didn't know what you want to achieve, but now it's clear.

cujomalainey, 2021-09-03:
As much as I agree with the consistency aspect of targeted tests, we do have a fuzzer running continuously against the FW. I could talk to my security team and see if we can get one for the kernel as well.