thesofproject/sof #686: 4ch 16-bit DMIC recording for v0.55 has dc at the beginning of recording

ranj063, 2018-12-07:
Please find the recorded sample here: https://drive.google.com/file/d/1Oz6uWxcdCedkXCkLsj-Z6iCWnx1zL8xE/view?usp=sharing

singalsu, 2018-12-10:
Here's a plot of the captured waveform

[image]

The channels 1 and 2 look normal to me. Channels 3 and 4 with apparently no microphones contain a DC value of about -0.48 once the DCCOMP in HW and gain ramp controlled by FW is settled.

The tonal sound heard in the beginning is from the DCCOMP vs. gain ramp update caused "sawtooth" waveform. Since the non-connected microphones need to be removed by the audio framework I don't consider it as issue. The DCCOMP can compensate only about half of full scale DC level and the remaining DC level is what's seen after 300 ms time.
 
[image]

If the channels 3-4 are omitted, the start of channels 1-2 looks normal to me. I don't see any excessive DC level there. Can you please confirm if the bug is about channels 3-4.

[image]

What's strange here is that channels 1 and 2 seem to be identical except one LSB difference in the very beginning. The next plot is ch1 minus ch2.

[image]

Unless your device has only one microphone it's a severe bug somewhere. Can you please confirm the number of microphones.

ranj063, 2018-12-11:
@singalsu My recording was on a board with only one mic.

But according to Naveen the issue still persists when he checked with a board that has 2 mic's. He will send out an email to you

singalsu, 2018-12-11:
@ranj063 The issue doesn't look like a bug in SOF. Maybe we can close this?

naveen-manohar, 2018-12-12:
Can you please recheck with the captured file from UI ?

[myRecording02.zip](https://github.com/thesofproject/sof/files/2670917/myRecording02.zip)

singalsu, 2018-12-13:
It looks and sounds quite OK but there's some strange tonal background noise. Is there a fan in the device or is the lab space noisy? Are the click sounds from keyboard typing?

There's no DC problems seen, so the this topic can be closed. But if there's no fan and the space where you tested was silent you may want to open another issue about noise. 

Generally microphones should be tested in silent condition and it would be good if you could in addition playback a chirp / sine sweep from about 100 Hz to 20 kHz via some high quality speaker (maybe a good over-ear headphone very near to microphone can be used too for output) . Audacity can make a chirp waveform via menu Generate -> Chirp.

singalsu, 2018-12-17:
@naveen-manohar Can we close this issue?

naveen-manohar, 2018-12-17:
Sure Seppo. Thanks for detailed explanation and follow-up.
We can close this for now.

Regarding BG noise - You are right, there was a fan in the lab.
Thanks fro the tip on Chirp wave generation.

wenqingfu, 2018-12-17:
> Regarding BG noise - You are right, there was a fan in the lab.

@singalsu  :sweat_smile:

singalsu, 2018-12-18:
Excellent, I'll close this. @wenqingfu Yep, exactly!

closed 2018-12-18 by singalsu