thesofproject/sof #694: DSP panic occurred after the firmware boot on BYT with ALC5651

markyang, 2018-12-10:
**Summary:**
  DSP panic occurred after the firmware boot on BYT with ALC5651. 
  This issue cannot be reproduced with the same kernel & topology, and the firmware on Dec. 7, so it seems to be a firmware regression.
  sof-logger failed: ```Unable to open in file /sys/kernel/debug/sof/etrace```  
  sof-logger -t is empty
**dmesg:**
```
[   16.728153] sof-audio sof-audio: booting DSP firmware
[   16.730066] sof-audio sof-audio: ipc rx: 0x70000000: FW_READY
[   16.730072] sof-audio sof-audio: ipc: DSP is ready 0x70000000 offset 0x144000
[   16.730152] sof-audio sof-audio:  Firmware info: version 1:1-8ef28 build 32 on Dec  9 2018:21:49:45
[   16.730169] sof-audio sof-audio: error: have no window info
[   16.730248] sof-audio sof-audio: booting DSP firmware completed
[   16.730257] sof-audio sof-audio: ipc rx done: 0x70000000: FW_READY
[   16.730385] sof-audio sof-audio: firmware boot complete
[   16.730622] sof-audio sof-audio: generating page table for 00000000c6b22433 size 0x10000 pages 16
[   16.730626] sof-audio sof-audio: dma_trace_pages: 16
[   16.730651] sof-audio sof-audio: stream_tag: -26374
[   16.730674] sof-audio sof-audio: ipc tx: 0x90010000: GLB_TRACE_MSG
[   16.730913] sof-audio sof-audio: error : DSP panic!
[   16.731159] sof-audio sof-audio: error: runtime exception
[   16.731218] sof-audio sof-audio: error: trace point 90010000
[   16.731279] sof-audio sof-audio: error: panic happen at :0
[   16.731339] sof-audio sof-audio: error: DSP Firmware Oops
[   16.731398] sof-audio sof-audio: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   16.734028] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0xff345000 PS       0x00070520 SAR     0x0000001d
[   16.736717] sof-audio sof-audio: EPC1     0xff2c0983 EPC2     0x00000000 EPC3     0xff2c0f03 EPC4    0xff2c0f03
[   16.739523] sof-audio sof-audio: EPC5     0xff2c0f03 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   16.742353] sof-audio sof-audio: EPS2     0x00000000 EPS3     0x00060720 EPS4     0x00060720 EPS5    0x00060720
[   16.745151] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00078f40 INTERRU 0x000000a2
[   16.747951] sof-audio sof-audio: stack dump from 0xff327c30
[   16.750750] sof-audio sof-audio: 0xff327c30: 0xff2d1b24 0x00060730 0x0000001d 0x484c0c59
[   16.753589] sof-audio sof-audio: 0xff327c34: 0xff306a94 0xff344ff8 0xfffffff8 0xff307684
[   16.756474] sof-audio sof-audio: 0xff327c38: 0x0000000d 0x0fffff3f 0xff2d1b1e 0xff2d1b34
[   16.759311] sof-audio sof-audio: 0xff327c3c: 0x00000000 0x00000000 0x7ffe8aff 0xff5df7fc
[   16.759315] sof-audio sof-audio: 0xff327c40: 0x24052016 0x5c44c218 0x0085c500 0x140a9941
[   16.759318] sof-audio sof-audio: 0xff327c44: 0x157f73fd 0x6de6fbdf 0xbf5fbf57 0x5b293dfb
[   16.759321] sof-audio sof-audio: 0xff327c48: 0x44a00608 0x0b400291 0x422106a0 0x090c0232
[   16.759324] sof-audio sof-audio: 0xff327c4c: 0x2f6fff0b 0xffff9fdf 0xe5e33fff 0xef63ffd6
[   17.032879] sof-audio sof-audio: error: ipc timed out for 0x90010000 size 0x1c
[   17.037581] sof-audio sof-audio: error: runtime exception
[   17.040491] sof-audio sof-audio: error: trace point 90010000
[   17.043367] sof-audio sof-audio: error: panic happen at :0
[   17.046185] sof-audio sof-audio: error: DSP Firmware Oops
[   17.048947] sof-audio sof-audio: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   17.051764] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0xff345000 PS       0x00070520 SAR     0x0000001d
[   17.054597] sof-audio sof-audio: EPC1     0xff2c0983 EPC2     0x00000000 EPC3     0xff2c0f03 EPC4    0xff2c0f03
[   17.057435] sof-audio sof-audio: EPC5     0xff2c0f03 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   17.060259] sof-audio sof-audio: EPS2     0x00000000 EPS3     0x00060720 EPS4     0x00060720 EPS5    0x00060720
[   17.063178] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00078f40 INTERRU 0x000000a2
[   17.066014] sof-audio sof-audio: stack dump from 0xff327c30
[   17.068861] sof-audio sof-audio: 0xff327c30: 0xff2d1b24 0x00060730 0x0000001d 0x484c0c59
[   17.071720] sof-audio sof-audio: 0xff327c34: 0xff306a94 0xff344ff8 0xfffffff8 0xff307684
[   17.074673] sof-audio sof-audio: 0xff327c38: 0x0000000d 0x0fffff3f 0xff2d1b1e 0xff2d1b34
[   17.077542] sof-audio sof-audio: 0xff327c3c: 0x00000000 0x00000000 0x7ffe8aff 0xff5df7fc
[   17.080411] sof-audio sof-audio: 0xff327c40: 0x24052016 0x5c44c218 0x0085c500 0x140a9941
[   17.083381] sof-audio sof-audio: 0xff327c44: 0x157f73fd 0x6de6fbdf 0xbf5fbf57 0x5b293dfb
[   17.086280] sof-audio sof-audio: 0xff327c48: 0x44a00608 0x0b400291 0x422106a0 0x090c0232
[   17.089174] sof-audio sof-audio: 0xff327c4c: 0x2f6fff0b 0xffff9fdf 0xe5e33fff 0xef63ffd6
[   17.092154] sof-audio sof-audio: error: can't set params for DMA for trace -110
[   17.095114] sof-audio sof-audio: warning: failed to initialize trace -110
[   17.095206] sof-audio sof-audio: created machine bytcr_rt5651
```

**Test steps:**
1. $ aplay -l
```
aplay: device_list:270: no soundcards found...
```

**Test env:**
sof master: 8ef2894a
kernel sof-dev: 8cf9a6a7
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg

**Log:**
[dmesg-byt.txt](https://github.com/thesofproject/sof/files/2661903/dmesg-byt.txt)

markyang, 2018-12-10:
This issue can be reproduced if sof-acce7428 was used,  but sof-be45dacd worked fine.
So it seems that sof-acce7428 caused this firmware regression.

mengdonglin, 2018-12-11:
Duplicate of #695 .  The error of DSP boot fail is same "error: have no window info"

markyang, 2018-12-17:
**Summary:**
The issue cannot be reproduced any more.

**Test env:**
sof master: 0dad0130
kernel sof-dev: 2d3fcf16
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg

closed 2018-12-17 by markyang