thesofproject/sof #695: Can't set params for DMA on APL with PCM512x

markyang, 2018-12-10:
**Summary:**
  Can't set params for DMA on APL with PCM512x. 
  This issue cannot be reproduced with the same kernel & topology, and the firmware on Dec. 7, so it seems to be a firmware regression.
  sof-logger failed: ```Unable to open in file /sys/kernel/debug/sof/etrace```  
  sof-logger -t is empty
**dmesg:**
```

[    4.064974] sof-audio sof-audio: booting DSP firmware
[    4.065069] sof-audio sof-audio: period_bytes:0x0
[    4.065071] sof-audio sof-audio: periods:1
[    4.086876] sof-audio sof-audio: unstall/run core: core_mask = 1
[    4.086881] sof-audio sof-audio: DSP core(s) enabled? 1 : core_mask 1
[    4.127936] sof-audio sof-audio: ipc rx: 0x70000000: FW_READY
[    4.127939] sof-audio sof-audio: ipc: DSP is ready 0x70000000 offset 0x81000
[    4.127996] sof-audio sof-audio:  Firmware info: version 1.1-8ef28 build 32 on Dec  9 2018:21:52:09
[    4.128007] sof-audio sof-audio: error: have no window info
[    4.128008] sof-audio sof-audio: booting DSP firmware completed
[    4.128010] sof-audio sof-audio: ipc rx done: 0x70000000: FW_READY
[    4.131881] sof-audio sof-audio: Firmware download successful, booting...
[    4.131884] sof-audio sof-audio: firmware boot complete
[    4.131891] sof-audio sof-audio: Debug PCIR: 00000000 at  00000048
[    4.131894] sof-audio sof-audio: Debug PCIW: 00000002 at  00000048
[    4.131898] sof-audio sof-audio: Debug PCIR: 00000000 at  00000044
[    4.131901] sof-audio sof-audio: Debug PCIW: 00000000 at  00000044
[    4.131950] sof-audio sof-audio: generating page table for 000000006c525691 size 0x10000 pages 16
[    4.131952] sof-audio sof-audio: dma_trace_pages: 16
[    4.131984] sof-audio sof-audio: period_bytes:0x0
[    4.131985] sof-audio sof-audio: periods:1
[    4.131991] sof-audio sof-audio: stream_tag: 1
[    4.132007] sof-audio sof-audio: ipc tx: 0x90010000: GLB_TRACE_MSG
[    4.403586] random: crng init done
[    4.403590] random: 7 urandom warning(s) missed due to ratelimiting
[    4.439898] sof-audio sof-audio: error: ipc timed out for 0x90010000 size 0x1c
[    4.439957] sof-audio sof-audio: unknown ROM status value ffffffff
[    4.440072] sof-audio sof-audio: error: unexpected fault 0xffffffff trace 0xffffffff
[    4.440118] sof-audio sof-audio: error: can't set params for DMA for trace -110
[    4.440169] sof-audio sof-audio: warning: failed to initialize trace -110
[    4.440253] sof-audio sof-audio: created machine bxt-pcm512x
[    4.467487] HDMI HDA Codec ehdaudio0D2: Max dais supported: 3
[    4.474193] bxt-pcm512x bxt-pcm512x: info: override FE DAI link SSP5-Codec
[    4.474196] bxt-pcm512x bxt-pcm512x: info: override FE DAI link iDisp1
[    4.474198] bxt-pcm512x bxt-pcm512x: info: override FE DAI link iDisp2
[    4.474199] bxt-pcm512x bxt-pcm512x: info: override FE DAI link iDisp3
[    4.474264] sof-audio sof-audio: loading topology:intel/sof-apl-pcm512x.tplg
[    4.492076] sof-audio sof-audio: tplg: ready widget id 0 pipe 1 type 23 name : PCM0P stream Passthrough Playback 0
[    4.492081] sof-audio sof-audio: loaded host PCM0P
[    4.492083] sof-audio sof-audio:  config: periods snk 2 src 0 fmt 0
[    4.492105] sof-audio sof-audio: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    4.799878] sof-audio sof-audio: error: ipc timed out for 0x30010000 size 0x44
[    4.799938] sof-audio sof-audio: unknown ROM status value ffffffff
[    4.800054] sof-audio sof-audio: error: unexpected fault 0xffffffff trace 0xffffffff
[    4.800102] sof-audio sof-audio: error: DSP failed to add widget id 0 type 23 name : PCM0P stream Passthrough Playback 0 reply 0
[    4.800166] sof-audio sof-audio: ASoC: failed to load widget PCM0P
[    4.800200] sof-audio sof-audio: error: tplg component load failed -110
[    4.800246] sof-audio sof-audio: error: failed to load DSP topology -22
[    4.800281] sof-audio sof-audio: ASoC: failed to probe component -22
[    4.800330] bxt-pcm512x bxt-pcm512x: ASoC: failed to instantiate card -22
[    4.800396] bxt-pcm512x bxt-pcm512x: snd_soc_register_card failed -22
[    4.800442] bxt-pcm512x: probe of bxt-pcm512x failed with error -22

```

**Test steps:**
1. $ aplay -l
aplay: device_list:270: no soundcards found...

**Test env:**
sof master: 8ef2894a
kernel sof-dev: 8cf9a6a7
tplg: sof-apl-pcm512x.tplg

**Log:**
[dmesg-apl.txt](https://github.com/thesofproject/sof/files/2661981/dmesg-apl.txt)

xiulipan, 2018-12-11:
@markyang 
From the log here
`   [ 4.128007] sof-audio sof-audio: error: have no window info`
The FW boot is failed due to no mem window info. Changed the issue title.

Root caused to be https://github.com/thesofproject/sof/pull/657 changed IPC ready structure size make mismatch between kernel and fw.

keqiaozhang, 2018-12-11:
This issue also can be reproduced on GLK.
cros-sof-v4.14-rebase:676c0362
sof-master:8ef289

Jocelyn-Li, 2018-12-11:
@mrajwa some misunderstanding here :( the issue I talked in the meeting is https://github.com/thesofproject/linux/issues/370.

mengdonglin, 2018-12-12:
@mrajwa,

Your commit https://github.com/thesofproject/sof/commit/acce742819e55f57086058a7dc6949b3a93a4076
introduced an ABI chnage. It's recommended to use 1 of the reserved 4 uint32_t for the new flag "gdb_enabled".

uint64_t locks_verbose:1;
	} debug;		} debug;

+ 	uint32_t gdb_enabled;
+ 
 	/* reserved for future use */		/* reserved for future use */
	uint32_t reserved[4];		uint32_t reserved[4];
} __attribute__((packed));	} __attribute__((packed));

mengdonglin, 2018-12-14:
@markyang  SOF has merged the fix https://github.com/thesofproject/sof/commit/f28d3fbe227deaaf75c53ac09da959eace5e9cce

Could you help to verify this solution for this issue  and its duplicate #694?

keqiaozhang, 2018-12-17:
This issue is gone on GLK, logger trace works well on GLK.

markyang, 2018-12-17:
**Summary:**
The issue cannot be reproduced on APL with PCM512x.

**Test env:**
sof master: 0dad0130
kernel sof-dev: 2d3fcf16
tplg: sof-apl-pcm512x.tplg

closed 2018-12-17 by markyang