thesofproject/sof #710: Remove Xrun recovery inside FW.

keyonjie, 2018-12-13:
SOF FW will need to remove recovery trying code, e.g. pipeline_prepare(), trigger start, and schedule copy.

Just keep sending IPC to host driver side, host framework(e.g. ALSA) will help to handle this recovery if possible, e.g. trigger stop and start.

lgirdwood, 2018-12-13:
ok, no objections from me but we need to make sure drivers *does* do the recovery.

keyonjie, 2018-12-14:
@lgirdwood sure, @bardliao is working on that.

bardliao, 2018-12-14:
@keyonjie @lgirdwood My plan is to return SNDRV_PCM_POS_XRUN in sof_pcm_pointer when kernel receive the XRUN message from FW.

slawblauciak, 2019-02-18:
FW xrun recovery will be disabled once driver fully handles the situation
https://github.com/thesofproject/linux/issues/660

jajanusz, 2019-06-06:
WIP #1439

tlauda, 2019-10-25:
It's disabled in FW.

closed 2019-10-25 by tlauda