thesofproject/sof #717: Regression: volume change doesn't work

keqiaozhang, 2018-12-14:
First, the default value of all right channels are zero. see the following screenshot.

[image]

Second, the volume cannot be changed.
trace error:
```
    0      2          IPC           4567233.489583       210.260422            apl-ipc.c:114    ipc: msg rx -> 0x50010000
    0      2          IPC           4567239.635417         6.145833            handler.c:794    ipc: hdr 0x50010000 tx (108) > rx (92)
    0      2          IPC           4567243.385417         3.750000            handler.c:796    ipc: comp 9 -> cmd 0
    0      2       VOLUME           4567249.947917         6.562500             volume.c:432    volume_cmd()
    0      2       VOLUME           4567254.010417         4.062500             volume.c:326    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, cdata->comp_id = 9
    0      2       VOLUME           4567257.864583         3.854167             volume.c:331    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 536884292, value = 2047
    0      1       VOLUME           4567261.145833         3.281250             volume.c:339    volume_ctrl_set_cmd() error: SOF_CTRL_CMD_VOLUME, invalid i = 536884292
    0      2       VOLUME           4567266.406250         5.260417             volume.c:331    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 1, value = 0
    0      2          IPC           4602111.250000     34844.843750            apl-ipc.c:114    ipc: msg rx -> 0x50020000
    0      2          IPC           4602117.395833         6.145833            handler.c:794    ipc: hdr 0x50020000 tx (108) > rx (92)
    0      2          IPC           4602121.145833         3.750000            handler.c:796    ipc: comp 9 -> cmd 0
    0      2       VOLUME           4602127.552083         6.406250             volume.c:432    volume_cmd()
    0      2       VOLUME           4602131.614583         4.062500             volume.c:403    volume_ctrl_get_cmd(), SOF_CTRL_CMD_VOLUME / SOF_CTRL_CMD_SWITCH, cdata->comp_id = 9
    0      2       VOLUME           4602135.156250         3.541667             volume.c:410    volume_ctrl_set_cmd(), channel = 0, value = 65536
    0      2       VOLUME           4602138.489583         3.333333             volume.c:410    volume_ctrl_set_cmd(), channel = 1, value = 0
    0      2          IPC           4602306.145833       167.656250            apl-ipc.c:114    ipc: msg rx -> 0x50010000
    0      2          IPC           4602312.031250         5.885417            handler.c:794    ipc: hdr 0x50010000 tx (108) > rx (92)
    0      2          IPC           4602315.885417         3.854167            handler.c:796    ipc: comp 9 -> cmd 0
    0      2       VOLUME           4602322.395833         6.510417             volume.c:432    volume_cmd()
    0      2       VOLUME           4602326.458333         4.062500             volume.c:326    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, cdata->comp_id = 9
    0      2       VOLUME           4602330.260417         3.802083             volume.c:331    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 536884292, value = 2047
    0      1       VOLUME           4602333.697917         3.437500             volume.c:339    volume_ctrl_set_cmd() error: SOF_CTRL_CMD_VOLUME, invalid i = 536884292
```
How to reproduce:
1. alsamixer -c 0
2. change the PGA* items.

Tested with:
1. FW: SOF master/78f13ed + PR #709
2. Kernel: topic/cros-sofv4.14-rebase/40fae8a
3. topology: SOF master/78f13ed

[dmesg.log](https://github.com/thesofproject/sof/files/2679349/dmesg.log)
[logger-t.log](https://github.com/thesofproject/sof/files/2679350/logger-t.log)

keqiaozhang, 2018-12-17:
@dabekjakub 
This issue has basically been fixed, but 2 display issues need to be addressed:
1. The default value of all right channels are still 0.
2. Volume display error. Volume can be set and effective, but display is wrong as default in alsamixer like below. I think there may be some mismatched parts between driver and firmware in ABI.
Besides, the errors are still there in trace:
    0      1       VOLUME            105545.885417         3.593750             volume.c:339    volume_ctrl_set_cmd() error: SOF_CTRL_CMD_VOLUME, invalid i = 536884268

[image]

Tested with：
sof-glk-005-drop-stable:75410581

wenqingfu, 2018-12-17:
> This issue has basically been fixed, but 2 display issues need to be addressed:

Should this be a new bug?

keqiaozhang, 2018-12-18:
Confirmed that this issue is fixed on latest GLK005 branch,  let's close this bug.
Fixed by commit 23da63.

closed 2018-12-18 by keqiaozhang