thesofproject/sof #723: Add timeout for HDA DMA Buffer full wait to avoid FW hang

keyonjie, 2018-12-14:
We are seeing FW	hang at waiting HDA DMA buffer full(looks we can't get buffer full at some cases, no matter how long we wait).

To avoid this, we need to implement TODO in below comment in hda-dma.c:hda_dma_host_preload(),

```
while (!(host_dma_reg_read(dma, chan->index, DGCS) & DGCS_BF) &&
	       (chan->state & HDA_STATE_BF_WAIT))
		; /* TODO: this should timeout and not wait forever */
```

keyonjie, 2018-12-14:
@mmaka1 Hi Marcin, I am assigning this to you, please feel free to assign it to anybody in your team who can help on it.

lgirdwood, 2018-12-17:
Lets make this timeout PLATFOM specific and define in platform.h

keyonjie, 2018-12-27:
some local change which actually workaround  this issue:
```
diff --git a/src/drivers/intel/cavs/hda-dma.c b/src/drivers/intel/cavs/hda-dma.c
index 7250f68d..9ea45906 100644
--- a/src/drivers/intel/cavs/hda-dma.c
+++ b/src/drivers/intel/cavs/hda-dma.c
@@ -327,12 +327,14 @@ static int hda_dma_host_preload(struct dma *dma, struct hda_chan_data *chan)
        };
        int i;
        int period_cnt;
+       int count = 20;
 
        /* waiting for buffer full after start
         * first try is unblocking, then blocking
         */
        while (!(host_dma_reg_read(dma, chan->index, DGCS) & DGCS_BF) &&
-              (chan->state & HDA_STATE_BF_WAIT))
+//            (chan->state & HDA_STATE_BF_WAIT))
+              (chan->state & HDA_STATE_BF_WAIT) && (count-- > 0))
                ; /* TODO: this should timeout and not wait forever */
 
        if (host_dma_reg_read(dma, chan->index, DGCS) & DGCS_BF) {
@@ -348,7 +350,9 @@ static int hda_dma_host_preload(struct dma *dma, struct hda_chan_data *chan)
                }
        } else {
                /* next call in pre-load state will be blocking */
-               chan->state |= HDA_STATE_BF_WAIT;
+//             chan->state |= HDA_STATE_BF_WAIT;
+               
+               chan->state &= ~(HDA_STATE_HOST_PRELOAD | HDA_STATE_BF_WAIT);
        }
 
        return 0;
```

wenqingfu, 2019-01-09:
according to the discussion in #783 , this issue can be closed. @keyonjie will open new issue if there are timeouts on top of it. Please be sure to include more details how to reproduce it next time.

closed 2019-01-09 by wenqingfu