thesofproject/sof #732: Topology: PIPELINE_PCM_ADD() with 50 frames per 1 ms is rejected by firmware

singalsu, 2018-12-17:
The firmware complains about the playback host sink buffer size if it is not a multiple of 32 bytes. E.g. 3x buffers of S24_LE for 50 frames needs 1200 bytes that is not accepted by FW.

Such 50 frames buffers are needed for topologies where SSP is configured for 50 kHz rate. The 50 kHz rate is needed to add flexibility to some platforms to work with some peripherals with those require 32/64 bit I2S word length.

plbossart, 2018-12-17:
@singalsu can you provide the topology file used?

tlauda, 2019-03-23:
@singalsu Please check with current master. The constraint is gone on cAVS platforms.

singalsu, 2019-03-28:
@tlauda @michalgrodzicki @wenqingfu Oops, thanks for reminder I'll check this now!

singalsu, 2019-03-28:
I've tried this but so far no success. One of the key problem is that the pipeline params/prepare walk propagates the 50 frames per 1ms (dev->frames) from DAI to PCM. The presence of SRC type component in the pipeline is not recognized. SRC might be able to alter it in prepare walk, maybe though. So the 50 constraint for buffers gets propagated to where it should not be needed. 

I was not able to get past 1) a host buffer size error, 2) a dw-dma config error, or 3) aplay buffers mismatch error. Also SRC is broken due to other SOF changes so I can't run it without corrupted audio. I need to study better the SOF and ALSA buffers system and control from topology to make better test topologies. I'll do it while updating SRC component first.

lgirdwood, 2019-03-29:
@singalsu can you align with @keyonjie , he is working on updates to make sure all params and constraints are propagated correctly through the pipeline. You should be able to resolve any remaining SRC issues after @keyonjie work is merged.

singalsu, 2019-03-29:
Excellent, I'll restart with this in beginning of next week. I'll share to @keyonjie my test topology.

tlauda, 2019-04-10:
@lgirdwood @singalsu @keyonjie @plbossart  The problem is also visible for the PCM Media pipeline, which is set to 192 frames. When we want to play some of the different sampling rates e.g. 8 kHz we are getting error: "host_params() error: component buffer size is not a divisor of host buffer size", which is basically true. Do we even need such constraint for DW-DMA on host side? It was already removed for Host HD-DMA and also I can see the code in host callback, which handles split transfers, so it looks like this condition is deprecated. @plbossart @ranj063 @keyonjie Any thoughts?

lgirdwood, 2019-04-15:
@tlauda restraint can probably be removed since @keyonjie did some updates in the past that means host transfers can be unaligned and detect wraps.

tlauda, 2019-04-22:
@singalsu Buffer constraint is gone and also SRC should be fixed. Could you tell us what is the currently received error?

singalsu, 2019-04-23:
OK, I'll check.

singalsu, 2019-04-23:
So there are many problems and I don't know how to work around the limitations. Essentially we want to  have DAI configuration like this

```
#SSP 5 (ID: 0)
DAI_CONFIG(SSP, 5, 0, SSP5-Codec,
	SSP_CONFIG(I2S, SSP_CLOCK(mclk, 19200000, codec_mclk_in),
		SSP_CLOCK(bclk, 3200000, codec_slave),
		SSP_CLOCK(fsync, 50000, codec_slave),
		SSP_TDM(2, 32, 3, 3),
		SSP_CONFIG_DATA(SSP, 5, 24)))

```

This can be tested with UP2 + HiFiberry DAC+ since it can work also with the 3.2 MHz bit clock and 50 kHz frame rate. However the next setting cannot be done 

```
# playback DAI is SSP5 using 2 periods
# Buffers use s24le format, with 50 frame per 1000us on core 0 with priority 0
DAI_ADD(sof/pipe-dai-playback.m4,
	1, SSP, 5, SSP5-Codec,
	PIPELINE_SOURCE_1, 2, s24le,
	50, 1000, 0, 0)

```

It result to this error in sof-logger

```
    0      2         PIPE 1.7         52209.427083         4.010417 src/audio/pipeline.c:320 	pipeline_params()
    0      2         HOST             52213.072917         3.645833     src/audio/host.c:514 	host_params()
    0      2          DMA             52219.739583         6.666667 intel/cavs/hda-dma.c:464 	hda-dmac: 5 channel 0 -> get
    0      2          DMA             52223.541667         3.802083 intel/cavs/hda-dma.c:685 	hda-dmac: 5 channel 0 -> config
    0      1          DMA             52227.031250         3.489583 intel/cavs/hda-dma.c:747 	hda-dmac: 5 chan 0 - buffer not DMA aligned 0x4b0
    0      1         HOST             52233.645833         6.614583     src/audio/host.c:610 	host_params() error: dma_set_config() failed
    0      1         PIPE             52239.739583         6.093750 src/audio/pipeline.c:329 	pipeline_params() error: ret = -22, host->comp.id = 0
    0      1          IPC             52245.833333         6.093750    src/ipc/handler.c:318 	ipc: pipe 1 comp 0 params failed -22
    0      2         PIPE 1.7         52252.083333         6.250000 src/audio/pipeline.c:583 	pipeline_reset()

```

However changing the previous DAI_ADD to use parameter 48 instead of 50 the pipeline will run without errors. I'm now working remotely this afternoon and UP2 is at my work desk so I can't listen if the playback was clean. I'll update about it tomorrow.

```
# playback DAI is SSP5 using 2 periods
# Buffers use s24le format, with 48 frame per 1000us on core 0 with priority 0
DAI_ADD(sof/pipe-dai-playback.m4,
	1, SSP, 5, SSP5-Codec,
	PIPELINE_SOURCE_1, 2, s24le,
	48, 1000, 0, 0)

```

singalsu, 2019-04-24:
@tlauda I get now clean playback with my 50 kHz test topology, thanks for working with the previous blockers!

Since it works now I think this issue would be best to close and the not-so-clean looking things like the one above in topology could be handled with separate issues. I think I'll show the test topology as a RFC / do not merge PR and discuss next actions there. I don't have myself test HW with non-standard rate I2S need so this UP2+HiFiBerry topology that I've used is a toy example.

singalsu, 2019-04-24:
Closing this.