thesofproject/sof #735: cmocka: can't build make check for host target.

lgirdwood, 2018-12-17:
make check fails for host target.
```
./scripts/host-build-all.sh
....
make[3]: Entering directory '/home/lrg/source/reef/sof.git/test/cmocka'
make  mixer  preproc_varargs_count preproc_concat preproc_seq preproc_defer preproc_get_arg debugability_macros rstrcmp rstrlen strcheck volume_process buffer_new buffer_write buffer_wrap buffer_copy comp_set_state list_init list_is_empty list_item_append list_item_del list_item_is_last list_item_prepend list_item gcd ceil_divide find_equal_int16 find_min_int16 find_max_abs_int32 norm_int32 sin_fixed
make[4]: Entering directory '/home/lrg/source/reef/sof.git/test/cmocka'
depbase=`echo src/audio/mixer/mixer_test.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I. -I../../src/include    -O3 -Wall -Werror -Wl,-EL -Wmissing-prototypes -Wimplicit-fallthrough=3 -I./include -I../../src/arch/host/include -I /home/lrg/source/reef/sof.git/src/include -g -O2 -MT src/audio/mixer/mixer_test.o -MD -MP -MF $depbase.Tpo -c -o src/audio/mixer/mixer_test.o src/audio/mixer/mixer_test.c &&\
mv -f $depbase.Tpo $depbase.Po
In file included from ../../src/include/sof/trace.h:39,
                 from ../../src/include/sof/ipc.h:36,
                 from src/audio/mixer/mixer_test.c:38:
../../src/include/sof/mailbox.h:35:10: fatal error: platform/mailbox.h: No such file or directory
 #include <platform/mailbox.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [Makefile:1382: src/audio/mixer/mixer_test.o] Error 1
make[4]: Leaving directory '/home/lrg/source/reef/sof.git/test/cmocka'
make[3]: *** [Makefile:1900: check-am] Error 2
make[3]: Leaving directory '/home/lrg/source/reef/sof.git/test/cmocka'
make[2]: *** [Makefile:1903: check] Error 2
make[2]: Leaving directory '/home/lrg/source/reef/sof.git/test/cmocka'
make[1]: *** [Makefile:382: check-recursive] Error 1
make[1]: Leaving directory '/home/lrg/source/reef/sof.git/test'
make: *** [Makefile:433: check-recursive] Error 1
```

lgirdwood, 2018-12-17:
@xiulipan once this is working, host cmocka build should be part of CI container tests.

xiulipan, 2018-12-18:
@lgirdwood 
Sure, I am looking forward to enable unit test into sof-ci and travis.

jajanusz, 2018-12-18:
@xiulipan 
We need cmocka 1.1+. On our side we use 1.2, so if you want to have cmocka built from exactly the same codebase you can use https://github.com/jajanusz/cmocka - I had to make fork because original cmocka doesn't support building for embedded platforms - You should be able to build with it for host and all xtensa platforms.

lgirdwood, 2018-12-19:
Fixed #750

PR #765 (2018-12-20): Fix ut host
I'm not sure, but I think you may need this to really close #735 issue for now

Fixes ut for host:
Mixer - fixed by increasing platform streams count

Disables ut for host:
Debugability - not designed to be run on host at all
Volume - needs some tuning to not fail
Buffer - needs some fixes to not have segfault
closed 2018-12-19 by lgirdwood