thesofproject/sof #736: cmocka: support for xtensa-gcc

lgirdwood, 2018-12-17:
cmocka must also be able to use xtensa GCC as compiler.

jajanusz, 2019-03-26:
@lgirdwood 
These unit tests has to be run somewhere.
Is qemu usable? If so, then I'll need some info how to run it.
In xtensa simulator I can run simple program like `main() { printf("Hello"); }` and it will print to stdout. I should be able to do it somehow in qemu.
AFAIK we use qemu in our ROM boot tests. If you could point me to some info how to setup it locally, it'd be helpful.

michalgrodzicki, 2019-03-26:
@lgirdwood any comment?

lgirdwood, 2019-03-26:
@jajanusz qemu simulates bare metal DSP from BYT -> ICL so it's missing C library support. In order to run the UTs on qemu (including CI) we would have to follow ROM test approach. i.e.

1) UT includes xtensa-crt.S that creates C runtime environment and jumps to main()
2)  printf() would call trace() infrastructure which would write to special memory device (which qemu printf()).

It seems we are missing qemu build instructions on sof-docs atm so (in qemu directory)
```
'./configure' '--prefix=.' '--target-list=xtensa-softmmu,x86_64-softmmu' '--enable-gtk' '--enable-sdl' '--enable-spice' '--audio-drv-list=alsa' '--enable-libusb' '--enable-usb-redir' '--enable-coroutine-pool' '--disable-opengl' '--enable-fdt'
make
``` 
Then you can run it by
```
xtensa-host.sh
usage: ./xtensa-host.sh device [-k kernel] [-t] [-d] [-i] [-r rom] [-c] [-g] [-o time log]
supported devices: byt, cht, hsw, bdw, bxt, sue, cnl, icl, skl, kbl, hky
[-k] | [--kernel]: load firmware kernel image
[-r] | [--rom]: load firmware ROM image
[-t] | [--trace]: trace DSP instructions
[-i] | [--irqs]: trace DSP IRQs
[-d] | [--debug]: enable GDB debugging - uses tcp::1234
[-c] | [--console]: Stall DSP and enter console before executing
[-g] | [--guest]: Display guest errors
[-o] | [--timeout]: Kill after timeout seconds
```

xiulipan, 2019-03-28:
@jajanusz 
you can check the 
https://github.com/thesofproject/sof/blob/master/scripts/qemu-check.sh

In this script, there are some example about how to read DSP => HOST IPC payload(IPC ready message)

lgirdwood, 2019-03-28:
@xiulipan would you be able to add some instructions on sof-decs for building and using qemu. You can aslo copy and pase the docs from the SOF wiki on alsa-project.org

xiulipan, 2019-03-29:
@lgirdwood good idea.
I create a issue in sof-docs to track this https://github.com/thesofproject/sof-docs/issues/102

jajanusz, 2019-04-30:
> It seems we are missing qemu build instructions on sof-docs atm so (in qemu directory)
>
> ```
> './configure' '--prefix=.' '--target-list=xtensa-softmmu,x86_64-softmmu' '--enable-gtk' '--enable-sdl' [...]

With that instructions I get:
```
  CC      util/memfd.o
util/memfd.c: In function ‘qemu_memfd_alloc’:
util/memfd.c:77:15: error: implicit declaration of function ‘memfd_create’; did you mean ‘timer_create’? [-Werror=implicit-function-declaration]
         mfd = memfd_create(name, MFD_ALLOW_SEALING | MFD_CLOEXEC);
               ^~~~~~~~~~~~
               timer_create
util/memfd.c:77:15: error: nested extern declaration of ‘memfd_create’ [-Werror=nested-externs]
cc1: all warnings being treated as errors
/home/admin/sof/repo/qwd/qemu/rules.mak:66: recipe for target 'util/memfd.o' failed
make: *** [util/memfd.o] Error 1
```

I used `git clone https://github.com/thesofproject/qemu` and then your instructions. Maybe some submodules use top commits instead of specific commits and it broke in meantime.

lgirdwood, 2019-04-30:
@jajanusz what distro and version are you using, it could be an older version of a dependency library....

lgirdwood, 2019-04-30:
@jajanusz see here too, could be old dependency https://github.com/systemd/systemd/issues/8099

jajanusz, 2019-04-30:
@lgirdwood 
```
admin@Linux-Ubuntu-Dev:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.5 LTS
Release:        16.04
Codename:       xenial
```

lgirdwood, 2019-04-30:
@jajanusz 16.4 is ancient, 3 years old. Please update. In general using LTS versions > 1 year old are not good for developers I'm on 18.10, will be moving to 19.04 soon.

plbossart, 2019-04-30:
out of curiosity has anyone used 19.04 for SOF work since it was released on April 19?

xiulipan, 2019-04-30:
@lgirdwood @plbossart 
It seems there are some update in QEMU repo?
I am now testing use Dockerhub to auto build the image used for CI test.

lgirdwood, 2019-04-30:
@xiulipan don't autobuild qemu for each SOF PR (too long, travis will timeout), just use the latest prebuilt binary in container. I will tag it in future so you can keep uptodate.

xiulipan, 2019-04-30:
@lgirdwood 
I am trying to track the https://github.com/thesofproject/qemu on DockerHub only to auto update the image used for CI.
It will make the binary used for CI always auto update after we have change in QEMU. I think we will have more patches in QEMU.

Related PR here: https://github.com/thesofproject/qemu/pull/4

jajanusz, 2019-05-02:
@lgirdwood I checked with ubuntu 19.04 and QEMU built successfully.
@plbossart I built SOF (FW) today on 19.04 with instructions for 18.10 from https://thesofproject.github.io/latest/getting_started/build-guide/build-from-scratch.html 
Need just minor fix for crosstool-ng -> https://github.com/thesofproject/crosstool-ng/pull/3

kv2019i, 2025-12-17:
I think this has been implemented long time a go. Anyways, no activity in 5 years, closing. Please reopen if still relevant.

closed 2025-12-17 by kv2019i