thesofproject/sof #737: cmocka: tests should also be able to run on xtensa-qemu

lgirdwood, 2018-12-17:
cmocka tests for xtensa platforms currently run on xt-run emulator. These test should also be to run on qemu too when statically linked with xtensa-newlib.

lgirdwood, 2018-12-17:
@xiulipan once working this should be part of CI.

michalgrodzicki, 2019-03-26:
@jajanusz what do you think?

jajanusz, 2019-03-26:
> @jajanusz what do you think?

Everything is possible, but I'd like to have some more info about our qemu: https://github.com/thesofproject/qemu

+what I already wrote #736 - these issues are related

lgirdwood, 2019-03-26:
@jajanusz it appears the qemu docs did not transfer over to sof-docs. I will take the action.

jajanusz, 2019-06-06:
@lgirdwood  @mmaka1 Should w invest time into that? Imo we should stick to one of xt-run or qemu. Supporting both may be extra effort we don't want. At most we should care of 1 simulator + host tests, actively supporting multiple simulators from our side may be too time-consuming. It's not like you just have to run it on qemu. Later we gonna have for example coverage so you'll have to add profiler support for xt-run and gcov somehow to qemu, and it's just DSP, what if we gonna test something more?

lgirdwood, 2019-06-06:
@jajanusz I'm happy with xtrun today, but we probably want to keep this feature "open" for the community as this may be useful to run cmocka on non xtensa targets with qemu.