thesofproject/sof #746: EQ components fail to ABI check with their configuration data

singalsu, 2018-12-18:
dmesg:

```
joulu 18 15:17:58 piikkisika kernel: sof-audio sof-audio: error: ipc error for 0x50030000 size 0x180
joulu 18 15:17:58 piikkisika kernel: sof-audio sof-audio: error: failed to set control 1325331 values
joulu 18 15:17:58 piikkisika kernel: sof-audio sof-audio: error: failed kcontrol value set for widget: 3
joulu 18 15:17:58 piikkisika kernel: sof-audio sof-audio: error: failed to restore pipeline after resume -22
```
sof-logger:

```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         COMP             70300.885417     70300.882812 ../../src/include/sof/audio/component.h:297 	comp_cmd() error: invalid version, data->magic = 4607827, data->abi = 33554432
    0      1          IPC             70306.458333         5.572917            handler.c:862 	ipc: comp 3 cmd 3 failed -22
```

singalsu, 2018-12-18:
After updating ABI version in configuration blobs to 3, 0, 0 I get this new error

```
    0      2          IPC             48044.375000         6.614583            handler.c:847 	ipc: hdr 0x50030000 tx (384) > rx (92)
    0      2          IPC             48047.916667         3.541667            handler.c:849 	ipc: comp 3 -> cmd 3
    0      2       EQ_FIR             48053.385417         5.468750             eq_fir.c:623 	eq_fir_cmd()
    0      2       EQ_FIR             48057.291667         3.906250             eq_fir.c:564 	fir_cmd_set_data(), SOF_CTRL_CMD_BINARY
    0      2       EQ_FIR             48062.760417         5.468750             eq_fir.c:590 	fir_cmd_set_data(): blob size: 56
    0      1          IPC             48068.802083         6.041667            handler.c:875 	ipc: comp 3 cmd 3 returned 384 bytes max 384

```

singalsu, 2018-12-18:
@lgirdwood Should EQ set the _data->rhdr.hdr.size to something once it has handled the set data command? I understand this check for get data but for set I wonder how the component should respond. The error above is from this code in handler.c

```
	/* write component values to the outbox */
	if (_data->rhdr.hdr.size <= MAILBOX_HOSTBOX_SIZE &&
	    _data->rhdr.hdr.size < SOF_IPC_MSG_MAX_SIZE) {
		mailbox_hostbox_write(0, _data, data.rhdr.hdr.size);
		ret = 1;
	} else {
		trace_ipc_error("ipc: comp %d cmd %u returned %d bytes max %d",
				data.comp_id, data.cmd, _data->rhdr.hdr.size,
				MIN(MAILBOX_HOSTBOX_SIZE,
				    SOF_IPC_MSG_MAX_SIZE));
		ret = -EINVAL;
	}

```

ranj063, 2018-12-19:
@singalsu the reply size should depend on what you set the replay message to be in the driver. In this case, I don't think we need to return the entire data back in the reply

singalsu, 2018-12-19:
I tried to set the rhdr.hdr.size to the length of binary data payload, this time 56 bytes (0x38). The DSP side is fine but kernel returns this:

```
[   67.682666] sof-audio sof-audio: error: reply expected 0x180 got 0x38 bytes
[   67.682880] sof-audio sof-audio: error: ipc error for 0x50030000 size 0x180
[   67.682905] sof-audio sof-audio: error: failed to set control 3 values
[   67.682910] sof-audio sof-audio: error: failed kcontrol value set for widget: 3
[   67.682915] sof-audio sof-audio: error: failed to restore pipeline after resume -22
```

Next I try to set it to 384 (0x180) and get this error on FW side:

```
    0      2       EQ_FIR             70259.635417         4.635417             eq_fir.c:626 	eq_fir_cmd()
    0      2       EQ_FIR             70262.552083         2.916667             eq_fir.c:564 	fir_cmd_set_data(), SOF_CTRL_CMD_BINARY
    0      2       EQ_FIR             70266.666667         4.114583             eq_fir.c:590 	fir_cmd_set_data(): blob size: 56
    0      1          IPC             70271.979167         5.312500            handler.c:875 	ipc: comp 3 cmd 3 returned 384 bytes max 384

```

This looks like a bug in IPC @lgirdwood ?

singalsu, 2018-12-19:
I can make it work with this change. Should I include this to EQ blobs update PR?

```
diff --git a/src/ipc/handler.c b/src/ipc/handler.c
index 45042e2e..5d4843db 100644
--- a/src/ipc/handler.c
+++ b/src/ipc/handler.c
@@ -865,7 +865,7 @@ static int ipc_comp_value(uint32_t header, uint32_t cmd)
 
        /* write component values to the outbox */
        if (_data->rhdr.hdr.size <= MAILBOX_HOSTBOX_SIZE &&
-           _data->rhdr.hdr.size < SOF_IPC_MSG_MAX_SIZE) {
+           _data->rhdr.hdr.size <= SOF_IPC_MSG_MAX_SIZE) {
                mailbox_hostbox_write(0, _data, data.rhdr.hdr.size);
                ret = 1;
        } else {

```

singalsu, 2018-12-19:
@plbossart @lgirdwood That should be fine, hope so -- looks symmetrical now at least.  But I don't understand why kernel uses and accepts only the max IPC size while e.g. this offending set data message triggered by topology sof-apl-eq-pcm512x.tplg used only 56 bytes of payload.

singalsu, 2018-12-19:
Fix for this issue is in PR #756

singalsu, 2019-01-02:
The PR is merged so this can be closed.

closed 2019-01-02 by singalsu