thesofproject/sof #747: Playback of > 48 kHz content fails with topology sof-apl-src-pcm512x.tplg

singalsu, 2018-12-18:
The aplay command reports just "aplay: pcm_write:2011: write error: Input/output error". The sof-logger trace contains nothing from the event.

The dmesg contains this:

j```
oulu 18 16:35:22 piikkisika kernel: sof-audio sof-audio: error: ipc timed out for 0x60050000 size 0xc
joulu 18 16:35:22 piikkisika kernel: sof-audio sof-audio: status: fw entered - code 00000005
joulu 18 16:35:22 piikkisika kernel: sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
joulu 18 16:35:22 piikkisika kernel: sof-audio sof-audio: error: waking up any trace sleepers
joulu 18 16:35:22 piikkisika kernel:  Port5: ASoC: trigger FE failed -110
joulu 18 16:35:22 piikkisika kernel: sof-audio sof-audio: pcm: free stream 0 dir 0
joulu 18 16:35:22 piikkisika kernel: sof-audio sof-audio: error: ipc timed out for 0x60030000 size 0xc
joulu 18 16:35:22 piikkisika kernel: sof-audio sof-audio: status: fw entered - code 00000005
joulu 18 16:35:22 piikkisika kernel: sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
joulu 18 16:35:22 piikkisika kernel: sof-audio sof-audio: error: waking up any trace sleepers
joulu 18 16:35:22 piikkisika kernel: sof-audio sof-audio: pcm: close stream 0 dir 0
joulu 18 16:35:25 piikkisika kernel: sof-audio sof-audio: error: ipc timed out for 0x40010000 size 0x4c
joulu 18 16:35:25 piikkisika kernel: sof-audio sof-audio: status: fw entered - code 00000005
joulu 18 16:35:25 piikkisika kernel: sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
joulu 18 16:35:25 piikkisika kernel: sof-audio sof-audio: error: ctx_save ipc error during suspend -110

```

lgirdwood, 2018-12-19:
@juimonen @singalsu can you desribe the pipeline, this looks like setting 96kHz is causing a crash for your pipe.

lgirdwood, 2019-01-23:
@juimonen any update ? Can we close ?

juimonen, 2019-01-24:
I didn't have this issue with pr842, but can check after rebase.

lgirdwood, 2019-01-24:
@juimonen ok pls close if all is ok after rebase.

lgirdwood, 2019-01-24:
Ok, can you use the crash dump tool to dump the crash and attach logs. Also what is the pipeline you are running ?

juimonen, 2019-01-24:
Yes I'll figure this out. and yes I'm running src pipeline with dai set to 48 and pcm range 8-192kHz. This is with my PR842...

singalsu, 2019-02-01:
@juimonen @keqiaozhang I just tested that playback of all support high rates 64, 88.2, 96, 176.4, and 192 kHz works now. The high rates needs xt-xcc compiled firmware. Thanks Jaska for fixing this!

closed 2019-02-01 by singalsu

lgirdwood, 2019-02-01:
@singalsu we need to add some compile logic that detects compiler type and makes SRC reject at runtime rates that are too high for GCC.

juimonen, 2019-02-01:
@lgirdwood I think the logic is there. after the host dma channel oops was fixed I can see with my gcc compiled fw that src is rejecting rates it can't do...