thesofproject/sof #749: Potental race condition between the host driver and the dsp when going to D3

mmaka1, 2018-12-18:
DSP performs memory power down routine right after the `SOF_IPC_PM_CTX_SAVE` is handled and the response is written to the uplink mailbox allocated inside the memory being turned off (see `ipc_platform_do_cmd()` flow).
If the host driver is late on reading the mailbox content when signaled by the interrupt, the content is gone. Note that memory window is not disabled by the current implementation as well.

No additional synchronization would be required, if specific messages (including the ctx_save reply) are re-designed to be short ones and fit into the IPC registers. So that no memory window is required for additional content. Note that there are other flows required in low power states (no mem wnd avialable) being discussed.

lbetlej, 2018-12-18:
I agree that we need to have option for short IPC messages (register only). One solution is to extend current API to have a flag in register based communication that indicates if there is data in memory window that should be interpreted. Adding a reference to ongoing discussion on IPC ABI https://github.com/thesofproject/linux/issues/414

lgirdwood, 2018-12-18:
We have to be careful here and abstract any register read part into the HW abstraction layer as the IPC layer is HW agnostic.  A simpler approach should use something like UDP messaging where messages can be sent by the sender whilst the receiver never replies or acknowledges a message. This would be HW agnostic and simple to abstract into the IPC layer.

ranj063, 2018-12-19:
@lgirdwood I agree with the UDP analogy but our driver currently only schedules messages upon receiving a reply. So that will need to change as well

lgirdwood, 2018-12-19:
@ranj063 in this case we have a flag for "no reply" messages which would force the IPC to suceed and reset any "waiting" logic to be ready for subsequent new IPCs.

mmaka1, 2018-12-19:
@lgirdwood 
> We have to be careful here and abstract any register read part into the HW abstraction layer 

I agree. I had assumed that this is purely the implementation of the low level IPC transport, specific for a given platform. So we are free to decide what part of the message is written into which piece of that platform specific HW. It might be completely different for another one which does not have IPC registers pairs but something else. So this should be hidden inside the IPC driver's send()/recv() implementation which we may (must in some cases) implement as mem-less for short ones.
Of course, in the application layer, the IPC message content, Rx/Tx'ed from the underlying transport, should follow a single common definition.

ranj063, 2018-12-19:
> @ranj063 in this case we have a flag for "no reply" messages which would force the IPC to suceed and reset any "waiting" logic to be ready for subsequent new IPCs.

@lgirdwood does this "no reply" flasg already exist in the code today or is it something we need to implement?

lgirdwood, 2018-12-19:
@ranj063 something that needs added.

lbetlej, 2018-12-19:
I'm OK with "no reply" flag as a general capability but how this approach addresses potential race issue caused by driver executing D0->D3 flow after SOF_IPC_PM_CTX_SAVE is sent in parallel with FW executing its D0->D3 flow ? In my opinion there is a need to secure synchronization between the SW and FW. Driver needs to know that FW finished handling SOF_IPC_PM_CTX_SAVE (i.e. power gating is finished by FW) and can proceed with HW power gating done by host.

plbossart, 2018-12-19:
On 12/19/18 3:11 PM, Lech Betlej wrote:
> I'm OK with "no reply" flag as a general capability but how this
> approach addresses potential race issue caused by driver executing
> D0->D3 flow after SOF_IPC_PM_CTX_SAVE is sent in parallel with [...]
I'm with Lech, i don't see what the noreply brings. My understanding is 
that you want a 'secure' way of sending a command that isn't affected by 
power transitions, e.g. memory windows being power gated. You'd need 
some sort of shared hardware register or leave parts of the memory bank 
powered.

lgirdwood, 2018-12-20:
Ok, I've had a look at the power down code and it does have a pre-existing comment for IPC since there is an obvious potential for race.
@ranj063 I've created a new branch pm-d3 which sends an IPC at the completion of power down rather than at start of power down. Compile tested only.
https://github.com/thesofproject/sof/tree/lrg/topic/pm-d3

wenqingfu, 2018-12-24:
Is this fixed by #772 ?

ranj063, 2019-01-15:
@wenqingfu yes this is fixed by #772 and the patches are both on glk-004 and the master branches.

wenqingfu, 2019-01-15:
Thank you, @ranj063 ! 

Close as fixed by #772

closed 2019-01-15 by wenqingfu