thesofproject/sof #751: dmic trace error at boot on GLK

keqiaozhang, 2018-12-19:
There're some dmic config error in trace log at boot. The DMIC function is normal.

```
    0      2         DMIC            103989.687500         4.218750               dmic.c:1119   dmic_set_config()
    0      1         DMIC            103997.395833         7.708333               dmic.c:1155   dmic_set_config() error: prm driver_ipc_version = 1, prm->num_pdm_active = 2
    0      1         DMIC            104003.750000         6.354167               dmic.c:1159   dmic_set_config() error: pdmclk_min = 500000, pdmclk_max = 4800000, fifo_fs_a = 48000, fifo_fs_b = 0
    0      1         DMIC            104010.416667         6.666667               dmic.c:1163   dmic_set_config() error: prm fifo_bits_a = 16, fifo_bits_b = 16, duty_min = 40, duty_max = 60
    0      2         DMIC            104034.635417        24.218750               dmic.c:1216   dmic_set_config(), cfg clkdiv = 8
    0      2         DMIC            104038.750000         4.114583               dmic.c:1219   dmic_set_config(), mcic = 25, mfir_a = 2, mfir_b = 0, cic_shift = 3
    0      2         DMIC            104042.604167         3.854167               dmic.c:1223   dmic_set_config(), cfg fir_a_length = 101, fir_b_length = 0, fir_a_shift = 0, fir_b_shift = 0
```
How to reproduce:
1. After system boot, check the trace log:
sof-logger -l sof-apl.ldc -t 

Tested with:
SOF/TPLG: glk-005-drop-stable/4f58fdb
Kernel: cros-sof-v4.14-rebase/730ace9

ranj063, 2018-12-19:
@keqiaozhang is dmic capture working at all?

ranj063, 2018-12-19:
fixed by https://github.com/thesofproject/sof/pull/754

keqiaozhang, 2018-12-19:
@ranj063 
dmic capture works well except for the channel 4. 
I'm curious about the data of channel 4. I can only hear the noise( marked as red) at the beginning, then I cannot hear anything, is this a normal behavior?
[image]

keqiaozhang, 2018-12-19:
@singalsu 
Could you help to confirm the 4th channel issue of the dmic? I recorded that I got the same waveform on UP^2 with nocodec topology, I'm not sure whether these are all the normal behavior?

singalsu, 2018-12-19:
@keqiaozhang Can you please attach the waveform. It would be normal if channels 3 and 4 would be like this if there's only one or two microphones in the device.

If channel 3 contains valid audio then something is pulling the data line high when mic 3 doesn't drive the PDM bus data line. A steady logic one data line appears as negative maximum but as you see in the plot the DCCOMP feature in HW causes it to remain at half of full scale (a HW limitation in amount of how large DC values it can suppress). The curve in the beginning is the logarithmic gain ramp to slowly enable the channel. When there's a mic connected the DC value is usually settled once the gain ramp is complete.

keqiaozhang, 2018-12-20:
@singalsu 
The recorded wav file is attached, the 3th channel contains valid data.
[dmic.tar.gz](https://github.com/thesofproject/sof/files/2697590/dmic.tar.gz)

singalsu, 2018-12-20:
Is that wrong file? The wav file is very short and corrupt. Normally file command reports the format of a wav file.

```
$ ls -l dmic.wav 
-rw-r--r-- 1 singalsu singalsu 10240 joulu 20 08:01 dmic.wav
$ file dmic.wav 
dmic.wav: data
```

closed 2018-12-24 by keqiaozhang