thesofproject/sof #752: [GLK] SOF does not handle XRUNs correctly

naveen-manohar, 2018-12-19:
There appears to be bug with how SOF is handling XRUNs which causes audio distortion (buffer alignment error?)

How to reproduce
Run speaker-test -D hw:0,0 -t sine -c 2
Suspend and resume the task repeatedly (using C-z and fg)
Eventually you will get an iteration where the sine wave it noticeably no longer playing the correct tune and sounds distorted.

keqiaozhang, 2018-12-20:
Verified that this issue is fixed on glk-005-drop-stable branch.
Tested with: 
SOF: glk-005-drop-stable/4d7904d.
Driver: cros-sof-v4.14-rebase/2aa548d

@naveen-manohar, please help to confirm it at your end.

naveen-manohar, 2018-12-20:
Tried remotely and I hit a reboot when trying speaker-play cmd in Terminal-1 and suspend/resume in T-2. Shall recheck tmrw at Lab and update whether the changes are needed in 004 branch too. Thanks

mengdonglin, 2018-12-29:
Fixed by #755 and #757, closing this issue

closed 2018-12-29 by mengdonglin