thesofproject/sof #758: IIR EQ should be optimized with HiFi3 intrinsics

singalsu, 2018-12-19:
GCC built generic C version is slow though it's fast enough for realtime operation while there is not much other processing in firmware. Compilation with xt-xcc gives better performance but good MCPS performance and low power is achieved only with intrinsics optimized code like volume, FIR, and SRC components.

singalsu, 2019-06-06:
@jajanusz OK, got it. It's a fairly small task, I'll start doing next week.

jajanusz, 2019-06-06:
@singalsu It would be great. However it's low priority task, so feel free to do it only if you have nothing better to do.

lgirdwood, 2019-06-06:
@singalsu lets have this for v1.4 since it's a small task, Btw does FIR require the same ?

singalsu, 2019-06-07:
@lgirdwood FIR is optimized already since it was hit worst by generic C implementation. Fortunately I did the intrinsics version for year ago OSTS demo.

singalsu, 2019-06-14:
@lgirdwood I didn't find time for this, new ETA is beginning of July, after my vacation ends plus need some work days.

singalsu, 2019-07-11:
SOF PR #1624 addresses this requirement.

singalsu, 2019-08-19:
This can be closed since PR #1624 is merged.

closed 2019-08-19 by singalsu