thesofproject/sof #766: comp_new ipc failure due to invalid ipc size

ranj063, 2018-12-20:
I ran into this issue today where the comp_new ipc failed because of invalid size.
The ipc size is 76 which seems to be correct but the config.hdr.size doesnt match the expected size.

dmesg logs: 
[  253.116672] sof-audio sof-audio: ipc tx: 0x30010000
[  253.116796] sof-audio sof-audio: ipc tx succeeded: 0x30010000
[  253.116811] sof-audio sof-audio: ipc tx: 0x30100000
[  253.116905] sof-audio sof-audio: ipc tx succeeded: 0x30100000
[  253.116934] sof-audio sof-audio: ipc tx: 0x30010000
[  253.117204] sof-audio sof-audio: error: ipc error for 0x30010000 size 0x14
[  253.117223] sof-audio sof-audio: error: failed to load widget type 23 with ID: 27
[  253.117226] sof-audio sof-audio: error: failed to restore pipeline after resume -22

logger:
    

0      2         PIPE             95961.562500        18.177084           pipeline.c:245 	pipeline_new()
    0      2          IPC             96060.781250        99.218750            apl-ipc.c:114 	ipc: msg rx -> 0x30010000
    0      2          IPC             96114.687500        53.906250            handler.c:908 	ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC             96118.333333         3.645833            handler.c:911 	ipc: pipe 5 comp 27 -> new (type 1)
    0      2         HOST             96131.093750        12.760417               host.c:338 	host_new()
    0      1         HOST             96135.156250         4.062500               host.c:341 	ipc: size 0 expected 36
    0      1         COMP             96141.354167         6.197917          component.c:90  	comp_new() error: unable to create the new component
    0      1          IPC             96146.406250         5.052083                ipc.c:119 	ipc_comp_new() error: component cd = NULL

lgirdwood, 2018-12-20:
@jajanusz @dabekjakub, @ranj063 mentioned a similar bug on this mornings call, can we confirm what the hdr.size is on the DSP and host sides. It could be some corruption at one end....

ranj063, 2018-12-20:
@dabekjakub I have pushed some changed to the kernel to print debug messages with the ipc data before sending the ipc to the same branch as below:
https://github.com/ranj063/linux/tree/debug/suspend_stress_test

ranj063, 2018-12-21:
@dabekjakub some more logs from the changes I made today. I printed the ipc header and payload on both the host and DSP and I can clearly see that all the data is there but the ipc still fails. This is really baffling. 

logger: 
   0      2          IPC             96065.156250       125.156250            apl-ipc.c:114 	ipc: msg rx -> 0x30010000
    0      2          IPC             96070.572917         5.416667            handler.c:1064	ipc: hdr 0x30010000 tx (80) > rx (28)
    0      2          IPC             96073.854167         3.281250            handler.c:1067	ipc: pipe 5 comp 31 -> new (type 2)
    0      2          IPC             96077.395833         3.541667            handler.c:80  	ipc header cmd: 0x30010000 size: 80

    0      2          IPC             96081.406250         4.010417            handler.c:145 	id: 31 type: 2 pipeline_id: 5
    0      2          IPC             96085.156250         3.750000            handler.c:80  	ipc header cmd: 0x30010000 size: 80

    0      2          IPC             96089.062500         3.906250            handler.c:145 	id: 31 type: 2 pipeline_id: 5
    0      2          IPC             96092.604167         3.541667            handler.c:150 	config header

    0      2          IPC             96095.989583         3.385417            handler.c:80  	ipc header cmd: 0x0 size: 36

    0      2          IPC             96099.947917         3.958333            handler.c:153 	periods_sink: 0 periods_source: 2
    0      2          IPC             96103.229167         3.281250            handler.c:155 	preload_count: 2 frame_fmt: 2
    0      2          IPC             96106.302083         3.072917            handler.c:156 	xrun_action: 0

    0      2          IPC             96109.479167         3.177083            handler.c:184 	dir: 0 dai_index: 3
    0      2          IPC             96112.656250         3.177083            handler.c:186 	type: 3 dmac_config: 0

    0      2          DAI             96125.729167        13.072917                dai.c:178 	dai_new()
    0      1          DAI             96183.489583        57.760418                dai.c:204 	dai_new() error: dai_get() failed to create DAI.
    0      1         COMP             96191.718750         8.229167          component.c:90  	comp_new() error: unable to create the new component
    0      1          IPC             96196.614583         4.895833                ipc.c:119 	ipc_comp_new() error: component cd = NULL
    0      1          IPC             96201.770833         5.156250            handler.c:1075	ipc: pipe 5 comp 31 creation failed -22

driver logs:
 1686.285024] sof-audio sof-audio: ipc header cmd: 0x30010000 size: 80
[ 1686.285025] sof-audio sof-audio: ipc comp id: 31 type: 2 pipeline id: 5
[ 1686.285026] sof-audio sof-audio: config header
[ 1686.285028] sof-audio sof-audio: ipc header cmd: 0x0 size: 36
[ 1686.285029] sof-audio sof-audio: periods_sink: 0 periods_source: 2 preload_count: 2 frame_fmt: 2 xrun_action: 0
[ 1686.285031] sof-audio sof-audio: dir: 0 dai_index: 3 type: 3 dmac_config: 0
[ 1686.285055] sof-audio sof-audio: ipc tx: 0x30010000
[ 1686.285265] sof-audio sof-audio: ipc rx: 0x90020000
[ 1686.285282] sof-audio sof-audio: ipc rx done: 0x90020000
[ 1686.285302] sof-audio sof-audio: error: ipc error for 0x30010000 size 0x14
[ 1686.285311] sof-audio sof-audio: error: failed to load widget type 27 with ID: 31
[ 1686.285314] sof-audio sof-audio: error: failed to restore pipeline after resume -22

keyonjie, 2018-12-21:
> @dabekjakub some more logs from the changes I made today. I printed the ipc header and payload on both the host and DSP and I can clearly see that all the data is there but the ipc still fails. This is really baffling.

> [ 1686.285314] sof-audio sof-audio: error: failed to restore pipeline after resume -22

@ranj063 can you share the FW patch you used for this log also?

ranj063, 2018-12-21:
@keyonjie here it is:
https://github.com/ranj063/sof/tree/debug/suspend_stress_test

Please make sure to change the xtensa-build-all.sh to use "xtensa" instead of xtensa-smp for ARCH in apollolake.

dabekjakub, 2018-12-21:
@ranj063 that is in line with what i observed. i will have to track where the change of size to 0 occurs.

ranj063, 2018-12-21:
@dabekjakub, @xiulipan made an amazing discovery that this could be related to the multicore support in the FW. We are testing this with xtensa arch option instead of xtensa-smp and it seems to be doing better. Do give it a try and let us know.

dabekjakub, 2018-12-21:
I don't think that disabling smp architecture is the way i will be investigating it as is. Also issue only occurs on experimental kernel. I believe we did not see it in any other circumstances.

ranj063, 2018-12-21:
@dabekjakub agree but it could be a real issue when it comes to high frequency memory reads/writes. Not seen with our official kernel

keyonjie, 2018-12-21:
@dabekjakub please consider disabling smp for the test, the issue #743 is disappear also on my side.

dabekjakub, 2018-12-21:
@keyonjie Why would i disable it if You confirmed it fixes it :P I will use that knowledge to investigate it.
@ranj063 That is right.

lgirdwood, 2018-12-21:
@ranj063 sorry, not following can you confirm 
1) IPC structure data is valid and correct on host side prior to the IPC error.
2) IPC structure and data on DSP size != host data ?

ranj063, 2018-12-21:
@lgirdwood here's an example of the difference between the host and the DSP.
The top part is the logger from the DSP and bottom part is the host side for the ipc that fails.

    0      2          IPC            100208.645833        10.052083            handler.c:1066	ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC            100212.708333         4.062500            handler.c:1069	ipc: pipe 6 comp 34 -> new (type 1)
    0      2          IPC            100216.354167         3.645833            handler.c:80  	ipc header cmd: 0x30010000 size: 76

    0      2          IPC            100220.729167         4.375000            handler.c:145 	id: 34 type: 1 pipeline_id: 6
    0      2          IPC            100224.739583         4.010417            handler.c:80  	ipc header cmd: 0x30010000 size: 76

    0      2          IPC            100228.645833         3.906250            handler.c:145 	id: 34 type: 1 pipeline_id: 6
    0      2          IPC            100232.395833         3.750000            handler.c:150 	config header

    0      2          IPC            100235.833333         3.437500            handler.c:80  	ipc header cmd: 0x1388 size: 0

    0      2          IPC            100239.739583         3.906250            handler.c:153 	periods_sink: 48 periods_source: 0
    0      2          IPC            100243.125000         3.385417            handler.c:155 	preload_count: 0 frame_fmt: 2
    0      2          IPC            100246.302083         3.177083            handler.c:156 	xrun_action: 0

[  171.514999] sof-audio sof-audio: ipc header cmd: 0x30010000 size: 76
[  171.515000] sof-audio sof-audio: ipc comp id: 34 type: 1 pipeline id: 6
[  171.515001] sof-audio sof-audio: config header
[  171.515003] sof-audio sof-audio: ipc header cmd: 0x0 size: 36
[  171.515004] sof-audio sof-audio: periods_sink: 2 periods_source: 0 preload_count: 2 frame_fmt: 0 xrun_action: 0
[  171.515019] sof-audio sof-audio: direction: 0 no_irq: 0 dmac_config: 0

lgirdwood, 2018-12-21:
@ranj063 thanks, btw can you try the checks in my email and can you also try and read back the data on the host side when it fails and compare it to what was sent. i.e. has it been altered on host side too ?
Please also use the debug macros for dumping that data in FW as they can be read from debugFS and easily compared to driver side dumps. The debug macro will copy data to another SRAM window that is exposed to userspace via debugFS.

Fwiw, te 0x1388 in the debug above for cmd actually looks like a 16bit bit mask for something ?

ranj063, 2018-12-21:
@lgirdwood I am trying the delay after powering down memory banks. I will keep you posted on the results.

Also which debug macros should I do using? Is it the ones in debug.h?

dabekjakub, 2019-01-08:
@ranj063 I debuged it and found that ipcs that fail call to create dai of type 0 

Whilist we do not have type 0 in our predefined structs (We only have ssp dmic and hda avalible). We do overall have enum SOF_DAI_INTEL_NONE but .

```
enum sof_ipc_dai_type {
	SOF_DAI_INTEL_NONE = 0,		/**< None */
	SOF_DAI_INTEL_SSP,		/**< Intel SSP */
	SOF_DAI_INTEL_DMIC,		/**< Intel DMIC */
	SOF_DAI_INTEL_HDA,		/**< Intel HD/A */
};
```

in my enchanced logs You can see

    0      2          DAI             99590.937500         3.750000                dai.c:184 	ipc size ipc_dai ptr 0xbe051480 , config size: 24comp hdr size: 50
    0      2          IPC             99594.375000         3.437500                dai.c:185 	comp ptr 0xbe051480
    0      2          IPC             99597.760417         3.385417                dai.c:190 	ipc data byte:0 data: 50
    0      2          IPC             99601.250000         3.489583                dai.c:190 	ipc data byte:1 data: 30010000
    0      2          IPC             99604.531250         3.281250                dai.c:190 	ipc data byte:2 data: 26
    0      2          IPC             99607.864583         3.333333                dai.c:190 	ipc data byte:3 data: 2
    0      2          IPC             99611.302083         3.437500                dai.c:190 	ipc data byte:4 data: 6
    0      2          IPC             99614.635417         3.333333                dai.c:190 	ipc data byte:5 data: 0
    0      2          IPC             99617.968750         3.333333                dai.c:190 	ipc data byte:6 data: 0
    0      2          IPC             99621.406250         3.437500                dai.c:190 	ipc data byte:7 data: 24
    0      2          IPC             99624.739583         3.333333                dai.c:190 	ipc data byte:8 data: 0
    0      2          IPC             99628.177083         3.437500                dai.c:190 	ipc data byte:9 data: 0
    0      2          IPC             99631.510417         3.333333                dai.c:190 	ipc data byte:a data: 2
    0      2          IPC             99634.843750         3.333333                dai.c:190 	ipc data byte:b data: 2
    0      2          IPC             99638.281250         3.437500                dai.c:190 	ipc data byte:c data: 2
    0      2          IPC             99641.562500         3.281250                dai.c:190 	ipc data byte:d data: 0
    0      2          IPC             99644.843750         3.281250                dai.c:190 	ipc data byte:e data: 0
    0      2          IPC             99648.281250         3.437500                dai.c:190 	ipc data byte:f data: 0
    0      2          IPC             99651.562500         3.281250                dai.c:190 	ipc data byte:10 data: 0
    0      2          IPC             99654.947917         3.385417                dai.c:190 	ipc data byte:11 data: 0
    0      2          IPC             99658.229167         3.281250                dai.c:190 	ipc data byte:12 data: 0
    0      2          IPC             99661.510417         3.281250                dai.c:190 	ipc data byte:13 data: 0

byte 12 should be of type either 1 2 or 3 for ssp dmic or hda but is sent as 0. ()

Second fail I saw was when creating pipeline and again sent ipc has 0 in a struct field that is handling size 

> 
>     0      2          IPC             92530.520833        11.979167            apl-ipc.c:114 	ipc: msg rx -> 0x30010000
>     0      2          IPC             92583.489583        52.968750            handler.c:909 	ipc comp hdr size: 76
>     0      2          IPC             92587.343750         3.854167            handler.c:914 	ipc data byte:0 data: 4c
>     0      2          IPC             92591.041667         3.697917            handler.c:914 	ipc data byte:1 data: 30010000
>     0      2          IPC             92594.375000         3.333333            handler.c:914 	ipc data byte:2 data: 8
>     0      2          IPC             92597.708333         3.333333            handler.c:914 	ipc data byte:3 data: 1
>     0      2          IPC             92601.145833         3.437500            handler.c:914 	ipc data byte:4 data: 2
>     0      2          IPC             92604.479167         3.333333            handler.c:914 	ipc data byte:5 data: 0
>     0      2          IPC             92607.812500         3.333333            handler.c:914 	ipc data byte:6 data: 0
>     0      2          IPC             92611.145833         3.333333            handler.c:914 	ipc data byte:7 data: 0 
>     0      2          IPC             92614.479167         3.333333            handler.c:914 	ipc data byte:8 data: 1388
>     0      2          IPC             92617.916667         3.437500            handler.c:914 	ipc data byte:9 data: 30
>     0      2          IPC             92621.250000         3.333333            handler.c:914 	ipc data byte:a data: 0
>     0      2          IPC             92624.583333         3.333333            handler.c:914 	ipc data byte:b data: 0
>     0      2          IPC             92628.072917         3.489583            handler.c:914 	ipc data byte:c data: 0
>     0      2          IPC             92631.406250         3.333333            handler.c:914 	ipc data byte:d data: 0
>     0      2          IPC             92634.739583         3.333333            handler.c:914 	ipc data byte:e data: 0
>     0      2          IPC             92638.072917         3.333333            handler.c:914 	ipc data byte:f data: 0
[... cut ...]
To sum up every issue i see is due to data that is beeing sent in the ipc (and i checked ipc integrity it is the same when recived and when used). 
@ranj063 please confirm with my fw build that data You send is the same as data i recive in event of fail. If Yes then above issue is due to wrong data sent in ipc message. Please use smp for this build. I can see no connection with this fails and smp build (unless there are references in kernel)

You can find my build with full dai ipcs loged on my fork:
https://github.com/dabekjakub/sof/tree/ipc_data_show

ranj063, 2019-01-09:
@lgirdwood @dabekjakub, @fredoh9 also hit this issue while trying to load the nocodec topology on the up squared. So it is no longer a stress test issue only. 
Fred, could you please help with checking if the data is getting corrupted at any point prior to the IPC being sent in the driver?

fredoh9, 2019-01-16:
My case is a bit different from this thread. I built nocodec build for UP2 but I don't have sound card listed. I was trying to digest my problem/solution and issues in this thread. I attached my logs here.

[logger_0108_fails.log](https://github.com/thesofproject/sof/files/2765699/logger_0108_fails.log)
[UP2_0108_sof-dev_nocodec_tplg.log](https://github.com/thesofproject/sof/files/2765700/UP2_0108_sof-dev_nocodec_tplg.log)

I rebased to 2981a6f22c23624e15a5e5efb3d19cca12c044f6 in yesterday, I tried nocodec build again with ARCH=xtensa-smp, now I can see sof-nocodec sound card. I see some related changes are in recently. If you need to use my setup with the baseline as of Jan 8th, I can happy to try more testing also.

lgirdwood, 2019-01-23:
@ranj063 @fredoh9 any update, can we close if resolved ?

ranj063, 2019-01-24:
@lgirdwood I dont want to close this issue yet. We have met IPC errors lately in several different situations. This script is the only way to reproduce it consistently. I will work with @fredoh9 to check if the IPC message is intact at the point when the doorbell is sent to the DSP

lgirdwood, 2019-01-25:
@ranj063 ok, fwiw the recent IPC timeout stress test failures could be related to scheduling. e.g. IPC handler is being scheduled, but not being run. I agree, pls do check message intactness through out the IPC flow.  The script can also be upstreamed when you have time.

michalgrodzicki, 2019-02-13:
> I dont want to close this issue yet. We have met IPC errors lately in several different situations. This script is the only way to reproduce it consistently. I will work with @fredoh9 to check if the IPC message is intact at the point when the doorbell is sent to the DSP

@ranj063 I'll asign this issue to you.

naveen-manohar, 2019-04-02:
Issue observed on CML v4 series during stress cycle
2019-03-28T14:33:40,309989+05:30 sof-audio-pci 0000:00:1f.3: error: ipc error for 0x30010000 size 0x14
2019-03-28T14:33:40,309997+05:30 sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 27 name : SSP1.OUT stream SSP1-Codec reply -22
2019-03-28T14:33:40,309999+05:30 sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget SSP1.OUT
2019-03-28T14:33:40,310001+05:30 sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
2019-03-28T14:33:40,310005+05:30 sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22

cc: @sathyap-chrome

sathyap-chrome, 2019-04-02:
The issue noted by @naveen-manohar is seen on power cycling stress test on CML.
This is seen 1 / 2 times in 100 power cycles.

lgirdwood, 2019-04-02:
@naveen-manohar @sathyap-chrome could someone attach dmesg and sof-logger output. Thanks.

naveen-manohar, 2019-04-02:
[failure_dmesg.txt](https://github.com/thesofproject/sof/files/3033510/failure_dmesg.txt)
[pass_dmesg.txt](https://github.com/thesofproject/sof/files/3033512/pass_dmesg.txt)

PFA dmesg between working and non-working scenario

tlauda, 2019-04-17:
@naveen-manohar @sathyap-chrome Could you verify with FW having this change: https://github.com/thesofproject/sof/commit/95b896b08484d5625d9603fad6b135a1e3b2331a? It should fix wrong IPC data reported in this issue.

jajanusz, 2019-04-30:
@naveen-manohar @sathyap-chrome ping

sathyap-chrome, 2019-05-06:
We failed with similar error.
The test was done using
Topology : cml_drop_0003 branch
Firmware : master

Signature 1:
2019-04-29T12:59:40,045482+05:30 sof-audio-pci 0000:00:1f.3: error: ipc error for 0x30010000 size 20
2019-04-29T12:59:40,045491+05:30 sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 27 name : SSP0.OUT stream SSP0-Codec reply -22
2019-04-29T12:59:40,045495+05:30 sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget SSP0.OUT

Signature 2:
2019-04-29T04:20:08,655337-07:00 sof-audio-pci 0000:00:1f.3: error: ipc error for 0x30010000 size 20
2019-04-29T04:20:08,655346-07:00 sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 27 name : HDA2.OUT stream iDisp3 reply -22
2019-04-29T04:20:08,655350-07:00 sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget HDA2.OUT
[dmesg_cold.txt](https://github.com/thesofproject/sof/files/3147526/dmesg_cold.txt)
[dmesg_warm.txt](https://github.com/thesofproject/sof/files/3147527/dmesg_warm.txt)

sathyap-chrome, 2019-05-06:
Update: Topology and kernel : SOF V0.3 for CML.

tlauda, 2019-06-06:
@naveen-manohar @sathyap-chrome Do we still observe such issues with kernel without MSI?

jajanusz, 2019-06-17:
@naveen-manohar @sathyap-chrome Closing as we don't have this error on RVPs and probably the problem is gone on your side on kernel w/o MSI. Feel free to reopen this issue, if you still have this problem.

closed 2019-06-17 by jajanusz