thesofproject/sof #767: DMIC should use SOF_ABI_VERSION instead of SOF_TKN_INTEL_DMIC_DRIVER_VERSION

ranj063, 2018-12-20:


singalsu, 2018-12-20:
I added it for extra extra safety in case the configuration selection logic would be updated. There's a simple criteria to pick a most preferred option of some decimator and clock dividers programming possibilities. Updated criteria could cause the mic clock frequency to change within allowed min/max limits with identical IPC. A change in bus clock frequencies could cause different product EMI/EMC approval challenges. The idea is that if such change is done into modes selection algorithm this driver version would be incremented.

We have now a working ABI major/minor system that covers the IPC changes so a major non-compatibility with parameters shouldn't happen. Maybe instead of blocking DMIC access with non-matching driver version token should cause warning messages sent to trace. That should trigger
re-tests for a device and if OK just match in topology file the value.

plbossart, 2018-12-20:
On 12/20/18 3:08 AM, Seppo Ingalsuo wrote:
> I added it for extra extra safety in case the configuration selection
> logic would be updated. There's a simple criteria to pick a most
> preferred option of some decimator and clock dividers [...]
Isn't this a generic issue though? If the IPC with a firmware component 
remains the same, but the implementation changes and produces different 
results, then is this really the same component? You could have the same 
problem with all acoustic processing, volume control, etc., and at that 
point it might be a better idea to have a module identifier rather than 
a version counter. You could even have someone substitute a generic 
module for another custom one, so I would assert using a version # is 
not future-proof. Maybe time to look at UUID?

singalsu, 2018-12-20:
Yes, I agree with your view and it's a generic issue. E.g. two totally different AEC implementations could share the same configuration parameters and have not the same performance.