thesofproject/sof #771: GLK: Audio stops playing while listening with any android "Online music apps" after connect 3.5mm Headset and play

naveen-manohar, 2018-12-21:
Description:
All audio output broken after attempting to play audio in Play Music.

What steps will reproduce the problem?
1. Enable ARC++
   a. Boot to ChromeOS(any sof enabled version is fine)
   b. In system tray, go to settings and in search bar, input ‘Playstore’ and enable it. Now DUT will support 
       ARC++ and playstore apps.
2. How to repro issue:
   a. Install any ARC++ music app [Google play music ,Youtube app, Gaana etc..]
   b. Start playing Audio
   c. Connect 3.5mm HS (Audio plays)
   d. Pause the Audio
   e. Wait 1 -2 munites and then press play --> Observe that from here Audio recovers only after reboot.

What is the expected output?
- Audio should work fine without any issue.
What do you see instead?
- Audio stops playing – from here Audio recovers only after reboot

[LOG.log](https://github.com/thesofproject/sof/files/2703459/LOG.log)

Debug Info: Need to debug Topology
ARC always tries to use 4-channel device.
Speaker and headphone device both reports they supports channel 2 to 8.

There is software DSP for speaker. The number of output channel is 2 in it, so CRAS opens two channel on speaker device.
On the other hand, headphone uses 4-channel, as requested by ARC.

plbossart, 2018-12-21:
> Debug Info: Need to debug Topology
> ARC always tries to use 4-channel device.
> Speaker and headphone device both reports they supports channel 2 to 8.
>
> There is software DSP for speaker. The number of [...]

Can you please try to play with 4 ch with aplay/speaker-test -Dhw:0. if it works then it's a Chrome issue not an SOF issue.

ranj063, 2018-12-21:
@plbossart @naveen-manohar this issue is specific to v0.55. FIxing the max channels to PIPELINE_CHANNELS fixes the issue. Already verified to work.

naveen-manohar, 2018-12-21:
Got the fix from Ranjani, many thanks to her.. This was a critical milestone blocker

closed 2018-12-21 by naveen-manohar