thesofproject/sof #773: PM D3 entry fixes for CNL, WHL, CML, ICL

Jocelyn-Li, 2018-12-22:
Make sure PM entry is completed prior to IPC.  Reference https://github.com/thesofproject/sof/pull/772 for APL fixes

RanderWang, 2018-12-24:
@Jocelyn-Li  I don't understand the idea of 772. And I discuss with other team member, they also don't understand it. We don't know whether kernel driver needs to be refined

lgirdwood, 2019-01-08:
@RanderWang this is for FW only. Please have a look at the PM shutdown sequence (PM ops and IPC ops) on APL and see if similar changes have to be made for other platforms.

lgirdwood, 2019-01-23:
@RanderWang ping, any questions ?

RanderWang, 2019-01-24:
Hi Liam,  sorry for late. I am busy with other issues.
One Question: I find there is no power_down.s on ICL. AS I know, ICL is almost the same as CNL in FW,
so could I copy from CNL and modify it for ICL? Thanks.

lgirdwood, 2019-01-24:
@RanderWang I think so, but @lbetlej will be able to confirm for sure.

closed 2019-01-24 by lgirdwood

lbetlej, 2019-01-25:
@RanderWang @lgirdwood - correct the power_down.s from CNL can be used for ICL . Note that I've just a bit cleaned up the file see PR#894 in case you want to copy it.

lbetlej, 2019-01-25:
@RanderWang  @lgirdwood - BTW> the driver shall make sure that is not accessing memory window (only IPC registers) during pm_ctx_save handling.

lgirdwood, 2019-01-25:
@ranj063 fyi, we cannot read mailbox after receiving PD IPC reply