thesofproject/sof #784: [Stress test]: IPC timed out after 967 times aplay on APL with PCM512x.

markyang, 2018-12-28:
**Summary:**
  IPC timed out after 967 times aplay on APL with PCM512x.
  sof-logger is empty. sof-logger -t log is attached below.
**dmesg:**
```
[ 3296.899229] sof-audio sof-audio: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 3296.899430] sof-audio sof-audio: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 3296.913495] sof-audio sof-audio: pcm: trigger stream 0 dir 0 cmd 1
[ 3296.913523] sof-audio sof-audio: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 3296.913808] sof-audio sof-audio: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 3298.633775] sof-audio sof-audio: pcm: trigger stream 0 dir 0 cmd 0
[ 3298.633810] sof-audio sof-audio: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 3298.937260] sof-audio sof-audio: error: ipc timed out for 0x60050000 size 0xc
[ 3298.937273] sof-audio sof-audio: status: fw entered - code 00000005
[ 3298.937418] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[ 3298.937423] sof-audio sof-audio: error: waking up any trace sleepers
[ 3298.937449]  Port5: ASoC: trigger FE failed -110
[ 3298.937502] sof-audio sof-audio: pcm: free stream 0 dir 0
[ 3299.241262] sof-audio sof-audio: error: ipc timed out for 0x60030000 size 0xc
[ 3299.241275] sof-audio sof-audio: status: fw entered - code 00000005
[ 3299.241421] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[ 3299.241426] sof-audio sof-audio: error: waking up any trace sleepers
[ 3299.256441] sof-audio sof-audio: pcm: close stream 0 dir 0
```

**Test steps:**
```
1. aplay-10000.sh
```

**Test env:**
sof master: a2e6f82e
kernel sof-dev: 62aa5698
tplg: sof-apl-pcm512x.tplg

**Log:**
[logger-t.txt](https://github.com/thesofproject/linux/files/2714422/logger-t.txt)
[aplay_error_967.log](https://github.com/thesofproject/linux/files/2714420/aplay_error_967.log)
[aplay-10000.sh.txt](https://github.com/thesofproject/linux/files/2714423/aplay-10000.sh.txt)

keyonjie, 2018-12-28:
good reporting, thanks, @markyang

lgirdwood, 2019-01-23:
@markyang I wonder if this is IPC related (since the FW has not crashed afaict). Do you have a similar stress test that can be run on same HW that changes a volume control value 10000 times.

dabekjakub, 2019-01-24:
@markyang can it be reproduced on Yorp ?

@wwittbrx please try to reproduce it in our test enviroment

lgirdwood, 2019-01-24:
@dabekjakub I have a feeling these similar IPC timeout issues we are seeing today are related to scheduling. i.e. the IPC reply is scheduled but not run since we don't see any exception.

dabekjakub, 2019-01-24:
@lgirdwood You are probably right and we are waiting for @bkokoszx to finish scheduler changes

markyang, 2019-01-25:
@lgirdwood yes, I got a similar IPC timeout after changing a volume control value about 1000 times.

markyang, 2019-01-25:
@dabekjakub sorry, I have no Yorp.

markyang, 2019-01-28:
**Summary:**  
w/ playback stream, volume change passed 10000 iterations

**Test steps:**
1. aplay -Dhw:0,0 -c2 -fS16_LE -vv -i music.wav  # Terminal 1
2. ./vol-change-10000.sh   # Terminal 2

**Test env:**
sof master: 1f5a9657
kernel sof-dev: 7b67e0bb
tplg: sof-apl-pcm512x.tplg

wenqingfu, 2019-01-29:
@markyang , can you please help to test with non-SMP configuration as @keqiaozhang did at [issue 780](https://github.com/thesofproject/sof/issues/780#issuecomment-458046254). I feel they are very likely to be the same. Thank you in advance for the testing!

markyang, 2019-01-29:
@wenqingfu  w/ playback stream, volume change passed 10000 iterations with the latest non-SMP firmware built by `./scripts/xtensa-build-all.sh -l -u -j apl`.

**Test steps:**
```
1. aplay -Dhw:0,0 -c2 -fS16_LE -vv -i /dev/zero  # in terminal 1
2. ./volume-change-loop.sh # in terminal 2, run 10000 times
```

**Test env:**
sof master: f9e5bba3
kernel sof-dev: 7b67e0bb
tplg: sof-apl-pcm512x.tplg

**Scripts**
[volume-change-loop.sh.txt](https://github.com/thesofproject/sof/files/2806705/volume-change-loop.sh.txt)

markyang, 2019-01-29:
**Summary:**
this issue can also be reproduced after 18 iterations on APL with PCM512x, with the latest non-SMP firmware built by `./scripts/xtensa-build-all.sh -l -u -j apl.`

**dmesg:**
```
[ 6593.140563] sof-audio sof-audio: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 6593.444448] sof-audio sof-audio: error: ipc timed out for 0x60050000 size 0xc
[ 6593.444469] sof-audio sof-audio: status: fw entered - code 00000005
[ 6593.444618] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[ 6593.444626] sof-audio sof-audio: error: waking up any trace sleepers
[ 6593.444648]  Port5: ASoC: trigger FE failed -110
[ 6593.444697] sof-audio sof-audio: pcm: free stream 0 dir 0
[ 6593.748429] sof-audio sof-audio: error: ipc timed out for 0x60030000 size 0xc
[ 6593.748451] sof-audio sof-audio: status: fw entered - code 00000005
[ 6593.748599] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[ 6593.748606] sof-audio sof-audio: error: waking up any trace sleepers
[ 6593.763584] sof-audio sof-audio: pcm: close stream 0 dir 0
[ 6596.580410] sof-audio sof-audio: error: ipc timed out for 0x40010000 size 0x4c
[ 6596.580432] sof-audio sof-audio: status: fw entered - code 00000005
[ 6596.580581] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[ 6596.580595] sof-audio sof-audio: error: ctx_save ipc error during suspend -110
[ 6596.580622] sof-audio sof-audio: FW Poll Status: reg=0x1010303 successful
[ 6596.581429] sof-audio sof-audio: FW Poll Status: reg=0x303 successful
[ 6596.581441] sof-audio sof-audio: DSP core(s) enabled? 0 : core_mask 3
```

**Test steps:**
```
1. ./aplay-loop.sh
======== aplay: 18 times ========
```

**Test env:**
sof master: f9e5bba3
kernel sof-dev: 7b67e0bb
tplg: sof-apl-pcm512x.tplg

**Log:**
[dmesg-aplay-loop-fail.txt](https://github.com/thesofproject/sof/files/2807080/dmesg-aplay-loop-fail.txt)
[logger-t.txt](https://github.com/thesofproject/sof/files/2807081/logger-t.txt)

markyang, 2019-01-30:
**Summary:**
This issue can also be reproduced on APL with PCM512x using the latest daily build.

**Test steps:**
```
$ sudo dmesg -C
$ ./aplay-loop.sh 1
======== aplay: 70 times ========
$ sudo dmesg -C
$ ./aplay-loop.sh 2
======== aplay: 43 times ========
$ sudo dmesg -C
$ ./aplay-loop.sh 4
======== aplay: 37 times ========
$ sudo dmesg -C
$ ./aplay-loop.sh 8
======== aplay: 5 times ========

```

**vi aplay-loop.sh:**
```
#!/bin/bash

function error_check(){
    local sof_error=$(dmesg | grep sof-audio | grep -E "error|failed|timed out|panic|oops")
    if [ ! -z "$sof_error" ]; then
        dmesg > dmesg-aplay-loop-fail.txt
        return 1
    fi
    return 0
}

function aplay_loop(){
    local sleep_sec=${1:-1}
    local times=${2:-10000}
    local i=0
    while [ $i -lt $times ]; do
        i=$((i + 1))
        printf "aplay: %d \r" $i
        aplay -Dhw:0,0 -c 2 -r 48000 -f s16_le -d 1 ~/Music/WAV_PCM_48000Hz_stereo_16bit.wav 2> /dev/null
        [ $sleep_sec -gt 0 ] && sleep $sleep_sec
        error_check
        [ $? -ne 0 ] && break
    done
    printf "======== aplay: %d times ========\n" $i
}

aplay_loop $@
```

**Test env:**
sof master: 365b132e
kernel sof-dev: 7b67e0bb
tplg: sof-apl-pcm512x.tplg

lgirdwood, 2019-01-30:
@markyang ok, just to confirm failure increases with delay between aplay ? I would expected the opposite.

markyang, 2019-01-31:
@lgirdwood  According to the following test results, it seems has nothing to do with  the delay between aplays:

**Test steps:**
```
1. sudo dmesg -C
2. ./aplay-loop.sh 1
======== aplay: 50 times ========

3. rmmod & modprobe sof-audio modules
4. sudo dmesg -C
5. ./aplay-loop.sh 8
======== aplay: 38 times ========

6. rmmod & modprobe sof-audio modules
7. sudo dmesg -C
8. ./aplay-loop.sh 2
======== aplay: 47 times ========

9. rmmod & modprobe sof-audio modules
10. sudo dmesg -C
11. ./aplay-loop.sh 4
======== aplay: 15 times ========

12. rmmod & modprobe sof-audio modules
13. sudo dmesg -C
14. ./aplay-loop.sh 16
======== aplay: 10 times ========

15. sudo reboot
16. sudo dmesg -C
17. ./aplay-loop.sh 4
======== aplay: 60 times ========

```

lgirdwood, 2019-01-31:
@markyang @wwittbrx delays above 4 seconds will cause the driver/DSP to enter D3, so this errors > 4 seconds are probably kernel bug, but delay < 4 seconds are FW related. @wwittbrx any update at your end ?

Jiangxinx, 2019-03-20:
Passed on daily test
Step:
1. aplay-10000.sh

Env:
sof master: 6e88bdf1
kernel sof-dev: 95b2fab5
tplg: sof-apl-pcm512x.tplg

closed 2019-03-20 by lgirdwood