thesofproject/sof #788: [PM] playback on HDMI can't continue after suspend/resume

keyonjie, 2019-01-03:
How to reproduce:
1. run aplay on HDMI output;
2. run 'rtcwake -m mem -s 5' on another terminal;

Expected result: HDMI playback continue after resume from suspend;
Actual result: the playback of HDMI will stop.

This will actually block the HDMI playback suspend/resume stress test on all platforms.

lgirdwood, 2019-01-08:
@keyonjie logs please.

markyang, 2019-01-14:
**Summary:**
  This issue can be reproduced on APL with PCM512x using the latest build.
  sof-logger is empty. sof-logger -t log is attached below.

**dmesg:**
```
[20790.943963] OOM killer enabled.
[20790.943965] Restarting tasks ... 
[20790.946168] sof-audio sof-audio: pcm: trigger stream 2 dir 0 cmd 6
[20790.946175] sof-audio sof-audio: PCM: resume stream 2 dir 0 DMA position 0
[20790.946179] sof-audio sof-audio: pcm: hw params stream 2 dir 0
[20790.946186] sof-audio sof-audio: generating page table for 00000000776bf9ad size 0xff00 pages 16
[20790.946211] sof-audio sof-audio: period_bytes:0xfc0
[20790.946214] sof-audio sof-audio: periods:16
[20790.946225] sof-audio sof-audio: stream_tag 1
[20790.946566] done.
[20790.946594] video LNXVIDEO:00: Restoring backlight state
[20790.946603] PM: suspend exit
[20790.947650] sof-audio sof-audio: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[20790.948178] sof-audio sof-audio: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[20790.948475] sof-audio sof-audio: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[20790.948671] sof-audio sof-audio: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[20791.099652] ata1: SATA link down (SStatus 4 SControl 300)
[20791.099680] ata2: SATA link down (SStatus 4 SControl 300)
[20792.060856] sof-audio sof-audio: pcm: trigger stream 2 dir 0 cmd 0
[20792.060895] sof-audio sof-audio: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[20792.364365] sof-audio sof-audio: error: ipc timed out for 0x60050000 size 0xc
[20792.364387] sof-audio sof-audio: status: fw entered - code 00000005
[20792.364536] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[20792.364543] sof-audio sof-audio: error: waking up any trace sleepers
[20792.364571]  HDMI2: ASoC: trigger FE failed -110
[20792.364611] sof-audio sof-audio: pcm: free stream 2 dir 0
[20792.668392] sof-audio sof-audio: error: ipc timed out for 0x60030000 size 0xc
[20792.668413] sof-audio sof-audio: status: fw entered - code 00000005
[20792.668561] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[20792.668568] sof-audio sof-audio: error: waking up any trace sleepers
[20792.668799] sof-audio sof-audio: pcm: close stream 2 dir 0
[20793.176053] r8169 0000:03:00.0 enp3s0: Link is Up - 100Mbps/Full - flow control rx/tx
[20798.244362] sof-audio sof-audio: error: ipc timed out for 0x40010000 size 0x4c
[20798.244383] sof-audio sof-audio: status: fw entered - code 00000005
[20798.244532] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[20798.244549] sof-audio sof-audio: error: ctx_save ipc error during suspend -110
[20798.244578] sof-audio sof-audio: FW Poll Status: reg=0x1010303 successful
[20798.245376] sof-audio sof-audio: FW Poll Status: reg=0x303 successful
[20798.245387] sof-audio sof-audio: DSP core(s) enabled? 0 : core_mask 3
[20798.245448] sof-audio sof-audio: Debug PCIR: 00000010 at  00000044
```

**Test steps:**
```
1. aplay -Dhw:0,2 -c2 -r48000 -fS16_LE -vv -i music.wav
2. sudo rtcwake -m mem -s 2
```

**Test env:**
sof master: 872cefe3
kernel sof-dev: 2ac82ee9
tplg: sof-apl-pcm512x.tplg

**Log:**
[dmesg-apl.txt](https://github.com/thesofproject/sof/files/2754494/dmesg-apl.txt)
[logger-t.txt](https://github.com/thesofproject/sof/files/2754495/logger-t.txt)

lgirdwood, 2019-01-14:
@keyonjie can you check that all the 
```
while(hw_reg_value() == val) {
...
};
```

Have a timeout clause that will break. I have a suspicion we probably have one ion the HDA logic since the stream appears to start (but then ALSA stops it as it's position is not updating).

keyonjie, 2019-01-15:
@lgirdwood I believe we have some HDA DMA logic issue here, I will take looking into it after Xrun cleanup finished.

lgirdwood, 2019-01-16:
@dabekjakub your understanding is the same as mine, I would expect streams to be stopped prior to power down.

plbossart, 2019-01-16:
On 1/16/19 6:07 AM, Liam Girdwood wrote:
>
> @dabekjakub <https://github.com/dabekjakub> your understanding is the 
> same as mine, I would expect streams to be stopped prior to power down.
>
ALSA has support for TRIGGER_SUSPEND and TRIGGER_RESUME, it's perfectly 
legit to close the lid and suspend while audio is playing, and resume 
playing later on.

That said, ChromeOS will stop the streams and restart them explicitly (I 
believe there is a shortcut to have the same behavior as plain vanilla 
ALSA, Curtis was looking into it).

So it depends which environment you are testing in.

lgirdwood, 2019-01-17:
@plbossart my expectation is that TRIGGER_SUSPEND/RESUME should STOP/START the stream from a FW perspective as the FW does not retain any context when the lid is shut. Iirc there is also a flag for each PCM that lets ALSA core know if TRIGGER_SUSPEND/RESUME is supported (not sure if we are setting this in topology).

dabekjakub, 2019-01-17:
So as You said , FW knows nothing of external conditions and in curent implementation we should just be told to stop any streams then to go to power down.  So this looks rather like kernel/alsa issue.

lgirdwood, 2019-01-17:
@dabekjakub do you have any logs here from FW side as my understanding is that this should work on Chrome as @plbossart says since Chrome will send a TRIGGER_STOP and TRIGGER_START which should mean the FW will work as expected.

plbossart, 2019-01-17:
@lgirdwood in theory the IP can support suspend and restart from where the DMA position was, so your assertion is only true because we don't support what the hardware could do :-)

dabekjakub, 2019-01-18:
We have no logs from this since on our platforms we have reproduction of
https://github.com/thesofproject/linux/issues/489

I was just throwing logical assumption out in the open that if we have tests that confirm working scenario of TRIGGER_STOP > SUSPEND > RESUME > TRIGGER_START (and we do). Then only issue possible would be kernel side.

However I looked into provided logs 

> [20792.060895] sof-audio sof-audio: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
> [20792.364365] sof-audio sof-audio: error: ipc timed out for 0x60050000 size 0xc
> [20792.364387] sof-audio sof-audio: [...]

This part should catch our attention. Issue here is that stream stopping ipc is getting timed out.

and that is not all since in provided FW logs we have stop and free ipcs handled but they are reported as timed out on kernel side.

@keyonjie Could You confirm that on the kernel side You have no answer when ipc timeouts here ? Because I can see ipcs handled in logs.

keyonjie, 2019-01-21:
@dabekjakub as we talked in the meeting, I believe FW error first happen at trigger start before this trigger stop, here we see the trigger stop IPC timeout as FW can't respond to IPC anymore.

keyonjie, 2019-01-21:
BTW, please use generic hda tplg and machine driver on UP2, it's very easy(100%) to reproduce this issue.

mengdonglin, 2019-01-28:
This seems to be a duplicate of the generic issue https://github.com/thesofproject/linux/issues/489 on multiple platforms.

michalgrodzicki, 2019-02-13:
> This seems to be a duplicate of the generic issue [thesofproject/linux#489](https://github.com/thesofproject/linux/issues/489) on multiple platforms.

@mengdonglin - if it's a duplicate who should be owner of this issue? @RanderWang ?

mengdonglin, 2019-02-25:
@dabekjakub @michalgrodzicki   I think @ranj063 has between working on this before her vacation. And this is a duplicate of [thesofproject/linux#489](https://github.com/thesofproject/linux/issues/489)

mengdonglin, 2019-03-04:
Closing because it seems a driver issue, will follow up by thesofproject/linux#489

closed 2019-03-04 by mengdonglin