thesofproject/sof #791: IIR and FIR EQ tools should receive their ABI from C headers

singalsu, 2019-01-04:
Currently there's parallel  Matlab m-language subset headers implementation since the languages are not source compatible. E.g. the sof-eqctl binary tool could provide in Matlab syntax the ABI information to create up-to-date configuration data for EQs. The compatibility breaks from overall SOF ABI maintenance could be avoided this way.

jajanusz, 2019-06-06:
@singalsu What do you mean by `should receive their ABI`? Do you need some struct defs in matlab scripts that are in C and you have to duplicate them in matlab?

singalsu, 2019-06-06:
Yes I've transformed the needed headers into Matlab code to create without manual work e.g. the ascii hex blob to include from topology or the ascii decimal blob to pass with sof-ctl. The solution for making the blob in Matlab is ugly and not good for maintenance.

I think some C helper tool would be more elegant since it could use the SOF ABI headers as such and insert the pretty stable EQs configuration blobs created by Matlab.

lgirdwood, 2019-06-06:
@singalsu we probably need a small C tool that includes the ABI header that can be used by your matlab code to insert the header with the correct ABI for the matlab tool.

singalsu, 2019-06-07:
@lgirdwood Yes that looks like the best approach. I think I should be able to do it in mid July.

lgirdwood, 2019-06-10:
@singalsu great, marked it for v1.4

lgirdwood, 2019-08-30:
@singalsu is this now merged, can we close ?

singalsu, 2019-09-02:
@lgirdwood Sorry, no it's not yet done. Could the feature to return ABI header data e.g. in simple comma separated numbers format be added to sof-ctl executable?

singalsu, 2019-09-02:
@lgirdwood I created PR #1792 that implements this enhancement.

singalsu, 2019-09-13:
The PR #1792 is merged now so we can close this!

closed 2019-09-13 by singalsu