thesofproject/sof #795: SOF doesnt handle XRUNS correctly

ranj063, 2019-01-08:
Creating a github issue based on the partner issue created by @cujomalainey 
https://partnerissuetracker.corp.google.com/u/2/issues/121161381

Steps to reproduce: 
Run speaker-test -D hw:0,0 -t sine -c 2
Suspend and resume the task repeatedly (using C-z and fg)
Eventually you will get an iteration where the sine wave it noticeably no longer playing the correct tune and sounds distorted.

keqiaozhang, 2019-01-08:
@ranj063 
I cannot reproduce this issue on one Bobba.
I tested the speaker-test on speaker, sound is good, no obvious noise, then I tested it on headset and recorded the sound, the waveform looks good also, no distortion.
Screen shot and the wav file are attached, please refer.
[image]
[record.zip](https://github.com/thesofproject/sof/files/2735421/record.zip)

Test recipe:
ChromeOS: R72-11316.61.0
Kernel: google release branch/release-R72-11316.B-chromeos-4.14 + your fix/6b0e12c(ASoc: SOF: use optimized register polling)

ranj063, 2019-01-08:
@cujomalainey looks like the issue cannot be reproduced on our boards. Could you please verify with the latest R72 build?

ranj063, 2019-01-08:
@keqiaozhang Thanks for testing. I'm closing this one as it has been confirmed to be fixed

closed 2019-01-08 by ranj063