thesofproject/sof #798: tools/coredumper: need readme or documentation about how to use it

xiulipan, 2019-01-08:
No usage to tools/coredumper. Need some simple doc about how to use it and a example would be better.
What should be the input file and what is the legal exception output.

mwierzbix, 2019-01-09:
There is usage:
```
./sof-coredump-reader.py --help                                                                                                                                                                                        
usage: sof-coredump-reader.py [-h] [-a ARCH] [-c] [-l COLUMNCOUNT] [-v]
                              (--stdout | -o OUTFILE)
                              [--stdin | -i INFILE]

Tool for processing FW stack dumps. In verbose mode it prints DSP registers,
and function call addresses in stack up to that which caused core dump. It
then prints either to file or to stdin all gdb commands unwrapping those
function call addresses to function calls in human readable format.

optional arguments:
  -h, --help            show this help message and exit
  -a ARCH, --arch ARCH  determine architecture of dump file; valid archs are:
                        LE32bit, LE64bit.
  -c, --colour          set output to be colourful!
  -l COLUMNCOUNT, --columncount COLUMNCOUNT
                        set how many colums to group the output in
  -v, --verbose         increase output verbosity
  --stdout              output is to stdout
  -o OUTFILE, --outfile OUTFILE
                        output is to FILE
  --stdin               input is from stdin
  -i INFILE, --infile INFILE
                        path to sys dump bin
```
And there is exemplatory snippet:
```
cat ./sof-coredump-to-gdb.sh                                                                                                                                                                                            !2123
#!/bin/bash
# Exemplatory usage of sof-coredump-reader.py ($reader_name)
# We read from dump file into sof-coredump-reader.py, then we pipe its output
# to xt-gdb, which operates on given elf-file.

THIS_SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null && pwd )"
elf="${1}"
dump="${2}"

reader_name="sof-coredump-reader.py" # in case it is changed
reader_output="$(${THIS_SCRIPT_DIR}/${reader_name} -vc -i ${dump} --stdout -l 4)"
reader_result="$?" # if $reader_name script fails, running xt-gdb is pointless
if [[ ${reader_result} -ne 0 ]] ; then
        echo "${reader_name} failed!"
        exit ${reader_result}
else
        xt-gdb "${elf}" < <(echo "${reader_output}")
fi
```

xiulipan, 2019-01-10:
@mwierzbix 
LGTM, can you also add this as a README inside tools/coredumper folder?
And a example maybe better for people to confirm they have a GOOD run.

lgirdwood, 2019-01-23:
@dabekjakub I think we need this reassigned as @mwierzbix is working on something else atm.

dabekjakub, 2019-01-24:
@lgirdwood assigned @akloniex / myself  however this is low priority - we will add it to docs as soon as someone has free bandwitch

lgirdwood, 2019-01-24:
@dabekjakub thanks, Btw I want to make this mandatory in any FW crash issues to help developers.

abonislawski, 2019-03-28:
Please check PR #1195 about readme
+ PR 101 in docs:
https://github.com/thesofproject/sof-docs/pull/101

closed 2019-04-02 by abonislawski