thesofproject/sof #800: DMIC test fail when audio could not play

naveen-manohar, 2019-01-08:
Codebase: ChromeOS-firmware-R72-11316.64.0
Topology - topology: https://github.com/ranj063/soft-1/tree/sof-v0.55
DSP Firmware - https://github.com/ranj063/sof/tree/sof-v0.55

The test item "SpeakerDMic" is to loop from speaker to digital microphone. You can refer to the factory test script as below.
https://chromium.googlesource.com/chromiumos/platform/factory/+/master/py/test/pytests/audio_loop.py

In this test, it will run the audiofuntest to test the function, something like below:
audiofuntest -r 48000 -i hw:0,0 -o hw:0,0 -l 5 -a 0

2019-01-06T21:30:36.373054-08:00 INFO kernel: [  329.372343] sof-audio sof-audio:  Firmware info: version 1.1-13be5 build 32 on Dec 24 2018:19:43:24
2019-01-06T21:30:36.375043-08:00 INFO kernel: [  329.374091] sof-audio sof-audio: firmware boot complete
2019-01-06T21:30:43.049121-08:00 ERR kernel: [  336.048321] sof-audio sof-audio: error: ipc timed out for 0x60040000 size 0xc
2019-01-06T21:30:43.049165-08:00 ERR kernel: [  336.048355] sof-audio sof-audio: status: fw entered - code 00000005
2019-01-06T21:30:43.049173-08:00 ERR kernel: [  336.048495] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
2019-01-06T21:30:43.049176-08:00 ERR kernel: [  336.048502] sof-audio sof-audio: error: waking up any trace sleepers
2019-01-06T21:30:43.049182-08:00 ERR kernel: [  336.048539]  Speakers: ASoC: trigger FE failed -110
2019-01-06T21:30:43.049184-08:00 ERR kernel: [  336.048864] sof-audio sof-audio: error: rx list empty but received 0x0
2019-01-06T21:30:43.049186-08:00 ERR kernel: [  336.048876] sof-audio sof-audio: error: can't find message header 0x60040000
2019-01-06T21:30:43.049188-08:00 ERR kernel: [  336.048935] sof-audio sof-audio: XRUN for unknown stream, msg_id 0
2019-01-06T21:30:43.052043-08:00 ERR kernel: [  336.051095] sof-audio sof-audio: XRUN for unknown stream, msg_id 0
2019-01-06T21:30:43.052066-08:00 ERR kernel: [  336.051218] sof-audio sof-audio: XRUN for unknown stream, msg_id 0
2019-01-06T21:30:44.175639-08:00 INFO goofy[1236]: Test main_fleex:SMT.SpeakerDMicManual.SpeakerDMic_16 (03e10db4-ec6c-4fdf-8e85-e04c0fe8ce68) completed: FAILED (Under output volume 10; Failed to parse audiofuntest output#012Traceback (most recent call last):

wenqingfu, 2019-01-09:
same error as #743 ?

xiulipan, 2019-01-09:
@keyonjie 
```
ERR kernel: [  336.048321] sof-audio sof-audio: error: ipc timed out for 0x60040000 size 0xc
ERR kernel: [  336.048355] sof-audio sof-audio: status: fw entered - code 00000005
ERR kernel: [  336.048495] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
ERR kernel: [  336.048502] sof-audio sof-audio: error: waking up any trace sleepers
ERR kernel: [  336.048539]  Speakers: ASoC: trigger FE failed -110
ERR kernel: [  336.048864] sof-audio sof-audio: error: rx list empty but received 0x0
ERR kernel: [  336.048876] sof-audio sof-audio: error: can't find message header 0x60040000
```
It seems the IPC is handled, but not in time here.

@naveen-manohar 
will try to reproduce it and have some feedback.

keyonjie, 2019-01-09:
we need to duplicate it on our side to root cause it, but simultaneous pipelines stress test fails in v0.55 is our known issue.
Shall we align on that we will focus on fixing it in PV release, but not v0.55?

keqiaozhang, 2019-01-09:
I used my test script to test dmic capture and speaker playback at same time, it passed 1000 iterations without errors.
CMD:
arecord -Dhw:0,99 -f dat -c 4 /dev/null & aplay -D hw:0,0 -f dat -c 2 /dev/zero & sleep 3; pkill aplay; pkill arecord 

I also tried to test it with audiofuntest, but failed with "Failed to exec client: No such file or directory"
```
localhost /usr/local/factory # audiofuntest -P -r 48000 hw:0,0 -C 2 -R -r 48000 -c 4 hw:0,99 -l 5
Config values.
        Speaker active channels: 0, 1
        Mic active channels: 0, 1, 2, 3
        Allowed delay: 1.5000(s)
        FFT size: 2048
        Confidence threshold: 3.0000
        Match window size: 7
        Player parameter: -r
        Player FIFO name:
        Recorder parameter: -r
        Recorder FIFO name:
        Sample format: s16
        Sample rate: 64000
        Number of Microphone channels: 4
        Number of Speaker channels: 2
        Number of test rounds: 10
        Tone length: 5.0000(s)
        Volume gain: 50
Failed to exec client: No such file or directory
Failed to exec client: No such file or directory
Killed
```
@naveen-manohar 
What have I missed? Could you share same docs about how to setup the env and run audiofuntest on DUT?

dabekjakub, 2019-01-09:
I see that topology and FW are from @ranj063 repo. 
Does this reproduce on FW master branch ?
@naveen-manohar

naveen-manohar, 2019-01-10:
@ranj063  Yes i'm tracking the same with ODMs
Shall update soon they give a positive nod

naveen-manohar, 2019-01-10:
Got customer's reply - "SOF v1.0 will not be backported to the factory branch. We need to make sure this gets fixed on the factory branch."

wenqingfu, 2019-01-10:
@dabekjakub  @keqiaozhang , you need to talk with @keyonjie if it's the same problem he run into in [this comment ](https://github.com/thesofproject/sof/issues/743#issuecomment-450275036). Before that, he reported 1000+ successful iterations of a similar test scenario as well. 

Good luck!

wenqingfu, 2019-01-11:
> This issue is easier to reproduce if three pipelines are running at same time.

interesting finding, thanks! can you please put down the steps you used for creating "three pipelines"?  What about  more than three pipelines? Would it make the issue reproduction even easier?

wenqingfu, 2019-01-11:
scouted symptom report in #784 shows similar error pattern, too.

naveen-manohar, 2019-01-15:
@keyonjie Do we have a fix for this ? Please help provide an ETA.
jfyi, If the solution could not be landed in before EOW, we may need to consider to raise up a waiver request with customer.

keyonjie, 2019-01-15:
@naveen-manohar we have just finished back-porting those patches to sof-v0.55 and finished sanity test here, stress test just triggered, in case you can help trigger test on your side also:
https://github.com/keyonjie/sof/tree/glk-004-fix

naveen-manohar, 2019-01-16:
Hi @keyonjie,
Validation Team tried the fix shared by you for 50 iterations and wasn't able to hit the issue with factory test - SMT.SpeakerDMic
Also basic sanity passed. Thanks

keyonjie, 2019-01-16:
> Hi @keyonjie,
> We tried the fix shared by you for 50 iterations and wasn't able to hit the issue with factory test - SMT.SpeakerDMic
> But Basic Audio Sanity Failed, Both Speaker & Headset playback observes bursting noise (more frequently observed during audio playback soon after boot)

@naveen-manohar can you confirm if this noise issue is regression after apply my new fixes?

naveen-manohar, 2019-01-16:
@keyon yes we did re-verify and bursting/crackling issue is reproduced only the firmware change.
Without the firmware change, default cpfe works fine.

keyonjie, 2019-01-16:
> @Keyon yes we did re-verify and bursting/crackling issue is reproduced only the firmware change.
> Without the firmware change, default cpfe works fine.

OK thanks for the information, I figured out the fixes in UP^2 with nocodec, so let me try to figure out which patch is the convict, I suspect this one at my first think:

pipeline: switch schedule_task_idle to schedule_task

can you do a quick try if removing this commit make difference?

I will do debugging here tomorrow.

naveen-manohar, 2019-01-17:
@keqiaozhang  Steps we tried:
1. Boot to R72-11316.75.0 OS  & replaced with dsp firmware /lib/firmware/intel/sof-glk.ri with below md5sum and reboot - edd33e76454176f45151fc842ad2c7d7 
2. Observation: Speaker audio playback to notice bursting noise soon after boot

keqiaozhang, 2019-01-17:
@naveen-manohar 
Now I can reproduce this bursting noise issue, but this issue only happens with browser, no such noise with aplay.
After bisection, we found that this noise is caused by Keyon's new fix(debug: hda-dma: set timeout for preload). Now he is working on it.

naveen-manohar, 2019-01-17:
Thanks @keyonjie can we get an ETA for #856 ?
This will help us to communicate better with our customers.

keqiaozhang, 2019-01-30:
I cannot reproduce this issue any more.
@naveen-manohar 
Please help to close this issue if you are satisfied with it. Thanks!

naveen-manohar, 2019-01-30:
Thanks a Ton, fix has been validated & we can close this bug

closed 2019-01-30 by naveen-manohar