thesofproject/sof #802: [CML] playback/capture has noise on CometLake, with analog HDMI/DP, still found on WHL

zhuyingjiang, 2019-01-09:
On the CML RVP board (WHL RVP reworked add rt5682), the playback/capture for Analog,HDMI, DMIC all have noise.
But for loopback mode, there is no noise.

zhuyingjiang, 2019-01-09:
sine wave in loopbakc mode:
[image]

lgirdwood, 2019-01-09:
@zhuyingjiang you need to zoom in on the audacticy screenshot and attach a small 5 second wav with the noise. It does sound like codec integration domain given that loopback works. @bardliao any comment ?

plbossart, 2019-01-09:
@zhuyingjiang can you please clarify what the platform is? if you've reworked the platform to add an I2S codec then HDMI and "analog" have nothing in common and HDMI should work regardless of the codec.
Also please add logs so that we can check what is happening during playback.

bardliao, 2019-01-10:
@lgirdwood I think what @zhuyingjiang said "loopback mode" is to short I2S TX and RX pin physically. So it is not on codec domain. @zhuyingjiang as Liam said could you attach 5 second wav with the noise? We need the noise wav for all rt5682, HDMI, and DMIC. Let's see what kinds of noise they are and check if they look the same first. Also. please attach your topology file and show the command you tested.

zhuyingjiang, 2019-01-10:
> @zhuyingjiang can you please clarify what the platform is? if you've reworked the platform to add an I2S codec then HDMI and "analog" have nothing in common and HDMI should work regardless of the codec.
@plbossart @lgirdwood  @bardliao The platform is a reworked WHL RVP board plus a rt5682 board. The noise from HDMI is attached. I will also grasp noise from SSP and DMIC

zhuyingjiang, 2019-01-10:
[image]
[cml-evaluate-board-dmic-rec.zip](https://github.com/thesofproject/sof/files/2744550/cml-evaluate-board-dmic-rec.zip)
The wav record from DMIC, there is a noise at the end, as picture showed.

lgirdwood, 2019-01-10:
@zhuyingjiang the HDMI playback shows no digital issues (i.e. missing or repeated samples) but the waveform is not smooth. Does legacy HDMI playback to the same monitor give the same or better output ? 
What about HDA analog audio ? Does it have the same noise ?
[image]

zhuyingjiang, 2019-01-10:
@lgirdwood I have checked the same monitor with UP2, the waveform is smooth, And I will still check with legacy HDMI, HDA analog audio still have noise. with attach that tomorrow.

zhuyingjiang, 2019-01-11:
@lgirdwood  I have tried multiple device and case, the result is as follows:
WHL RVP with HDA driver => no noise
WHL RVP with SOF and sof-hda-generic.tplg => no noise
WHL reworked with rt5682 and sof-cnl-rt5682.tplg => noise
WHL reworked with rt5682 and sof-hda-generic.tplg => noise
WHL reworked with HDA driver => audio can't work because the rework.
Up2 with sof-apl-pcm512x.tplg with the same cable and monitor => no noise
So deduce here the noise may come from the manual rework, I will go on verify this when the real cometLake board comes.

lgirdwood, 2019-01-11:
@zhuyingjiang ok, good work - probably noise from rework maybe jitter....

zhuyingjiang, 2019-01-18:
> Update status. I don't see HDMI noise with my monitor. Probably we can ignore HDMI noise atm. For rt5682 playback noise issue, it is root caused to unsynchronized clock. rt5682 needs 24.576MHz for sysclk, but its PLL function can't generated an exactly 24.576MHz sysclk from 24MHz MCLK. So we need to enable rt5682's ASRC function.

@bardliao Yes, the noise from rt5682 should because of the clock. and have proved that no noise of HDMI/DMIC

lgirdwood, 2019-01-18:
@bardliao @zhuyingjiang is there any PR to be merged before we close ?

bardliao, 2019-01-18:
@lgirdwood Yes, I just updated https://github.com/thesofproject/linux/pull/553 with  rt5682_sel_asrc_clk_src() added. And no other PR needed so far.

lgirdwood, 2019-01-18:
@bardliao ok, thanks. Please close when it's merged.

closed 2019-01-21 by bardliao