thesofproject/sof #811: [CML] when playback and capture simultaneously, and stop playback by Ctrl+C, the capture will also stop

zhuyingjiang, 2019-01-10:
On the CML evaluation board(WHL RVP reworked with audio codec rt5682), when  playback and capture simultaneously, and stop playback by Ctrl+C, the capture will also exit.
Can be reproduce 100%.

keyonjie, 2019-01-10:
which playback/capture dais are you running? ssp, hda, dmic, or hdmi?

zhuyingjiang, 2019-01-10:
@keyonjie SSP

lgirdwood, 2019-01-10:
@zhuyingjiang we are missing logs etc so no one can help.

zhuyingjiang, 2019-01-10:
@lgirdwood I will attach them tomorrow.

zhuyingjiang, 2019-01-11:
[sim-play-cap-log.txt](https://github.com/thesofproject/sof/files/2749367/sim-play-cap-log.txt)
[cml-rt5682-aplay.txt](https://github.com/thesofproject/sof/files/2749368/cml-rt5682-aplay.txt)
[cml-rt5682-arecord-simultaneously.txt](https://github.com/thesofproject/sof/files/2749369/cml-rt5682-arecord-simultaneously.txt)
log of the issue:
1 aplay -D hw:0,0 48000-16bit-stereo.wav -vvvv
2 the same time arecord -D hw:0,0 -f s32_le -c 2 -r 48000 5682rec.wav -vvvvv
3 then Ctrl+c in the aplay console
(if Ctrl+c in the arecord console, the aplay still will exit)

lgirdwood, 2019-01-11:
@zhuyingjiang what about FW logs ?

```
[ 1755.570939] sof-audio sof-audio: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1755.571224] sof-audio sof-audio: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1755.571246] sof-audio sof-audio: ipc rx done: 0x600a0000: GLB_STREAM_MSG: POSITION
[ 1755.571388] sof-audio sof-audio: pcm: free stream 0 dir 0
[ 1755.571425] sof-audio sof-audio: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1755.571611] sof-audio sof-audio: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1755.571763] sof-audio sof-audio: pcm: close stream 0 dir 0
[ 1756.060621] sof-audio sof-audio: pcm: trigger stream 0 dir 1 cmd 0
[ 1756.060672] sof-audio sof-audio: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1756.061021] sof-audio sof-audio: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1756.061064] sof-audio sof-audio: pcm: free stream 0 dir 1
[ 1756.061093] sof-audio sof-audio: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1756.061304] sof-audio sof-audio: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1756.061515] sof-audio sof-audio: pcm: close stream 0 dir 1
[ 1756.260600] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1756.260687] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1756.760552] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1756.760640] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
```
This looks like the second stream has also stopped in the FW this then causes the driver to stop the stream, via IPC is it does not see stream movement within 100ms.

markyang, 2019-01-22:
**Summary:**  
  This issue cannot be reproduced on APL-nocodec, BYT-nocodec and BYT w/ ALC5651.

**Test steps:**
```
1. aplay -Dhw:0,0 -c2 -fS16_LE -vv -i ~/48000Hz_stereo_16bit.wav  # Terminal 1
2. arecord -Dhw:0,0 -c2 -fS16_LE -vv -i ~/test.wav  # Terminal 2
3. Ctrl+C  # Terminal 1 or 2
```

**Test env:**
sof master: 03ababfb
kernel sof-dev: a5dd3186
tplg: sof-apl-nocodec.tplg
tplg: test-ssp0-mclk-0-I2S-volume-s16le-s16le-48k-24000k-codec.tplg
tplg: test-ssp0-mclk-0-I2S-volume-s16le-s16le-48k-24000k-nocodec.tplg

lgirdwood, 2019-01-22:
thanks @markyang for testing. @zhuyingjiang so it's CML only issue. Can you retry playback on HDA and capture on SSP.  If this works then it could be DMA related.

bardliao, 2019-01-22:
@lgirdwood playback on HDA and capture on SSP works. I also tried playback on SSP and capture on DMIC and it didn't work.

lgirdwood, 2019-01-22:
@bardliao @zhuyingjiang ok, both DMIC and SSP share the DW-DMA whilst HDA uses a different DMA engine. The issue here seems to be changes in DW-DMA implementation for CML. It may be worth putting in some extra debug to dw-dma.c and see the DMA engine status during the issue.

bardliao, 2019-01-24:
Thanks @lgirdwood, #871 solve the issue.

jajanusz, 2019-01-29:
@bardliao @zhuyingjiang 
Can we close it?

bardliao, 2019-01-29:
@jajanusz Yes. Thanks.

closed 2019-01-29 by bardliao