thesofproject/sof #837: qume boot status for skl, kbl and cnl can not be judged

xiulipan, 2019-01-14:
[qemudump.zip](https://github.com/thesofproject/sof/files/2754594/qemudump.zip)

Upload the apl/bxt as good example and skl, kbl, cnl.
```bash
dump-apl.txt  dump-kbl.txt  qemu-bridge-bxt-hp-sram  qemu-bridge-kbl-hp-sram
dump-cnl.txt  dump-skl.txt  qemu-bridge-cnl-hp-sram  qemu-bridge-skl-hp-sram
```

The success APL had right IPC ready message in outbox
```bash
$ hexdump qemu-bridge-bxt-hp-sram |grep "006c 0000 0000 7000"
000f000 006c 0000 0000 7000 0000 0000 0000 0000
```

The three problem device have different behavior
CNL have no IPC ready at all
SKL and KBL should have IPC ready at offset 0xF000 as APL do I think, but they seems to have
the IPC ready message at offset 0x00322e8?
```bash
$ hexdump qemu-bridge-cnl-hp-sram |grep "006c 0000 0000 7000"
$ hexdump qemu-bridge-skl-hp-sram |grep "006c 0000 0000 7000"
00322e0 fffe ffff 0001 0000 006c 0000 0000 7000
$ hexdump qemu-bridge-kbl-hp-sram |grep "006c 0000 0000 7000"
00322e0 fffe ffff 0001 0000 006c 0000 0000 7000
```

xiulipan, 2019-01-14:
@lgirdwood Please have a look for the QEMU boot. If I search the right bridge files?

lgirdwood, 2019-01-14:
Cmd line shlould include ROM
```
./xtensa-host.sh apl -k ../sof.git/src/arch/xtensa/sof-apl.ri -r ../sof.git/src/arch/xtensa/rom-apl.bin
```

On APL
```
hexdump -C /dev/shm/qemu-bridge-hp-sram-mem | grep -A 10 "6c 00 00 00 00 00 00 70"
0000f000  6c 00 00 00 00 00 00 70  00 00 00 00 00 00 00 00  |l......p........|
0000f010  00 00 00 00 00 00 00 00  3c 00 00 00 01 00 01 00  |........<.......|
0000f020  00 00 24 00 4a 61 6e 20  31 31 20 32 30 31 39 00  |..$.Jan 11 2019.|
0000f030  31 35 3a 34 31 3a 34 35  00 00 64 64 65 32 30 00  |15:41:45..dde20.|
0000f040  00 00 00 03 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0000f050  00 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
0000f060  00 00 00 00 00 00 00 00  00 00 00 00 b8 00 00 00  |................|
0000f070  00 00 00 70 01 00 00 00  07 00 00 00 00 00 00 00  |...p............|
0000f080  05 00 00 00 00 00 00 00  00 00 00 00 00 10 00 00  |................|
0000f090  00 00 00 00 00 00 00 00  01 00 00 00 00 00 00 00  |................|
0000f0a0  00 00 00 00 00 10 00 00  00 10 00 00 00 00 00 00  |................|
```
on CNL
```
hexdump -C /dev/shm/qemu-bridge-hp-sram-mem | grep -A 10 "6c 00 00 00 00 00 00 70"
00005000  6c 00 00 00 00 00 00 70  00 00 00 00 00 00 00 00  |l......p........|
00005010  00 00 00 00 00 00 00 00  3c 00 00 00 01 00 01 00  |........<.......|
00005020  00 00 25 00 4a 61 6e 20  31 31 20 32 30 31 39 00  |..%.Jan 11 2019.|
00005030  31 35 3a 34 32 3a 31 31  00 00 64 64 65 32 30 00  |15:42:11..dde20.|
00005040  00 00 00 03 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00005050  00 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
00005060  00 00 00 00 00 00 00 00  00 00 00 00 b8 00 00 00  |................|
00005070  00 00 00 70 01 00 00 00  07 00 00 00 00 00 00 00  |...p............|
00005080  05 00 00 00 00 00 00 00  00 00 00 00 00 10 00 00  |................|
00005090  00 00 00 00 00 00 00 00  01 00 00 00 00 00 00 00  |................|
000050a0  00 00 00 00 00 10 00 00  00 10 00 00 00 00 00 00  |................|
```
on ICL
```
hexdump -C /dev/shm/qemu-bridge-hp-sram-mem | grep -A 10 "6c 00 00 00 00 00 00 70"
00005000  6c 00 00 00 00 00 00 70  00 00 00 00 00 00 00 00  |l......p........|
00005010  00 00 00 00 00 00 00 00  3c 00 00 00 01 00 01 00  |........<.......|
00005020  00 00 27 00 4a 61 6e 20  31 31 20 32 30 31 39 00  |..'.Jan 11 2019.|
00005030  31 35 3a 34 33 3a 31 34  00 00 64 64 65 32 30 00  |15:43:14..dde20.|
00005040  00 00 00 03 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00005050  00 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
00005060  00 00 00 00 00 00 00 00  00 00 00 00 b8 00 00 00  |................|
00005070  00 00 00 70 01 00 00 00  07 00 00 00 00 00 00 00  |...p............|
00005080  05 00 00 00 00 00 00 00  00 00 00 00 00 10 00 00  |................|
00005090  00 00 00 00 00 00 00 00  01 00 00 00 00 00 00 00  |................|
000050a0  00 00 00 00 00 10 00 00  00 10 00 00 00 00 00 00  |................|
```

lgirdwood, 2019-01-14:
I also see
```
ipc.io: *write* 0xc = 	0x00000080 (old 0x80) 	(00000128) 	|....|
ipc.io: *write* 0x8 = 	0xf0000000 (old 0xf0000000) 	(4026531840) 	|....|

```
SKL and KBL appear to work also. 

I would have the CI check check for both the FW boot IPC register write and mbox message header.

xiulipan, 2019-01-15:
@lgirdwood I added the rom support in https://github.com/thesofproject/sof/pull/832
And what I upload is dump from my device
Could you have a look about if there are un-pushed commit from your local repo?

From the log and bridge file I uploaded as attachment I can could see anything like that

```bash
now loading:
 kernel ../sof.git/src/arch/xtensa/sof-kbl.ri
 ROM ../sof.git/src/arch/xtensa/rom-kbl.bin
bridge-io: 0 messages are currently on child queue.
Header $AM1 found at offset 0x284 bytes
found 2 modules
using file offset 0x284
checking module 0 : BRNGUP
 IMR segment 0 file offset 0x2000 IMR addr 0xb000a000 offset 0xa000 size 0x1000
 IMR segment 1 file offset 0x3000 IMR addr 0xb0002000 offset 0x2000 size 0x1000
checking module 1 : BASEFW
 HP segment 0 file offset 0x4000 HP-SRAM addr 0xbe010000 offset 0x10000 size 0x17000
 HP segment 1 file offset 0x21c00 HP-SRAM addr 0xbe02d000 offset 0x2d000 size 0x6000
QEMU 2.11.0 monitor - type 'help' for more information^M
(qemu) l2.io: *write* 0x8 =     0x00000000 (old 0x0)    (00000000)      |....|
shim.io: *write* 0xd0 =         0x00000000 (old 0x0)    (00000000)      |....|
hostwin0.io: *write* 0x4 =      0x00002007 (old 0x0)    (00008199)      |....|
hostwin0.io: *write* 0x0 =      0xbe00e003 (old 0x0)    (3187728387)    |....|
```

lgirdwood, 2019-01-15:
@xiulipan can you provide the command line you are using above.

lgirdwood, 2019-01-15:
> QEMU 2.11.0 monitor - type 'help' for more information^M

There is a ^M on your terminal, maybe something todo with your terminal ? Can you check the registers all look similar. My ciode is here
```
static inline void arch_wait_for_interrupt(int level)
{
	/* can only eneter WFI when at runlevel 0 i.e. not IRQ level */
	if (arch_interrupt_get_level() > 0)
be010cf6:	002816               	beqz	a8, be010cfc <do_task_master_core+0x3c>
		panic(SOF_IPC_PANIC_WFI);
be010cf9:	ffeb65               	call8	be010bb0 <__panic$constprop$2>

	asm volatile("waiti 0");
be010cfc:	007000               	waiti	0
		wait_for_interrupt(0);

		/* now process any IPC messages from host */
		ipc_process_msg_queue();
be010cff:	04ed25               	call8	be015bd0 <ipc_process_msg_queue>
```

xiulipan, 2019-01-15:
I just wrap cmd in scripts/qemu-check.sh
You need a simple modify to run kbl
```diff
diff --git a/scripts/qemu-check.sh b/scripts/qemu-check.sh
index 21c9e375..b1ac657a 100755
--- a/scripts/qemu-check.sh
+++ b/scripts/qemu-check.sh
@@ -1,8 +1,8 @@
 #!/bin/bash

-#SUPPORTED_PLATFORMS=(byt cht bdw hsw apl icl skl kbl cnl)
+SUPPORTED_PLATFORMS=(byt cht bdw hsw apl icl skl kbl cnl)
 # TODO: skl, kbl, cnl is unstable need to check
-SUPPORTED_PLATFORMS=(byt cht bdw hsw apl icl)
+#SUPPORTED_PLATFORMS=(byt cht bdw hsw apl icl)

 rm -f dump-*.txt
```

Here is what I got on my machine with my qemu build

```bash
commit d982f21f06efac09746ffb5fb2eb160d11a29296 (HEAD -> sof-stable, origin/sof-stable, origin/HEAD)
Author: Liam Girdwood <liam.r.girdwood@linux.intel.com>
Date:   Sat Nov 24 13:46:32 2018 +0000

    xtensa: adsp: Add support for manifest v1.5

    Can now load files with version 1.5 of manifest.

    Signed-off-by: Liam Girdwood <liam.r.girdwood@linux.intel.com>
```

```bash
$ ./xtensa-host.sh kbl -k ../sof.git/src/arch/xtensa/sof-kbl.ri -r ../sof.git/src/arch/xtensa/rom-kbl.bin
./xtensa-softmmu/qemu-system-xtensa -cpu broxton -M adsp_kbl -nographic -kernel ../sof.git/src/arch/xtensa/sof-kbl.ri -rom ../sof.git/src/arch/xtensa/rom-kbl.bin
bridge-io: qemu-bridge-kbl-l2-sram fd 10 region 0 at 0x7fbb86d8a000 allocated 2097152 bytes
bridge-io: qemu-bridge-kbl-hp-sram fd 11 region 1 at 0x7fbb86b8a000 allocated 2097152 bytes
bridge-io: qemu-bridge-kbl-lp-sram fd 12 region 4 at 0x7fbb982c6000 allocated 131072 bytes
bridge-io: qemu-bridge-kbl-imr fd 13 region 7 at 0x7fbb86b0a000 allocated 524288 bytes
bridge-io: qemu-bridge-kbl-rom fd 14 region 5 at 0x7fbb982c4000 allocated 8192 bytes
bridge-io: qemu-bridge-cmd-io fd 15 region 8 at 0x7fbb982c3000 allocated 16 bytes
bridge-io: qemu-bridge-res-io fd 16 region 9 at 0x7fbb982c2000 allocated 16 bytes
bridge-io: qemu-bridge-ipc-io fd 17 region 10 at 0x7fbb982c1000 allocated 32 bytes
bridge-io: qemu-bridge-idc0-io fd 18 region 11 at 0x7fbb982c0000 allocated 128 bytes
bridge-io: qemu-bridge-idc1-io fd 19 region 12 at 0x7fbb982bf000 allocated 128 bytes
bridge-io: qemu-bridge-hostwin0-io fd 20 region 13 at 0x7fbb982be000 allocated 8 bytes
bridge-io: qemu-bridge-hostwin1-io fd 21 region 14 at 0x7fbb982bd000 allocated 8 bytes
bridge-io: qemu-bridge-hostwin2-io fd 22 region 15 at 0x7fbb982bc000 allocated 8 bytes
bridge-io: qemu-bridge-hostwin3-io fd 23 region 16 at 0x7fbb982bb000 allocated 8 bytes
bridge-io: qemu-bridge-irq-io fd 24 region 17 at 0x7fbb9814b000 allocated 512 bytes
bridge-io: qemu-bridge-timer-io fd 25 region 18 at 0x7fbb9814a000 allocated 512 bytes
bridge-io: qemu-bridge-mn-io fd 26 region 19 at 0x7fbb98149000 allocated 512 bytes
bridge-io: qemu-bridge-l2-io fd 27 region 20 at 0x7fbb98148000 allocated 64 bytes
bridge-io: qemu-bridge-kbl-shim fd 28 region 2 at 0x7fbb98147000 allocated 256 bytes
bridge-io-mq: added /qemu-io-parent-kbl
bridge-io-mq: added /qemu-io-child-kbl
now loading:
 kernel ../sof.git/src/arch/xtensa/sof-kbl.ri
 ROM ../sof.git/src/arch/xtensa/rom-kbl.bin
bridge-io: 0 messages are currently on child queue.
Header $AM1 found at offset 0x284 bytes
found 2 modules
using file offset 0x284
checking module 0 : BRNGUP
 IMR segment 0 file offset 0x2000 IMR addr 0xb000a000 offset 0xa000 size 0x1000
 IMR segment 1 file offset 0x3000 IMR addr 0xb0002000 offset 0x2000 size 0x1000
checking module 1 : BASEFW
 HP segment 0 file offset 0x4000 HP-SRAM addr 0xbe010000 offset 0x10000 size 0x17000
 HP segment 1 file offset 0x21c00 HP-SRAM addr 0xbe02d000 offset 0x2d000 size 0x6000
[... 32 lines cut ...]
AR08=e0000000 AR09=00000000 AR10=00000000 AR11=00000000
AR12=fefe0856 AR13=be080000 AR14=00000000 AR15=00000000
AR16=00000000 AR17=be00e003 AR18=00000000 AR19=00000000
AR20=b000a07f AR21=be07ffd0 AR22=be00e008 AR23=b000a10c
AR24=00000220 AR25=b0004000 AR26=000000d5 AR27=00000000
AR28=b000a188 AR29=be07ffc0 AR30=be00e048 AR31=00000001
AR32=00000008 AR33=00000000 AR34=00000000 AR35=00000000
AR36=00000000 AR37=00000000 AR38=00000000 AR39=00000000
AR40=00000000 AR41=00000000 AR42=00000000 AR43=00000000
AR44=00000000 AR45=00000000 AR46=00000000 AR47=00000000
AR48=00000000 AR49=00000000 AR50=00000000 AR51=00000000
AR52=00000000 AR53=00000000 AR54=00000000 AR55=00000000
AR56=00000000 AR57=00000000 AR58=00000000 AR59=00000000
AR60=00000000 AR61=00000000 AR62=00000000 AR63=00000000

F00=00000000 (+0.00000000e+00) F01=00000000 (+0.00000000e+00)
F02=00000000 (+0.00000000e+00) F03=00000000 (+0.00000000e+00)
F04=00000000 (+0.00000000e+00) F05=00000000 (+0.00000000e+00)
F06=00000000 (+0.00000000e+00) F07=00000000 (+0.00000000e+00)
F08=00000000 (+0.00000000e+00) F09=00000000 (+0.00000000e+00)
F10=00000000 (+0.00000000e+00) F11=00000000 (+0.00000000e+00)
F12=00000000 (+0.00000000e+00) F13=00000000 (+0.00000000e+00)
F14=00000000 (+0.00000000e+00) F15=00000000 (+0.00000000e+00)

```

xiulipan, 2019-01-15:
@lgirdwood 
Update the qemu to 
```bash
commit 199c9215781df4e25c05fe5d7e541932c9b6fe44 (HEAD -> sof-stable, origin/sof-stable, origin/HEAD)
Merge: c52b125787 e738fd86ba
Author: Liam Girdwood <lgirdwood@gmail.com>
Date:   Mon Jan 14 17:41:01 2019 +0000

    Merge pull request #1 from jgunn0262/sof-stable

    xtensa: hikey960: Add initial qemu support for Hikey 960 DSP
```
And all the issues are gone.

I will update the docker and qemu-check scripts to make sure

> CI check check for both the FW boot IPC register write and mbox message header.

lgirdwood, 2019-01-15:
@xiulipan one more thing, we have https://github.com/thesofproject/sof/pull/776 still to be merged after build system update. So we could enable KBL/SKL after this PR is merged. I have this commit at my end for FW testing so that probably why I boot.

xiulipan, 2019-01-15:
@lgirdwood 
It seems I missed some QEMU commit, update it seems work.

I will update the scripts to directly check mbox/shim or hp-sram/ipc for the ready message.

lgirdwood, 2019-01-23:
Looks like this is resolved now.

closed 2019-01-23 by lgirdwood