thesofproject/sof #848: Stress-test: noise when injecting xrun for HDMI playback

keqiaozhang, 2019-01-16:
This issue is observed on PV recipe. we have exposed this issue before, please refer to #506. But now this issue still can be reproduced under stress test and only happens on HDMI playback.
There's no obvious errors in log or trace and this noise is more like a rustle, it can be recovered after the next xrun. I will try to capture it.

How to reproduce:
1. enable XRUN debug in chromeos-intel-pineview.flavour.config:
+CONFIG_SND_DEBUG=y
+CONFIG_SND_DEBUG_VERBOSE=y
+CONFIG_SND_PCM_XRUN_DEBUG=y
2. build and update kernel
3. sudo su -switch to superuser
4. run the stress test script:
```
#!/bin/bash

xrun_inject()
{
int=0
while [ $int -lt 100000 ]
do
        echo 0 > /proc/asound/card0/pcm5p/sub0/xrun_injection
        sleep 0.5
        echo $int
        let int++
done
}
aplay -Dhw:0,5 -f dat -c 2 $audio.wav & xrun_inject
```
Test recipe：
ChromeOS：R73-11559.0.0
Kernel:https://github.com/ranj063/linux.git/fix/suspend_streams
sof firmware/tplg: glk-005-drop-stable

keqiaozhang, 2019-01-16:
@markyang 
please help to check this issue on UP2. Thanks!

markyang, 2019-01-16:
**Summary:**  
  This issue can also be reproduced on UP2 with PCM512x.

**Test env:**
sof master: 9d10e1a9
kernel sof-dev: eec37ea0
tplg: sof-apl-pcm512x.tplg

lgirdwood, 2019-01-16:
@keqiaozhang can we get a short wav file (of playing sine wave) and audacity capture showing the wave (where individual waves can be seen alongside the noise).

keqiaozhang, 2019-01-17:
@lgirdwood 
FYI
[image]
[hdmi.zip](https://github.com/thesofproject/sof/files/2767616/hdmi.zip)

lgirdwood, 2019-01-23:
@dabekjakub looks like a mix up with DMA R/W pointers for HDMI

dabekjakub, 2019-01-24:
@slawblauciak shouldnt xrun have been handled in kernel ?

lgirdwood, 2019-01-24:
@dabekjakub @slawblauciak yes, XRUNs are handled in the kernel by sending STOP, PREPARE and then START to audio drivers and FWs. This test allows user space to inject this condition to test how driver and FW react to STOP, PREPARE and START. It does look in this case that maybe FW is not resetting pointers for the restart correctly.

slawblauciak, 2019-02-18:
The firmware is not capable of resetting the pointers. It should be handled by alsa/driver.
https://github.com/thesofproject/linux/issues/660

lgirdwood, 2019-02-25:
@slawblauciak the FW must be able to at least STOP a stream if it detects unrecoverable XRUNs and notify the host. That way the we minimise noise and allow the host to recover. This also allows pipeline developers to debug their code.

lbetlej, 2019-03-27:
agree , let's add STOP functionality to FW

tlauda, 2019-06-06:
@keqiaozhang Can you reverify? There were some changes in both HDMI and XRUN handling.

jajanusz, 2019-06-10:
@keqiaozhang @markyang ping. Can you recheck?

keqiaozhang, 2019-06-11:
@jajanusz @tlauda 
Verified this issue with latest code base and got 1K iterations pass. This issue can be closed now.

Test recipe:
sof: master/c0c858
kernel: sof-dev/a50a2
tplg: sof-glk-da7219.tplg
platform: GLK-I2S/Ubuntu 18.04

closed 2019-06-11 by keqiaozhang