thesofproject/sof #856: hda-dma will timeout at waiting for DGCS_BF true

keyonjie, 2019-01-17:
On APL platforms, we are randomly failed at hda_dma_host_preload() where hda-dma will timeout at waiting for DGCS_BF true.

Using simultaneous pipelines stress test can duplicate this easily. Scrip here:
https://github.com/thesofproject/sof/blob/master/tools/test/qa/tests/stability_test/multiplex_pipeline-test.sh

Setting DGCS_L1ETP at hda_dma_set_config() or Force Host DMA to exit L1 make the reproduce rate change, but can't eliminate it.

keyonjie, 2019-01-17:
Setting it to P1 as it is blocker now.

lgirdwood, 2019-01-17:
@keyonjie https://github.com/thesofproject/sof/pull/804 has been merged now. @slawblauciak fyi

keyonjie, 2019-01-18:
> @keyonjie #804 has been merged now, does it help ?. @slawblauciak fyi

I haven't tried it, but I don't think so, #804 is change for link dma only, no relationship with host dma BF waiting.

keyonjie, 2019-01-18:
@mmaka1 I have tried what Marcin suggested(https://github.com/thesofproject/sof/pull/783#discussion_r244994892) -- Setting DGCS_L1ETP at hda_dma_set_config() or Force Host DMA to exit L1 -- the result is that:
it makes the reproduce rate change, but can't eliminate it.

@mmaka1 can you help to look in this issue?

lgirdwood, 2019-01-18:
@keyonjie you have provided two different solutions, but you have not said which one improved the result and by how  much it improved ? Can you also try and add some delay after DMA L1 exit before the next IO operation.

wenqingfu, 2019-01-18:
looks like the story line is #723 (FW hang, how about time out?) -> #783 (you got it!, timeout is better than freeze) -> #800 (caused burst noise!) -> #856 (Oh no, there is noise, need a full fix). 

add @dabekjakub to decide who will own it.

keyonjie, 2019-01-21:
> @keyonjie you have provided two different solutions, but you have not said which one improved the result and by how much it improved ? Can you also try and add some delay after DMA L1 exit before the next IO operation.

@lgirdwood the result here looks random to me, so I can't say which one improve and by how much it improved? Then I realized I can't try that blindly, I need to understand how it should work, that's why I dropped an email about it.

keyonjie, 2019-01-21:
> looks like the story line is #723 (FW hang, how about time out?) -> #783 (you got it!, timeout is better than freeze) -> #800 (caused burst noise!) -> #856 (Oh no, there is noise, need a full fix).
> 
> add @dabekjakub to decide who will own it.

@wenqingfu yes, that's how the story happened, @dabekjakub can you help find a owner for it?

dabekjakub, 2019-01-21:
@keyonjie @wenqingfu  I think that @slawblauciak is our specialist in hda-dma related issues. However i will reasign this after we sync on bugscrub.

keyonjie, 2019-01-21:
Add more info as below:
At this BF fail cases, usually buffer is not empty(DGCS_BNE==1), and write pointer(DGBWP) is at 0x20 offset(period size is 0xc0, that means, dgbwp==0x20 or 0xe0), such as:
dgcs: 0x86810120 dgbrp: 0x0, dgbwp: 0xe0

dabekjakub, 2019-01-22:
@keyonjie Could You check this one with non-smp architecture ?

keyonjie, 2019-01-23:
@dabekjakub I actually checked this always with UP architeture.

keyonjie, 2019-01-24:
> @keyonjie is result the same with UP vs SMP ?

I think SMP fails much easier comparing with UP.

slawblauciak, 2019-01-29:
@keyonjie how did you verify that the buffer has been in fact fully preloaded if the register values don't indicate it?

keyonjie, 2019-02-18:
> @keyonjie how did you verify that the buffer has been in fact fully preloaded if the register values don't indicate it?

@slawblauciak I dumped the contents of those buffers to verify them.

wenqingfu, 2019-03-21:
Should this issue be closed by thesofproject/linux#705? @RanderWang @keyonjie @slawblauciak . If so, please close it

RanderWang, 2019-03-22:
yes, it should be closed by thesofproject/linux#705

closed 2019-03-22 by RanderWang